pyvisa/pyvisa #701: Communication issue using Keysight 34461A with pyvisa-py

mboskovic994, 2022-11-02:
Hi,

I'm using pyvisa to acquire data from DMM Keysight 34461A. Code: 

import pyvisa
rm=pyvisa.ResourceManager()
rm.list_resources()
print(rm.list_resources())
dmm=rm.open_resource('USB::0x2A8D::0x1401::MY53211121::0::INSTR')  
print(dmm.query('*IDN?'))
dmm.write("*RST")
dmm.write('DISP:VIEW TCHart')
dmm.write('CONF:VOLT:DC')
dmm.write('VOLT:DC:NPLC 100')
dmm.write("TRIG:SOUR INT")
import time
import matplotlib.pyplot as plt
import numpy as np
i=int(input('broj? '))
y=[]
timeList=[]
startTime=time.time()
for i in range(0,i):
    dmm.write("SAMP:COUN 1")
    dmm.write('INIT')
    dmm.write('*TRG')
    a=(dmm.query('FETC?'))
    print(a)
    y.append(float(a))
     timeList.append(float(time.time() - startTime))
     plt.plot(timeList,y , '-o',color='red')
    plt.pause(0.000000001)
    import pandas as pd
    list1 = timeList
    list2 = y
    col1 = "time"
    col2 = "voltage"
    data = pd.DataFrame({col1:list1,col2:list2})
    data.to_excel(r'/home/paul/python/python/p.xlsx', index = False)
plt.show()

At the beginning, it works, fine, data is read every 2 seconds (100 PLC). However after couple minutes it starts lagging. Interval between data acquisition becomes longer, and eventually becomes 10 seconds. 

Does anyone experienced same problem? 

Marko

yberlin, 2022-11-30:
I use visa to communicate with DSOX3034T.
Before, all things went correctly. Recently, I collect a power trace from NEWAE CW308 board, with an amplifier offered by NEWAE. However, it will get stuck when the code is running. And the time of the stuck is different. Sometimes I can get thousands of traces, but sometimes it only obtain handreds of traces and then raise an error.
`Traceback (most recent call last):
  File "C:\Users\zyr\AppData\Local\Programs\Python\Python37\lib\code.py", line 90, in runcode
    exec(code, self.locals)
  File "<input>", line 1, in <module>
  File "C:\Program Files\JetBrains\PyCharm 2022.2.1\plugins\python\helpers\pydev\_pydev_bundle\pydev_umd.py", line 198, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm 2022.2.1\plugins\python\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "E:/20221009test/capture/cap_for_kyber.py", line 157, in <module>
    preamble, waveform = scope.record_SaveOne(idx)
  File "E:\20221009test\capture\visa_ctrl_DSOX3034T.py", line 384, in record_SaveOne
    Wav_Data = self.data_valid_fix(self.myScope.read_raw())
  File "E:\20221009test\capture\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 405, in read_raw
    return bytes(self._read_raw(size))
  File "E:\20221009test\capture\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "E:\20221009test\capture\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "E:\20221009test\capture\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "E:\20221009test\capture\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).`

LongnoseRob, 2022-11-30:
Did you try a different USB port or USB-Cable?

MatthieuDartiailh, 2022-11-30:
Also could you test https://github.com/pyvisa/pyvisa-py/pull/335 ? Early report suggest it improves the stability of USB resource access.

yberlin, 2022-12-01:
> 

Thanks for reply!
Actually, we have tried two different PC, with different USB wires. And all of them will raise this error but without a pattern. Sometimes I tried to add some delay in python code, and it has some help, the collecting code will run more time. And I checked the error time by keysight monitor. I found that this error raises every time when the scope returns the trace data.

yberlin, 2022-12-01:
> Also could you test [pyvisa/pyvisa-py#335](https://github.com/pyvisa/pyvisa-py/pull/335) ? Early report suggest it improves the stability of USB resource access.

Thanks for the reply!
Do you mean I should clone the #335 branch and re-install this lib? But how can I use this exact repo file to install the lib? I mean in the Readme in this repo, it shows that do the installation by pip install.

MatthieuDartiailh, 2022-12-01:
You can install the branch directly using `pip install git+https://github.com/pyvisa/pyvisa-py.git@usb-fixes`

MatthieuDartiailh, 2022-12-05:
Do you observe the same behavior if you disable the live plot an systematic saving ?

mboskovic994, 2022-12-05:
> 

Systematic saving does not change anything, but if live plot is disabled, interval becomes constant (little bit bigger than corresponding PLC time).

MatthieuDartiailh, 2022-12-05:
Matplotlib is not really good at live plotting. Still you can achieve better results using line.set_ydata (and set_xdata in your case). You can take inspiration from this example https://matplotlib.org/stable/gallery/animation/simple_anim.html (just focusing on the plot update logic, you do not need the animation bits).

MatthieuDartiailh, 2022-12-05:
pyqtgraph is usually considered faster, but I would personally just try improving your use of matplotlib first. It will likely be less effort.

mboskovic994, 2022-12-05:
> pyqtgraph is usually considered faster, but I would personally just try improving your use of matplotlib first. It will likely be less effort.

Thank you very much! I will improve code and check results.