pyvisa/pyvisa #706: Issue with GPIB-USB_HS+ Connector for Agilent 6681a Power Supply with PyVISA

moberry123, 2022-12-01:
Hi, I am trying to connect my linux laptop (Ubuntu 20.04 OS running 5.15.0-50-generic kernel) to my Agilent 6681a power supply using a National Instruments GPIB-USB_HS+ Connector. I have NI VISA working and am able to query the instrument:

**XXXX$ sudo ibtest
[sudo] password for user: 
Do you wish to open a (d)evice or an interface (b)oard?
	(you probably want to open a device): d
enter primary gpib address for device you wish to open [0-30]: 6
trying to open pad = 6 on /dev/gpib0 ...
You can:
	w(a)it for an event
	write (c)ommand bytes to bus (system controller only)
	send (d)evice clear (device only)
	change remote (e)nable line (system controller only)
	(g)o to standby (release ATN line, system controller only)
	send (i)nterface clear (system controller only)
	ta(k)e control (assert ATN line, system controller only)
	get bus (l)ine status (board only)
	go to local (m)ode
	change end (o)f transmission configuration
	(q)uit
	(r)ead string
	perform (s)erial poll (device only)
	change (t)imeout on io operations
	request ser(v)ice (board only)
	(w)rite data string
	send group e(x)ecute trigger (device only)
: d
Device clear sent.
gpib status is: 
ibsta = 0x100  < CMPL >
iberr= 0

ibcntl = 0
You can:
	w(a)it for an event
	write (c)ommand bytes to bus (system controller only)
	send (d)evice clear (device only)
	change remote (e)nable line (system controller only)
	(g)o to standby (release ATN line, system controller only)
	send (i)nterface clear (system controller only)
	ta(k)e control (assert ATN line, system controller only)
	get bus (l)ine status (board only)
	go to local (m)ode
	change end (o)f transmission configuration
	(q)uit
	(r)ead string
	perform (s)erial poll (device only)
	change (t)imeout on io operations
	request ser(v)ice (board only)
	(w)rite data string
	send group e(x)ecute trigger (device only)
: 6
invalid selection
You can:
	w(a)it for an event
	write (c)ommand bytes to bus (system controller only)
	send (d)evice clear (device only)
	change remote (e)nable line (system controller only)
	(g)o to standby (release ATN line, system controller only)
	send (i)nterface clear (system controller only)
	ta(k)e control (assert ATN line, system controller only)
	get bus (l)ine status (board only)
	go to local (m)ode
	change end (o)f transmission configuration
	(q)uit
	(r)ead string
	perform (s)erial poll (device only)
	change (t)imeout on io operations
	request ser(v)ice (board only)
	(w)rite data string
	send group e(x)ecute trigger (device only)
: w
enter a string to send to your device: *IDN?
sending string: *IDN?

gpib status is: 
ibsta = 0x2100  < END CMPL >
iberr= 0

ibcntl = 6
You can:
	w(a)it for an event
	write (c)ommand bytes to bus (system controller only)
	send (d)evice clear (device only)
	change remote (e)nable line (system controller only)
	(g)o to standby (release ATN line, system controller only)
	send (i)nterface clear (system controller only)
	ta(k)e control (assert ATN line, system controller only)
	get bus (l)ine status (board only)
	go to local (m)ode
	change end (o)f transmission configuration
	(q)uit
	(r)ead string
	perform (s)erial poll (device only)
	change (t)imeout on io operations
	request ser(v)ice (board only)
	(w)rite data string
	send group e(x)ecute trigger (device only)
: r
enter maximum number of bytes to read [1024]: 100
trying to read 100 bytes from device...
received string: 'HEWLETT-PACKARD,6681A,0,fA.01.02sA.00.03pA.01.02
'
Number of bytes read: 49
gpib status is: 
ibsta = 0x2100  < END CMPL >
iberr= 0

ibcntl = 49**

However, when I try the test script, my power supply does not show up in the output:
>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
>>> print(rm.list_resources())
('ASRL5::INSTR',)

I was thinking this could be a permissions issue, so I added this line to /etc/udev/rules.d/99-com.rules:
SUBSYSTEM=="usb", MODE="0666", GROUP="usbusers"

This didn't seem to change anything.

I also tried to connect to the power supply directly and got this error:
>>> import pyvisa
>>> rm = pyvisa.ResourceManager('@py')
>>> check = rm.open_resource('GPIB::6::INSTR')
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: Permission denied
libgpib: error in is_cic()!
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mary.sobernheim/.local/lib/python3.8/site-packages/pyvisa/highlevel.py", line 3284, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/mary.sobernheim/.local/lib/python3.8/site-packages/pyvisa/resources/resource.py", line 278, in open
    self.session, status = self._resource_manager.open_bare_resource(
  File "/home/mary.sobernheim/.local/lib/python3.8/site-packages/pyvisa/highlevel.py", line 3209, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/mary.sobernheim/.local/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 167, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/home/mary.sobernheim/.local/lib/python3.8/site-packages/pyvisa_py/sessions.py", line 325, in __init__
    self.after_parsing()
  File "/home/mary.sobernheim/.local/lib/python3.8/site-packages/pyvisa_py/gpib.py", line 268, in after_parsing
    self.interface = Gpib(
  File "/home/mary.sobernheim/.local/lib/python3.8/site-packages/gpib_ctypes/Gpib.py", line 31, in __init__
    self.id = gpib.dev(name, pad, sad, timeout, send_eoi, eos_mode)
  File "/home/mary.sobernheim/.local/lib/python3.8/site-packages/gpib_ctypes/gpib/gpib.py", line 303, in dev
    raise GpibError("dev")
gpib_ctypes.gpib.gpib.GpibError: dev() error: Errno 9, Bad file descriptor

Any help is appreciated! I added info about the instrument and pyvisa below. Thank you

**Instrument details**
------------------
* Model: Agilent 6681a Power Supply
* Communication: GPIB (NI GPIB-U http://ridl.cfd.rit.edu/products/manuals/Agilent/power%20supplies/CD1/Model/669xprog.pdfSB_HS+)
* Link to the documentation (if available):

**Output of `pyvisa-info`:**
mary.sobernheim@TT-PF1JEX54-HQ:~/Downloads/linux-gpib-4.3.5/linux-gpib-user-4.3.5/language/python$ pyvisa-info
Machine Details:
   Platform ID:    Linux-5.15.0-53-generic-x86_64-with-glibc2.29
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.8.10
   Compiler:       GCC 9.4.0
   Bits:           64bit
   Build:          Jun 22 2022 20:18:18 (#default)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.22.5.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
   py:
      Version: 0.5.3
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.5')
      GPIB INTFC: Available via Linux GPIB (b'4.3.5')

dphaas, 2022-12-23:
@moberry123 did you make any progress with this?
I think in "check = rm.open_resource('GPIB::6::INSTR')" GPIB should be GPIB**0** (0 or whatever the bus number is).  "6" is the instrument address on bus 0.
The response "libgpib: ibBoardOpen failed to open device file /dev/gpib0" I would take to mean the GPIB-USB_HS+ is not found.

pyvisa-shell is a handy debugging tool:
`PS C:\Python\T+M> pyvisa-shell
Welcome to the VISA shell. Type help or ? to list commands.
(visa) ?
Documented commands (type help <topic>):
EOF  attr  close  exit  help  list  open  query  read  termchar  timeout  write     

(visa) list
( 0) USB0::0x0699::0x0401::C021095::INSTR
     alias: OSCOPE
( 1) TCPIP0::A-34972A-26309::inst0::INSTR
     alias: DAQ-LAN
( 2) ASRL3::INSTR
     alias: COM3
( 3) ASRL7::INSTR
     alias: COM7
( 4) ASRL18::INSTR
     alias: GPP-2323
( 5) GPIB0::3::INSTR
     alias: FUNC_GEN
(visa)`

The "list" command should list all the instruments reported by VISA.

MatthieuDartiailh, 2022-12-23:
Also did you try connecting directly using the ivi backend instead of the pyvisa-py backend ?

OldDaiBii, 2023-05-05:
Hello, I have also encountered a similar problem.I would appreciate your help.
pyvisa-info
Machine Details:
   Platform ID:    Linux-5.19.0-41-generic-x86_64-with-glibc2.35
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.10.6
   Compiler:       GCC 11.3.0
   Bits:           64bit
   Build:          Mar 10 2023 10:55:28 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.23.3.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 24118016
         Spec. Version: 7340032
   py:
      Version: 0.6.0
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      VICP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

I am unable to return errors when I communicate using pyvisa.
>>> import pyvisa as visa
>>> rm = visa.ResourceManager()
>>> print(rm.list_resources())
('ASRL1::INSTR',)

And when I try to use pyvisa-py
>>> import pyvisa as visa
>>> rm = visa.ResourceManager('@py')
>>> print(rm.list_resources())
('ASRL/dev/ttyS0::INSTR',)

However, I am able to send *IDN? and receive a response when using ibtest.

...
: r
enter maximum number of bytes to read [1024]: 1024
trying to read 1024 bytes from device...
received string: 'Rigol Technologies,DM3054,DM3E131600009,03.12.00.03.11.00
'
Number of bytes read: 58
gpib status is: 
ibsta = 0x2100  < END CMPL >
iberr= 0

ibcntl = 58
...

MatthieuDartiailh, 2023-05-05:
I have no good idea as to why NI-VISA fails to see your device. PyVISA-py explictly says you need extra library to get GPIB support so no surprise there.
Did you try opening the resource since you know its address ?

OldDaiBii, 2023-05-05:
There is a display when I use nimax on a Windows system.
GPIB0::4::INSTR
[image]
Should I try using GPIB0::4::INSTR directly?

MatthieuDartiailh, 2023-05-05:
I would say it is worth trying yes.

OldDaiBii, 2023-05-05:
sorry sir, my attempt has failed.When I try to open it directly, I get a mismatch error prompt. Could you please advise me on how to set it up properly?
>>> inst = rm.open_resource('GPIB0::4::INSTR')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/highlevel.py", line 3247, in open_resource
    info = self.resource_info(resource_name, extended=True)
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/highlevel.py", line 3174, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/ctwrapper/functions.py", line 2077, in parse_resource_extended
    ret = library.viParseRsrcEx(
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/ctwrapper/highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_INTF_NUM_NCONFIG (-1073807195): The interface type is valid but the specified interface number is not configured.

I'm getting an error when using pyvisa-py.
>>> import pyvisa as visa
>>> rm = visa.ResourceManager('@py')
>>> inst = rm.open_resource('GPIB0::4::INSTR')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/highlevel.py", line 3284, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/resources/resource.py", line 282, in open
    self.session, status = self._resource_manager.open_bare_resource(
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/highlevel.py", line 3209, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python3.10/dist-packages/pyvisa_py/highlevel.py", line 167, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/usr/local/lib/python3.10/dist-packages/pyvisa_py/sessions.py", line 267, in __init__
    raise ValueError(msg)
ValueError: Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
No module named 'gpib'

MatthieuDartiailh, 2023-05-05:
It looks like the adapter is not properly configured to be used by NI VISA. However I have no idea how to fix it, you may have better luck on a NI forum. 
To use pyvisa-py as the error message says please install linux-gpib (using apt should work)

OldDaiBii, 2023-05-05:
Thank you very much for your help. I will keep trying to solve this problem.

OldDaiBii, 2023-05-06:
I'm very happy that I solved the problem. After installing python-dev, I executed "python ./setup.py install" in linux-gpib-user/language/python directory of gpib driver. When I ran pyvisa-info again.
py:
      Version: 0.7.0
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.5')
      GPIB INTFC: Available via Linux GPIB (b'4.3.5')
      VICP INSTR:
         Please install PyVICP to use this resource type.

Although there was an exception during the use of pyvisa-py, it successfully returned the correct address in the end.
>>> import pyvisa as visa
>>> rm = visa.ResourceManager('@py')
>>> print(rm.list_resources())
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
('ASRL/dev/ttyS0::INSTR', 'GPIB0::4::INSTR')
I successfully received the information I needed during the communication.
>>> print(inst.query("*IDN?\n"))
Rigol Technologies,DM3054,DM3E131600009,03.12.00.03.11.00

Thank you very much for your help！

MatthieuDartiailh, 2023-05-06:
Happy you got things to work. The spurious message from libgpib comes from the fact that we need to try every single address to find listeners and hence find a lot of invalid ones.

closed 2023-05-06 by MatthieuDartiailh