pyvisa/pyvisa #707: [COM] Communication issue with CTS climate chamber using TCPIP::SOCKET

PyroTom, 2022-12-06:
Instrument details
------------------
* Model: CTS 
* Communication: TCPIP SOCKET
* Link to the documentation (if available): [ASCII Protocol](https://www.cts-umweltsimulation.de/en/component/emdown/downloadfile/10565.html?id=10565&Itemid=358)

Output of `pyvisa-info`
-----------------------
```text
Machine Details:
   Platform ID:    Windows-10-10.0.19045-SP0
   Processor:      Intel64 Family 6 Model 85 Stepping 7, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\hw0mes00local\.venv\scripts\python.exe
   Version:        3.8.10
   Compiler:       MSC v.1928 64 bit (AMD64)
   Bits:           64bit
   Build:          May  3 2021 11:48:03 (#tags/v3.8.10:3d8993a)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 22020096
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 22020096
         Spec. Version: 5244928
```

First issue verification
------------------------

### what works (directly using a socket)

```python
import socket

sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
sock.connect(("10.0.1.16", 1080))

sock.sendall(b"A0")  # sends 'A0'
reply = sock.recv(64)  # receives 'A0 25.0 25.0'
```

### using pyvisa

#### code example
```python
import pyvisa

rm = pyvisa.ResourceManager()
chamber = rm.open_resource("TCPIP::10.0.1.16::1080::SOCKET")
chamber.write_termination = chamber.read_termination = None
chamber.encoding = "ascii"

chamber.write("A0")  # sends 'A0'
chamber.read()  # Timeout
```

#### NI io-trace

| number | description | state |
| --- | --- | --- |
| 47 | viWrite(TCPIP0::10.0.1.16::1080::SOCKET (0x02), "A0", 2 (0x2), 2 (0x2)) | 0 |
| 48 | viRead(TCPIP0::10.0.1.16::1080::SOCKET (0x02), "A0 025.0 025.0", 64 (0x40), 14 (0xE)) | 0xBFFF0015 |

PyroTom, 2022-12-06:
Unfortunately this does not work.
I still get a timeout error.

The only way to get it run is when i exactly use the number of bytes to receive which are send by the device which varies depending on the command. 😞 
The read never succeeds because i have 64 bytes buffer and receive 2 to 14 bytes as answer depending on the command.

MatthieuDartiailh, 2022-12-13:
You may be better off using the socket module directly. Your instrument does not seem to implement any protocol to ensure the message received is complete which is fairly central to VISA and trying to go around it will likely prove painful.

9y2070m, 2024-06-24:
I used the `socket` interface directly.

closed 2024-06-24 by 9y2070m