pyvisa/pyvisa #708: Is there anything changed with TCPIP SOCKET connection after 1.12 version?

huizhiw, 2022-12-08:
The instrument I'm using is called AV4051B. I tested pyvisa library before Nov 5th and it works well. Later on, I tried the same code today, and it doesn't work anymore. It tells me that my library doesn't have VISA implementation... I installed pyvisa-py as told, which also updated my library to 1.12. However, I ran the code and got the error as below:

```
Exception: could not connect: StatusCode.error_timeout
```
```
The code:
import pyvisa as visa  
rm = visa.ResourceManager()
inst = rm.open_resource('TCPIP0::192.168.1.100::5000::SOCKET')
print(inst)
```
Is there any change on TCPIP SOCKET implementation in new version? 
I really have no idea what's going on here. Any suggestion? Thanks in advance!

To Reproduce
--------------
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Output of `pyvisa-info`**

Machine Details:
   Platform ID:    Windows-10-10.0.19044-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     E:\miniConda\envs\visaConnect\python.exe
   Version:        3.10.6
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Oct 24 2022 16:02:16 (#main)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.3
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

LongnoseRob, 2022-12-08:
Have you check the connection to your instrument (ping, telnet?) just to be sure that there is no other problem with the conneciton to the device?

huizhiw, 2022-12-09:
So I tested the following commends from my computer:
ping 192.168.1.100               - work
telnet 192.168.1.100 5000    - can't access and connection refused

I guess it's the ip conflict problem
My classmate confirm my guess:
- There are several instruments in our lab using the static IP configuration and some of them are configured in the same IP: 192.168.1.100
- when they are connected to the same router, it will cause this conflict
- The visa library is working and there's no bug here
I configure the IP and resolve this problem,

closed 2022-12-09 by huizhiw

MatthieuDartiailh, 2022-12-09:
Happy you got things working.