pyvisa/pyvisa #710: pyvisa-shell attr command appears to truncate the last character of VI_ATTR_RSRC_IMPL_VERSION

dphaas, 2022-12-08:
pyvisa-shell attr command truncates the last character of VI_ATTR_RSRC_IMPL_VERSION

I am using NI VISA 2022 Q3 version.  It is interesting to note that the Spec. Version: 5244928 (0x00500800) is indicating VISA version 5.8 when version 7.2 is current version.   Also, Impl. Version: 23069952 (0x01600500) leads me to believe that NI does not update this field.  Do you have any contacts at NI that could clarify if these fields are updated with new releases?

To Reproduce
--------------
Steps to reproduce the behavior:
1. Run pyvisa-info.
2. Run pyvisa-shell.
3. Execute rm.visalib.get_debug_info().
4. Connect to an instrument.
5. Run (open) attr
6. Compare results of each.

pyvisa-info:
    Impl. Version: 23069952 (0x01600500)
    Spec. Version: 5244928 (0x00500800)

pyvisa-shell attr command:
    |  VI_ATTR_RSRC_IMPL_VERSION  | 1073676291 |   implementation_version   |              2306995
    |  VI_ATTR_RSRC_SPEC_VERSION  | 1073676656 |        spec_version        |              5244928

rm.visalib.get_debug_info():
    Impl. Version: 23069952
    Spec. Version: 5244928

**Output of `pyvisa-info`**
Machine Details:
   Platform ID:    Windows-10-10.0.19045-SP0
   Processor:      Intel64 Family 6 Model 79 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\dhaas\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundationPython.3.10_qbz5n2kfra8p0\python.exe
   Version:        3.10.9
   Compiler:       MSC v.1934 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec  6 2022 20:01:21 (#tags/v3.10.9:1dd9be6)
   Unicode:        UCS4

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928

PS C:\Users\dhaas\DPH\OneDrive\DPH_Documents\Remote-Haas\Python> pyvisa-info -b "c:\no_visa.dl"
Machine Details:
   Platform ID:    Windows-10-10.0.19045-SP0
   Processor:      Intel64 Family 6 Model 79 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\dhaas\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundationPython.3.10_qbz5n2kfra8p0\python.exe
   Version:        3.10.9
   Compiler:       MSC v.1934 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec  6 2022 20:01:21 (#tags/v3.10.9:1dd9be6)
   Unicode:        UCS4
PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928

dphaas, 2022-12-09:
As a data point, I switched to R&S VISA and VI_ATTR_RSRC_IMPL_VERSION is not truncated but it may be because it is only seven characters long, "5245960" as opposed to the eight characters in the NI-VISA.

If you are going to touch this the VI_ATTR_RSRC_IMPL_VERSION and VI_ATTR_RSRC_SPEC_VERSION are defined as hex readable not decimal.  (see Impl. Version: and Spec. Version: below for example)   Impl. Version: 0x500c08 = FileVersion: 5.12.8

`StringFileInfo: C:\WINDOWS\system32\rsvisa32.dll
        Comments: None
        InternalName: RsVisa32.dll
        ProductName: R&S VISA
        CompanyName: Rohde & Schwarz GmbH & Co. KG
        LegalCopyright: © 2022 Rohde & Schwarz GmbH & Co. KG
        ProductVersion: 5.12.8
        FileDescription: R&S VISA Library
        LegalTrademarks: None
        PrivateBuild: None
        FileVersion: 5.12.8
        OriginalFilename: RsVisa32.dll
        SpecialBuild: None

VISA debug print:
        Version: 1.12.0 (bundled with PyVISA)
        #1: C:\WINDOWS\system32\visa32.dll
                found by: auto
                bitness: 64
                Vendor: Rohde & Schwarz GmbH
                Impl. Version: 0x500c08
                Spec. Version: 0x500800
        #2: C:\WINDOWS\system32\visa64.dll
                found by: auto
                bitness: 64
                Vendor: Rohde & Schwarz GmbH
                Impl. Version: 0x500c08
                Spec. Version: 0x500800

Python version:   3.10.9 (tags/v3.10.9:1dd9be6, Dec  6 2022, 20:01:21) [MSC v.1934 64 bit (AMD64)]       

List of instruments known to VISA:

Instrument 1 : ASRL18::INSTR
  Interface type: 4
    board number: 18
  Resource class: INSTR
            name: ASRL18::INSTR
           alias: GPP-2323
        Baudrate: 115200, Bits: 8, Parity: 0, FlowControl: 0

Instrument 2 : USB0::0x0699::0x0401::C021095::INSTR
  Interface type: 7
    board number: 0
  Resource class: INSTR
            name: USB0::1689::1025::C021095::0::INSTR
           alias: OSCOPE

Instrument 3 : USB0::0x0957::0x2007::MY49026309::INSTR
  Interface type: 7
    board number: 0
            name: USB0::2391::8199::MY49026309::0::INSTR
           alias: DAQ

Instrument 4 : ASRL3::INSTR
  Interface type: 4
    board number: 3
  Resource class: INSTR
            name: ASRL3::INSTR
           alias: None
        Baudrate: 9600, Bits: 8, Parity: 0, FlowControl: 0

Instrument 5 : ASRL7::INSTR
  Interface type: 4
    board number: 7
  Resource class: INSTR
            name: ASRL7::INSTR
           alias: None
        Baudrate: 9600, Bits: 8, Parity: 0, FlowControl: 0`

MatthieuDartiailh, 2022-12-13:
Thanks for the report. Since the VISA specs is rather stable I am not surprised the value dies not change much  where did you get the value of 7.2 ?

dphaas, 2022-12-13:
https://ivifoundation.org/downloads/Shared%20Components/VISA%20Shared%20Components%20Release%20Notes_.txt

MatthieuDartiailh, 2022-12-14:
It is possible the NI instrumentation is lagging behind the specs. Keysight (https://www.keysight.com/fr/en/lib/software-detail/computer-software/io-libraries-suite-downloads-2175637.html) mention support for hislip 2.0 so it could be interesting checking the reported spec implementation version.

dphaas, 2022-12-14:
You got me curious about Keysight VISA.  I had tried previously installing this without success.  I just downloaded the latest installer, IOLibSuite_18_2_28229.exe, and cannot get PyVISA to recognize the driver (error below). NI and R&S VISA installs work fine with PyVISA.  I was frustrated that NI VISA does not pass the instrument configurations (e.g. serial baud rate) on to PyVISA.  R&S VISA seems to pass instrument configuration as expected.  I guess I'll stick with R&S since it works and is about 90% smaller install then the bloat that Keysight and NI stick into their install.

VISA files after Keysight install:
VISA files found in C:\WINDOWS\system32\:
         C:\WINDOWS\system32\agvisa32.dll
         C:\WINDOWS\system32\agvisaext.dll
         C:\WINDOWS\system32\ktvisa32.dll
         C:\WINDOWS\system32\ktvisaext.dll
         C:\WINDOWS\system32\visa32.dll
         C:\WINDOWS\system32\visa64.dll
         C:\WINDOWS\system32\visaConfMgr.dll
         C:\WINDOWS\system32\visaext.dll
         C:\WINDOWS\system32\visaUtilities.dll 

Error:
`Traceback (most recent call last):
  File "c:\Users\dhaas\DPH\OneDrive\DPH_Documents\Remote-Haas\Python\T+M\pyvisa_scan.py", line 123, in <module>
    main()
  File "c:\Users\dhaas\DPH\OneDrive\DPH_Documents\Remote-Haas\Python\T+M\pyvisa_scan.py", line 93, in main
    rm = visa.ResourceManager()
  File "C:\Users\dhaas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyvisa\highlevel.py", line 3001, in __new__
    obj.session, err = visa_library.open_default_resource_manager()
  File "C:\Users\dhaas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyvisa\ctwrapper\functions.py", line 1871, in open_default_resource_manager
    ret = library.viOpenDefaultRM(byref(session))
  File "C:\Users\dhaas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\dhaas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.`

MatthieuDartiailh, 2022-12-15:
I use Keysight VISA regularly so I will try updating and see what spec version I get. In some cases, Python is unhappy because the actual dll is not in System32 and one need to add the Keysight folder to the trusted locations from which dll can be loaded.

dphaas, 2022-12-16:
I am back on NI-VISA and the problem does not reproduce.  I think this may have been simply a problem with my terminal width moving the last character to then next line.  I see that when I pasted the line from pyvisa-shell output I did not capture the closing "|" character.  It was probably wrapped onto the following line.

Closing issue due to operator error.  Sorry for the annoyance.

closed 2022-12-16 by dphaas

MatthieuDartiailh, 2022-12-16:
Thanks for double-checking. I was a bit surprised that you observed the issue in a single place since I think the formatting logic lives in a single location but I had no time to double check.