pyvisa/pyvisa #717: Problem of checking code with  static type checker - mypy

Matjaz-B, 2022-12-14:
I'm trying to up my python game by using type checking. So far everything was going as expected, but now I've stumbled across a problem where I can not google a solution.

So when I run `mypy` over my code for communication with instrument I get errors.

As far as I understand the problem, is that open_resourse returns a "Resource" without `query` attribute. Which is correct - as query method is attached (registered) during runtime (again, as I understand it).

I imagine, that this is something where mypy is not really helpfull (static analysis, runtime registrations?). But I'd still like to use type checks when using visa module.
Any suggestion how to use static type checking (mypy) when `pyvisa `is used?

Thanks

To Reproduce
--------------
Content of pyvisa_test.py
```python
import pyvisa

rm = pyvisa.ResourceManager()
osc = rm.open_resource("TCPIP::192.168.99.104::INSTR")
print(osc.query("*IDN?"))
```
Result of `mypy pyvisa_test.py`

```
$ mypy pyvisa_test.py
pyvisa_test.py:5: error: "Resource" has no attribute "query"
Found 1 error in 1 file (checked 1 source file)
```

**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Linux-5.10.102.1-microsoft-standard-WSL2-x86_64-with-glibc2.29
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.8.10
   Compiler:       GCC 9.4.0
   Bits:           64bit
   Build:          Nov 14 2022 12:59:47 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.3
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```

MatthieuDartiailh, 2022-12-14:
The problem is that the type returned depend on the resource name so we cannot be more specific than saying resource. What you can do is help mypy to doing the following:
```python
from typing import cast

import pyvisa

rm = pyvisa.ResourceManager()
osc =cast( yvisa.resources.MessageBasedResource, rm.open_resource("TCPIP::192.168.99.104::INSTR"))
print(osc.query("*IDN?"))
```

Matjaz-B, 2022-12-15:
Yes, I figured this was the case (something like that).
Thanks for solution proposal.

What are you doing in such cases?
Are you casting, or just not running static checks?

Matjaz-B, 2023-01-20:
I see that I've forgotten to close this item.

Solution exists. Its not pretty but it does the job.

Thanks

closed 2023-01-20 by Matjaz-B