pyvisa/pyvisa #718: [COM] Communication issue with ARSL instruments using NI-VISA, configuration settings

dphaas, 2022-12-20:
NI-VISA comes with configuration utilities (MAX on Windows, visaconf on UNIX) for instrument communications settings.  These setting will not be used in pyVISA unless access_mode=constants.VI_LOAD_CONFIG is passed to open_resource().

`instr = rm.open_resource('ASRL18::INSTR', access_mode=constants.VI_LOAD_CONFIG)`

This NI-VISA behavior is consistent with the VISA library spec VPP-4.3 rev. 7.2. 
Keysight and R&S VISA pass the configured settings even if access_mode!=constants.VI_LOAD_CONFIG.

**This issue is for informational purposes only and can be closed.**

NI-VISA info:
C:\WINDOWS\system32\nivisa64.dll
        Comments: 2022/07/01 17:45:56, nivisa64/win64U/x64/msvc-14.0/release    
        InternalName: NIVISA_CORE 22.5.0f60
        ProductName: NIVISA_CORE
        CompanyName: National Instruments Corporation
        LegalCopyright: Copyright © 2000-2022 National Instruments Corporation. 
All Rights Reserved.
        ProductVersion: 22.5.0f60
        FileDescription: NI-VISA Library
        LegalTrademarks: None
        PrivateBuild: None
        FileVersion: 22.5.0f60
        OriginalFilename: nivisa64.dll
        SpecialBuild: NI-Spy

MatthieuDartiailh, 2022-12-20:
Could you make a PR against the documentation ? I believe it would be a valuable addition.

dphaas, 2022-12-21:
I'll give it a shot.

dphaas, 2023-01-11:
@MatthieuDartiailh are you happy with the pull request I created?

MatthieuDartiailh, 2023-01-12:
Sorry I went through it quickly once but did not comment. I will make a thorough review by the end of next week.

dphaas, 2023-01-12:
OK, I would be glad to change anything.  I wasn't sure about the best way to fit this into existing documentation.