pyvisa/pyvisa #720: Setting timeout = None still throws "VI_ERROR_TMO (-1073807339)" after 2 minutes.

C-Ingo, 2023-01-03:
**This is my first time posting about an issue, so if this is the wrong place to do so, please let me know and I will remove the post.**

When trying to read a measurement on a Keithley 2450 SourceMeter, my code timed out with the error ```pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.```.
As a workaround I tried to set ```my_instrument.timeout = None```, so the host PC would wait indefinitely for the measurement to finish and then receive the data.
This works perfectly for shorter measurements, but once the measurement time exceeds 2 minutes the same timeout error appears.
I seemed to be able to manually set the timeout longer than 2 minutes (e.g. ```my_instrument.timeout = 300000```), but this seems like an improper solution, when ```my_instrument.timeout = None``` should wait indefinitely anyway.

To Reproduce
--------------
1. Send a command that takes a long time to execute to the device.
    ```my_instrument.write(r":SOUR:SWE:VOLT:LIN 0, 10, 1001, 0, 1, AUTO, ON, OFF, "defbuffer1"")```
2. Set the VISA timeout to infinite
    ```my_instrument.timeout = None```
3. Send a read command to the device.
    ```my_instrument.write(r":TRACE:DATA? 1, 1001, "defbuffer1", SOUR, READ")```
4. Wait 2 minutes.
5. See error ```pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.```

**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Windows-10-10.0.19045-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\shark\.virtualenvs\LabEquipmentPy3-HCZK_If1\Scripts\python.exe
   Version:        3.11.0
   Compiler:       MSC v.1933 64 bit (AMD64)
   Bits:           64bit
   Build:          Oct 24 2022 18:26:48 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1377166368
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1377166368
         Spec. Version: 5244928
   py:
      Version: 0.6.0
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      VICP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

MatthieuDartiailh, 2023-01-03:
This is weird. Can you read back the timeout value (simply access the attribute) ?

C-Ingo, 2023-01-03:
Yes, calling ```print(my_instrument.timeout)``` before and after setting ```timeout = None``` respectively returns ```2000``` and ```inf```.

MatthieuDartiailh, 2023-01-03:
What kind of connection are you using (GPIB, USB, TCPIP) ?

C-Ingo, 2023-01-03:
I've tested on both GPIB and TCPIP and the issue seems to exist on both.

MatthieuDartiailh, 2023-01-03:
Interesting. Are you using the IVI backend (`rm = ResourceManager()`) or PyVISA-py (`rm = ResourceManager("@py")`) ?

In the first case I am not sure there is anything we can do, since it means Keysight VISA is somehow not honoring the setting, in the second it is a pyvisa-py bug that we can try to address.

C-Ingo, 2023-01-03:
I am indeed using the IVI backend, so I guess there's not much I can do except use GPIB 😅 .

MatthieuDartiailh, 2023-01-03:
You can try to reach out to Keysight, figure out a decent timeout or try to use OPC?  or use a service request to know when your acquisition completes.

C-Ingo, 2023-01-03:
Actually, I tried using the "*WAI" statement, the "OPC?" command and the SRQ system for the specific problem I was having at first, but all three of these options failed, which is what led me to changing the timeout values. I might try reaching out to Keysight for their input though. Thanks for your help anyway!

dphaas, 2023-01-03:
I was going to suggest you use pyvisa-shell for debug but after trying it myself that may not be productive.  I am using NI-VISA. 

- When I use pyvisa-shell to set timeout on a **LXI** instrument using value 4294967295 (no timeout) or 4294967294 (max timeout) I get: "VI_ERROR_NSUP_ATTR_STATE (-1073807330): The specified state of the attribute is not valid, or is not supported as defined by the object."
- When I use pyvisa-shell to set timeout on a **GPIB** instrument using value 4294967295 (no timeout) I get: "timeout value is invalid"
- When I use pyvisa-shell to set timeout on a **GPIB** instrument using value 4294967294 (max timeout) I get: "Done".  Then readback timeout value = "Timeout: 1000000ms"

I suspect NI-VISA is doing some substitutions here and timeout error determination.  The pyVisa comments seem to indicate Keysight VISA does some substituting on timeout values.

Probably the best way to isolate the behavior between pyvisa and the VISA driver is to use Keysight instrument communications tools that come with the IO Libraries.

The VISA library spec v7.2 indicates INSTR resources VI_ATTR_TMO_VALUE shall support range [VI_TMO_IMMEDIATE, 1 to FFFFFFFEh, VI_TMO_INFINITE]