pyvisa/pyvisa #721: [COM] Communication issue with LeCroy oscilloscope using read_raw

SengerM, 2023-01-05:
Instrument details
------------------
* Model: LeCroy HDO6034-MS
* Communication: LXI and USB

Output of `pyvisa-info`
-----------------------

```
/home/me/.local/lib/python3.8/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:
   Platform ID:    Linux-5.4.0-126-generic-x86_64-with-glibc2.29
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.8.10
   Compiler:       GCC 9.4.0
   Bits:           64bit
   Build:          Nov 14 2022 12:59:47 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.22.5.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available
```

# Description

I have been using Pyvisa to get data from LeCroy oscilloscopes in the last years, no problems. Recently I got a 12 bit vertical resolution oscilloscope and cannot get the data with the 12 bit vertical resolution, it always comes in 8 bits no matter what. I don't know exactly where is the problem, if in the oscilloscope or in the PC, but just in case I am asking here as I already went through all the settings I could find in the oscilloscope and everything looks good. This is my code:

```Python
import pyvisa
import time
import plotly.express as px

o = pyvisa.ResourceManager().open_resource('TCPIP0::xxx.xxx.xxx.xxx::inst0::INSTR')

print(o.query('*IDN?'))

# Method 1 ---
# This works, but gets 8 bit data.

o.write(f'C2:WF?')
time.sleep(.1)
raw_data = o.read_raw()
raw_data = list(raw_data)

print(raw_data)

fig = px.line(
	raw_data,
	markers = True,
)
fig.write_html('raw.html')

# Method 2 ---

data = o.query("VBS? 'return=app.Acquisition.C2.Out.Result.DataArray'")
print(data) # This prints 'Cannot convert Variant to String'
```

MatthieuDartiailh, 2023-01-05:
Could you share the programming manual ? I could only find the user manual on Lecroy website.

SengerM, 2023-01-05:
Sorry for that. I usually look for those manuals [here](https://teledynelecroy.com/oscilloscope/waverunner-9000-oscilloscopes/resources/product-manuals) as they all share the same interface. With the manuals there I was able to properly control 4 different oscilloscopes:

1. LeCroy WaveRunner 640Zi (8 bit).
2. LeCroy WaveRunner 9254M (8 bit).
3. LeCroy WaveRunner HDO6034-MS (12 bit).
4. LeCroy WaveRunner MDA8208HD (12 bit).

In all cases I use exactly the same code. In all cases the data with `read_raw` comes as expected BUT in 8 bit and the "method 2" fails as described.

MatthieuDartiailh, 2023-01-06:
From what can be read on page 7-155 I believe you need to use to COMM_FORMAT (7-44) specify you want to transfer data as WORD instead of BYTE.

SengerM, 2023-01-06:
Yes, it was that, thanks for pointing it out. In case someone ever ends up here with a similar issue, the following code is getting the waveforms with the correct vertical resolution:

```Python
import pyvisa
import time
import plotly.express as px

o = pyvisa.ResourceManager().open_resource('TCPIP0::130.60.165.216::inst0::INSTR')

print(o.query('*IDN?'))

o.write('CORD HI')
o.write('COMM_FORMAT DEF9,WORD,BIN')
o.write('CHDR OFF')

o.write(f'C2:WF?')
time.sleep(.1)
raw_data = o.read_raw()

grouped_raw_data = [[raw_data[j] for j in [i,i-1]] for i in range(0,len(raw_data),2)]
samples = [int.from_bytes(group_of_raw, byteorder='little') for group_of_raw in grouped_raw_data[181:]] # First 180 pairs of bytes are a header.

fig = px.line(
	samples,
	markers = True,
)
fig.write_html('raw.html')
```

closed 2023-01-06 by SengerM

MatthieuDartiailh, 2023-01-06:
Happy I could help