pyvisa/pyvisa #722: [COM] Communication issue with JGR optical switch using TCPIP pyvisa

aamirlore, 2023-01-10:
Instrument details
------------------
* Model: Optical Switch JGR
* Communication: TCPIP, GPIB, ....
* Link to the documentation (if available): https://www.manualslib.com/manual/2120333/Jgr-Sx1.html

Output of `pyvisa-info`
-----------------------

aamirlore, 2023-01-10:
I am currently working with an optical switch instrument by JGR optics and I am facing the same issue.
There's an SCPI command to query the information of the module MODule#:INFO?  which return a '1Ax3,SM,9\xb5m\n' and here we have bytearray value in the end which is \xb5m\n. In my python program I am using cmd = self.inst.query_ascii_values(f'MODule{module}:INFO?') which throws a UniCodeDecodeError [undefined]UnicodeDecodeError:'ascii' codec can't decode byte 0xb5 in position 9: ordinal not in range(128)
and when I use cmd = self.inst.binary_ascii_values(f'MODule{module}:INFO?') it throws a valueError cmd = self.inst.query_ascii_values(f'MODule{module}:INFO?')
Could you please help me to understand what exactly has to be done to get or convert it into a normal string? I am beginner using pyvisa implementation.

aamirlore, 2023-01-10:
I am currently working with an optical switch instrument by JGR optics and I am facing the same issue.
There's an SCPI command to query the information of the module `MODule#:INFO?`  which return a `'1Ax3,SM,9\xb5m\n'` and here we have bytearray value in the end which is `\xb5m\n.` In my python program I am using ```cmd = self.inst.query_ascii_values(f'MODule{module}:INFO?')``` which throws a UniCodeDecodeError ```[undefined]UnicodeDecodeError:'ascii' codec can't decode byte 0xb5 in position 9: ordinal not in range(128)```
and when I use ```cmd = self.inst.binary_ascii_values(f'MODule{module}:INFO?')``` it throws a valueError ```[undefined]ValueError: Could not find hash sign (#) indicating the start of the block. The block begin by bytearray(b'1Ax3,SM,9\xb5m\n')```
Could you please help me to understand what exactly has to be done to get or convert it into a normal string? I am beginner using pyvisa implementation.

MatthieuDartiailh, 2023-01-10:
The easiest will be to either use (after a write):
- read_bytes with large number and requesting to stop when encountering a term char. Then you can deal with the bytes yourself
- read and specify utf8 as encoding 

By the way the documentation you linked does not mention the `Module#:INFO?` command.

aamirlore, 2023-01-10:
Oh yes I just realised the provided link doesn't have this particular command.
Here is the manual which has this command
[M-SX1-001.03-SX1-User-Manual.pdf](https://github.com/pyvisa/pyvisa/files/10384404/M-SX1-001.03-SX1-User-Manual.pdf)

aamirlore, 2023-01-10:
Thanks a lot for your help but I am quite confused with your explanation. @MatthieuDartiailh 
I didn't really get your first line which says `The easiest will be to either use (after a write):`
Could you please elaborate it for me :)

MatthieuDartiailh, 2023-01-11:
Playing a bit on the Python repr it turns out that `"\xb5m"` is actually "µm" but the bytes need to be decoded using the latin1 codec. So you can do the following:
```python
self.instr.read_termination = "\n"  # This way PyVISA trim the termination character for you
self.instr.write(f"MODule{module}:INFO?")
cmd = self.instr.read(encoding="latin1")
```
Hope this helps. You could also specify the encoding to use directly on `self.instr` in which case it will always be used.

aamirlore, 2023-01-11:
Thanks a lot @MatthieuDartiailh for your help. Now I am able to query it in the form of a string using the latin1 encoding.

MatthieuDartiailh, 2023-01-11:
Happy to help.

closed 2023-01-11 by MatthieuDartiailh