pyvisa/pyvisa #728: Read_raw stops at newline while read_termination is None

bmoneke, 2023-02-27:
`read_raw` stops reading at a newline character, even though `read_termination` had been set to None, see https://github.com/pymeasure/pymeasure/issues/863 (there is also a code example).

It happens with pyvisa-py and serial communication.

What is the expected behaviour of `read_raw` if no `read_termination` is defined?

Is the `termchar` ResourceAttribute in the `read_termination` attribute (`messagebased.py`) the culprit?
~~~python
            # The termchar is also used in VI_ATTR_ASRL_END_IN (for serial
            # termination) so return it to its default.
            self.set_visa_attribute(constants.ResourceAttribute.termchar, ord(self.LF))
            self.set_visa_attribute(
                constants.ResourceAttribute.termchar_enabled, constants.VI_FALSE
            )
~~~

Or because pyvisa-py ignores the `termchar_enabled` value?

bmoneke, 2023-02-27:
I just tested the behaviour with the NI backend: `read_raw` stops at the first line feed `\n`, even if `read_termination` is None.

**What is the expected behaviour of `read_raw`?**
- The docstring states "Read the unmodified string sent from the instrument to the computer. In contrast to read(), no termination characters are stripped.". That would suggest to read until a termination character is reached (like read).
- However, it is not transparent, that line feed is the default termination character.

I expected to read the whole buffer / until somehow termination is indicated (termination lines, timeout etc.), if `read_termiation is None`.

MatthieuDartiailh, 2023-02-27:
The IVI spec (page 188) does discuss the oddities of serial resources regarding viRead. I am not saying that PyVISA is doing it right but Serial resources are always special. We should at least document things better but I am not sure how persavice changes we can make since people may rely on the current behavior.

bmoneke, 2023-02-28:
It is a special case: `read_termination = None` and `read_raw`.
However, we cannot know, whether someone expects this behaviour, therefore, I agree that the behaviour should not change.

Obviously it should be documented ("If `read_termination` is None, it breaks at a linefeed character").
An option could be added to continue reading / not breaking at a term char.

MatthieuDartiailh, 2023-02-28:
It breaks at a line feed only for serial resource whose configuration have not been customized. In particular if a resource declares it uses DTR line for termination the behavior may be different.

RobertoRoos, 2023-02-28:
> It might be useful to snapshot VISA attributes when using the IVI backend (NI, R&S or Keysight) and see if there is any discrepancy between those and pyvisa-py ones. Then update pyvisa-py

I get the same result with the NI IVI backend (premature break on a line-feed when no termchar is supplied).

MatthieuDartiailh, 2023-02-28:
Out of curiosity could you check the behavior you get in NI-MAX ? I am curious to see if at least PyVISA default behavior match MAX default behavior.