pyvisa/pyvisa #732: [COM] Communication issue with Agilent 34410A/Keysight 34461A using TCPIP: slow reading resulting in timeout

JerPatton, 2023-03-24:
Hello all,
I am trying to make multiple frequency measurements  from an Agilent 34410A multimeter, in order to finally get an average value. 
However, when I set the instrument to make multiple measurements, my readings turn out to be really slow (around 1 value/sec, confirmed by the blinking star in the top left corner of the front panel). And thus, any subsequent query almost always timeout with the error:  `VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed `

<details>
<summary>  View full error stack trace </summary>

```
VisaIOError                               Traceback (most recent call last)
Input In [16], in <cell line: 4>()
      2 # dmm.write("INIT;*OPC")
      3 sleep(0.1)
----> 4 dmm.query('*OPC?')
      5 sleep(0.1)
      6 # freq_value = dmm.query('*OPC?;FETCh?')

File D:\MCm\Git\Notebooks\venv_notebook\lib\site-packages\pyvisa\resources\messagebased.py:650, in MessageBasedResource.query(self, message, delay)
    647 if delay > 0.0:
    648     time.sleep(delay)
--> 650 return self.read()

File D:\MCm\Git\Notebooks\venv_notebook\lib\site-packages\pyvisa\resources\messagebased.py:488, in MessageBasedResource.read(self, termination, encoding)
    486 if termination is None:
    487     termination = self._read_termination
--> 488     message = self._read_raw().decode(enco)
    489 else:
    490     with self.read_termination_context(termination):

File D:\MCm\Git\Notebooks\venv_notebook\lib\site-packages\pyvisa\resources\messagebased.py:444, in MessageBasedResource._read_raw(self, size)
    437     while status == loop_status:
    438         logger.debug(
    439             "%s - reading %d bytes (last status %r)",
    440             self._resource_name,
    441             size,
    442             status,
    443         )
--> 444         chunk, status = self.visalib.read(self.session, size)
    445         ret.extend(chunk)
    446 except errors.VisaIOError as e:

File D:\MCm\Git\Notebooks\venv_notebook\lib\site-packages\pyvisa\ctwrapper\functions.py:2337, in read(library, session, count)
   2335 buffer = create_string_buffer(count)
   2336 return_count = ViUInt32()
-> 2337 ret = library.viRead(session, buffer, count, byref(return_count))
   2338 return buffer.raw[: return_count.value], ret

File D:\MCm\Git\Notebooks\venv_notebook\lib\site-packages\pyvisa\ctwrapper\highlevel.py:226, in IVIVisaLibrary._return_handler(self, ret_value, func, arguments)
    223         # Set session back to a safe value
    224         session = None
--> 226 return self.handle_return_value(session, ret_value)

File D:\MCm\Git\Notebooks\venv_notebook\lib\site-packages\pyvisa\highlevel.py:251, in VisaLibraryBase.handle_return_value(self, session, status_code)
    248     self._last_status_in_session[session] = rv
    250 if rv < 0:
--> 251     raise errors.VisaIOError(rv)
    253 if rv in self.issue_warning_on:
    254     if session and rv not in self._ignore_warning_in_session[session]:

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed 
```
</details>

This is really strange, as when I return the instrument to _Local_ the measurement rate is immediately faster (around 5 values / sec)...

I tried to add a `query("*OPC?")` and/or some `time.sleep(0.1)` before my query commands and also tried to "tweak" the settings to make faster measurements according to the user manual (see code example below), yet still without any success so far. I am not even sure the `query("*OPC?")` works at all (?).

The only workaround I found is to wait for the full measurement duration by adding a `time.sleep(amount)` between the `"INIT" `command and the subsequent `"FETCh?" `query, where amout equals the number of desired samples in seconds...
But it is really bothering in practice, if not impossible, to wait for this full duration, as it is part of a final sensor calibration test. And if for any reason the measurement takes longer then a timeout would occur, "crashing" the test.

If you have any explanation for this or can point me on any direction to solve this, I'd greatly appreciate. I admit I am quite stuck at this point.

Instrument details
------------------
<details>
<summary>  View details </summary>

* Model: Agilent 34410A and also the more recent Keysight 34461A
* Communication: TCPIP (INSTR)
* Link to the documentation (if available): 
    - User Manual: https://www.keysight.com/us/en/assets/9018-05586/user-manuals/9018-05586.pdf
    - Programmer's reference: https://www.keysight.com/us/en/assets/9018-83366/help-files/9018-83366.zip
</details>

Output of `pyvisa-info`
-----------------------
<details>
<summary>  View details </summary>

Machine Details:
   Platform ID:    Windows-10-10.0.19044-SP0
   Processor:      Intel64 Family 6 Model 140 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     D:\MCm\Git\Notebooks\venv_notebook\Scripts\python.exe
   Version:        3.10.5
   Compiler:       MSC v.1929 64 bit (AMD64)
   Bits:           64bit
   Build:          Jun  6 2022 16:14:13 (#tags/v3.10.5:f377153)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
   py:
      Version: 0.5.4.dev10+g76bda99
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
</details>

Minimal working example to reproduce my issue
------------------
```
import pyvisa
from time import sleep

rm = pyvisa.ResourceManager()
dmm = rm.open_resource("TCPIP0::RD011::inst0::INSTR")
dmm.timeout = 7000

dmm.write("CONFigure:FREQ")
nb_samples = 10
dmm.write(f"SAMPle:COUNt {nb_samples}")

""" 
Optional settings below, uncomment below to observe difference.
Input signal is square (DCyc: 50%) at 250 kHz, 3.3 VPP (no offset)
"""
# dmm.write("SENSe:VOLT:AC:BAND MAX")
# dmm.write("SENSe:VOLTage:AC:RANGe 5V")
# dmm.write("SENSe:FREQuency:APERture .001")

# Trigger measurements
dmm.write("INIT")    # Measurements are started and stored in instrument
# sleep(nb_samples)    # Works with this, fails without
dmm.query("*OPC?")    # This does not change anything, without the sleep it still fails
sleep(0.1)
freq_value = dmm.query("FETCh?")
print(freq_value)
```

JerPatton, 2023-03-27:
Thanks for your quick reply.

So, I dug a little bit further into my settings according to your input. 
Here are my findings regarding your comments:

- **Autoranging**: it was already disabled when I used the command: `write(SENSe:VOLTage:AC:RANGe 5V")`. Yet as observed ealier this has unfortunately no visible effect on the measurement speed.
- **Instrument statistics**: In fact I initially used the instrument built-in averaging tool, but observed the same slowness. Then for the sake of simplicity, I went with the example of sample collection only ;)
- **`*OPC?`**: Actually, I was incorrect above and you are right, it waits for the previous commands to end before running the subsequent commands. For the built in averaging the `dmm.query("*OPC?")` command is furthermore necessary to wait for the collection of all points. Otherwise, the average value is computed on an empty or truncated list of points.
- **timeout**: Oh, my bad it was unclear maybe. I initially set it to 7000ms in my code to be sure to cope with a 5 samples measurement duration which takes about 5 sec. The latter being the maximum acceptable recording duration in my situation. Then, for the example here, I increased to 10 samples to show how surprisingly long it is no matter the settings, which in turn causes a timeout. So in practice I don't really need nor want to increase it further...

What I am still confused about and would like to better understand, is why the measuring speed in remote v.s. local is much much slower (around 5 times). Could this be a pyvisa issue or is it in fact more of a hardware issue?

MatthieuDartiailh, 2023-03-27:
Since you are measuring frequency you need to disable frequency auto ranging `SENSe:FREQ:VOLT:RANG:AUTO`, does this make any difference ?

JerPatton, 2023-03-29:
Argh thanks, this one I indeed missed. I misinterpreted the doc and mixed it with the `SENSe:VOLTage:AC:RANGe range` setting. Unfortunately, this has still very little to no effect whether it's turned ON or OFF in my case...

However, I think I found the culprit, which is in fact **not a Pyvisa problem but rather a hardware setting, my bad**.... 
To make a long story short, it is because of the 1 default second _Trigger delay_ for frequency measurements imposed by the instrument.

It was confirmed in my case by the result of the command `"TRIGger:DELay?"` which indeed returns "+1.00002613E+00" (seconds).
So setting it to MIN (i.e. 0 sec for this case) or any other small value solves the problem, even when letting the aperture to its default value.
Then one just have to account for the potential "settling time" of the signal (if any), but this is another story ;)

<Details>
<Summary> Quick Agilent 34410A sample acquisition timing info for the curious below</Summary>

According to the doc, in  `SAMPle:SOURce IMMediate` mode (default), the instrument **waits for this _trigger delay_ duration before each sample**, also in the case of multiple measurements.
In `SAMPle:SOURce TIMer` mode, then the insturment **just waits once for the _trigger delay_ duration and then takes samples every _timer_ seconds**.
Both the _trigger delay_ and the _sample timer_ can be set with their respective commands `TRIGger:DELay delay` and `SAMPle:TIMer timer`.
The default _trigger delay_ also depends on the function, range, integration time, and ac filter setting and vary accordingly to ensure accurate measurements. If you set it manually, ensure you know what you are doing and how your signals are.
</Details>

So, thanks again for your time  and quick reply every time. And sorry for actually bothering you with a non pyvisa related issue...

closed 2023-03-29 by JerPatton