pyvisa/pyvisa #736: OSError: exception: access violation reading 0x000002DECD70976F

jihengzhang, 2023-04-03:
This is a new issue I encountered recently. 
In the past it works about 1year. But for unknow reason, this issue happens.
I tried to reinstall pyvisa, and python. but it can not repair this issue.
This issue does not always happen. At most of time it happens after 10 test was implemented.
So I close the vscode, then restart, then it can run for 10+ cycles. Then close the vscode to repeat.

Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\212597558\AppData\Local\Programs\Python\Python39\lib\threading.py", line 954, in _bootstrap_inner
    self.run()
  File "c:\Users\212597558\Repos\MNS_test_N9030A\code\MNS_scan.py", line 147, in run
    f = self.mnsTest.scan(series)
  File "c:\Users\212597558\Repos\MNS_test_N9030A\code\MNS_scan.py", line 102, in scan
    trace_data = self.dsa.get_trace_xy(spectrum)
  File "c:\Users\212597558\Repos\MNS_test_N9030A\code\dsa800.py", line 543, in get_trace_xy
    trace = self.resource.query("trace:data? trace1")
  File "C:\Users\212597558\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\212597558\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\212597558\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\212597558\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
OSError: exception: access violation reading 0x000002DECD70976F

PS C:\Users\212597558> python -m visa info
C:\Users\212597558\AppData\Local\Programs\Python\Python39\lib\site-packages\visa.py:13: FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.

The reason for the deprecation is the possible conflict with the visa package provided by the https://github.com/visa-sdk/visa-python which can result in hard to debug situations.
  warnings.warn(
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 13, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\212597558\AppData\Local\Programs\Python\Python39\python.exe
   Version:        3.9.2
   Compiler:       MSC v.1928 64 bit (AMD64)
   Bits:           64bit
   Build:          Feb 19 2021 13:44:55 (#tags/v3.9.2:1a79785)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: Keysight Technologies
         Spec. Version: Keysight Technologies
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: Keysight Technologies
         Spec. Version: Keysight Technologies

arr-ee, 2023-04-03:
my speculation:

It looks like the error occurs while calling external VISA implementation (keysight's in this case). Have you updated it recently by any chance?

> `File "C:\Users\212597558\AppData\Local\Programs\Python\Python39\lib\threading.py", line 954, in _bootstrap_inner
self.run()`

in the beginning of your trace suggests you might be running your pyvisa code in a thread — are you calling pyvisa from multiple threads by any chance? I suspect this _might_ lead to memory access issues.

If you could re-run your program using `pyvisa-py` instead of Keysight VISA (see https://pyvisa.readthedocs.io/en/latest/introduction/configuring.html), it might help triaging the source of issues.

jihengzhang, 2023-04-03:
> WARNING: This email originated from outside of GE. Please validate the sender's email address before clicking on links or attachments as they may not be safe.

Hi，
Thanks a lot for your quick response.
A thread is used in the design. Here is how it used:
***@***.***

Here is how the thread called:
***@***.***

Is there any potential issue caused by the usage? There is only one place to use the threading. I did not make change to threading.py.

If the thread method has no problem, I will try using pyvisa-py.

Best Regards
Michael (Jiheng). Zhang 张计恒

Architect, Display
COE Hardware , GE Healthcare

T +86 181 0104 3456 | +86 10 5708 3306
BTP 2A-W13  |  NO. 1 Tongji Nanlu, Yizhuang, Beijing, China
Post code: 100176
北京市大兴区亦庄开发区同济南路1号通用电气

MatthieuDartiailh, 2023-04-03:
Since I do not know what transport layer (TCPIP, USB, etc) is used I would be careful suggesting to use pyvisa-py. It should work for TCPIP but using it for USB (in particular on windows) is not that obvious. 

Getting a minimal reproducer would help in debugging the issue obviously.