pyvisa/pyvisa #741: read_raw() broken

spukhafte-sys, 2023-04-14:
I recently updated from:
pyserial==3.4
pyusb==1.0.2
PyVISA==1.8
PyVISA-py==0.2

to:
pyserial==3.5
pyusb==1.0.2
PyVISA==1.13.0
PyVISA-py==0.6.1

After the update a program stopped detecting an instrument that uses 26 byte frames (ie not SCPI, no termination char). The problem is the ASRL read_raw() behaviour has changed. In PyVISA v1.8, read_raw() swept up whatever it could from the serial read buffer without exception. Now it throws an exception, no matter how I set read_termination, end_input, etc. The program below produces this output when run on a system with an FT232 (USB serial adapter) in loopback:
```
$ ./loopback.py 
read_bytes=b'\xaa\x03\xfe5\x00\xe0'
default read_raw() failed
rt=None ei=False read_raw() failed
rt=None ei=False read_bytes=b''
rt=None ei=True read_raw() failed
rt=None ei=True read_bytes=b''
rt= ei=False read_raw() failed
rt= ei=False read_bytes=b''
rt= ei=True read_raw() failed
rt= ei=True read_bytes=b''
read_bytes()=b'\xaa\x03\xfe5\x00\xe0'
```
I can get read_bytes() to work by looping until I get VI_ERROR_TMO, but would much prefer the old behaviour of read_raw(), as the documentation still describes. 

**loopback.py**
```
#!/usr/bin/env python3

# Test write_raw and read_raw on a USB loopback

#   python3-pyvisa
#   python3-pyvisa-py

import pyvisa

TEST = b'\xaa\x03\xfe5\x00\xe0'  # Read EEPROM location 0

rm = pyvisa.ResourceManager('@py')  # @py selects the pyVisa-py backend

# Discover what's where
for resource in rm.list_resources():
    if resource.split('::')[0].startswith('ASRL'):  # Serial device?
        i = rm.open_resource(resource, {'baud_rate': 9600})
        i.write_termination = ''
        i.write_raw(TEST)
        print(f'read_bytes={i.read_bytes(len(TEST))}')  # Works

        try:  # Test default settings
            print(f'read_raw={i.read_raw()}')
        except pyvisa.errors.VisaIOError:
            print('default read_raw() failed')

        for rt,ei in ((None, False), (None, True), ('', False), ('', True)):
            i.read_termination = rt
            i.end_input = ei

            i.write_raw(TEST)

            print(f'rt={rt} ei={ei} ', end='')
            try:
                print(f'read_raw()=={i.read_raw()}')
            except pyvisa.errors.VisaIOError:
                print('read_raw() failed')

            print(f'rt={rt} ei={ei} ', end='')
            try:
                print(f'read_bytes={i.read_bytes(-1)}')
            except pyvisa.errors.VisaIOError:
                print('read_bytes() failed')

        i.write_raw(TEST)
        print(f'read_bytes()={i.read_bytes(len(TEST))}')  # Works

        i.close()
```
As you can see I also tried the undocumented read_bytes(-1), which I infer was intended to work the same as the old read_raw()?
```
$ pyvisa-info
Machine Details:
   Platform ID:    Linux-5.10.0-16-armmp-armv7l-with-glibc2.31
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.9.2
   Compiler:       GCC 10.2.1 20210110
   Bits:           32bit
   Build:          Feb 28 2021 17:03:44 (#default)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.6.1
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      VICP INSTR:
         Please install PyVICP to use this resource type.
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

spukhafte-sys, 2023-04-15:
I dug into this myself and suspect the problem is in pyvisa_py/sessions.py. I get the behaviour I expect from read_raw() with the changes to pyvisa-py/sessions.py documented in [this issue](https://github.com/pyvisa/pyvisa-py/issues/366).

arr-ee, 2023-04-15:
Thank you for digging further!

Out of curiosity, what is your reason to use `pyvisa` for non-VISA/non-GPIB-ish instruments? I have a bunch (including BK 8500 that might have been the instrument at issue here), and I was idly debating with myself if trying to force pyvisa to talk to them is worth it. Would love to hear opinions from others already doing it.

Thanks.

spukhafte-sys, 2023-04-15:
> Out of curiosity, what is your reason to use `pyvisa` for non-VISA/non-GPIB-ish instruments? I have a bunch (including BK 8500 that might have been the instrument at issue here)

Good guess. It's a B&K 8542B.

We've a motley collection of gear that support SCPI, except two that use "frame" protocols. Many instruments have a USB serial interface, which enumerate differently every time Linux reboots. PyVISA does a great job of discovering all the local and network resources. To detect which instrument is associated with what resource, we use write_raw() and read_raw(). If there's a better way of discovering and identifying instruments, do share!

MatthieuDartiailh, 2023-04-17:
VISA always expects to have some sort of signal that a message is complete and does not handle well fixed size messages. 

That being said you could use `read_bytes` to specify exactly how many bytes you expect. It will be more cumbersome but also safer in the long term (random bugs arising from getting a partial message are no fun).

spukhafte-sys, 2023-04-17:
> That being said you could use `read_bytes` to specify exactly how many bytes you expect. It will be more cumbersome but also safer in the long term (random bugs arising from getting a partial message are no fun).

Both read_bytes() and read_raw() have their use-cases. The read_bytes() method works fine if you know how many bytes to expect, but when discovering instruments there's no way to know the state of the instrument and its data pipeline. For example, the pipeline could contain a truncated response because the instrument was forceably reset. 

Speaking of read_bytes(), could someone explain what read_bytes(-1) is supposed to do?

spukhafte-sys, 2023-04-17:
I discovered errors in my loopback.py test. Here's an update:
```
#!/usr/bin/env python3

# Test write_raw() and read_raw() on a USB loopback

#   python3-pyvisa
#   python3-pyvisa-py

import pyvisa

TEST = b'\xaa\x03\xfe5\x00\xe0'  # NCD Read EEPROM location 0

rm = pyvisa.ResourceManager('@py')  # @py selects the pyVisa-py backend

# Discover what's where
for resource in rm.list_resources():
    if resource.split('::')[0].startswith('ASRL'):  # Serial device?
        i = rm.open_resource(resource,
            {'baud_rate': 9600, 'timeout': 1000, 'write_termination': ''})

        # Test default settings
        i.write_raw(TEST)
        print(f'read_bytes(LEN)=={i.read_bytes(len(TEST))}')  # Works

        i.write_raw(TEST)
        try:  # Test default settings
            print(f'read_raw=={i.read_raw()}')
        except pyvisa.errors.VisaIOError:
            print('default read_raw() failed')

        for rt,ei in ((None, False), (None, True), ('', False), ('', True)):
            i.read_termination = rt
            i.end_input = ei

            print(f'rt={rt} ei={ei} ', end='')

            i.write_raw(TEST)
            try:
                print(f'read_raw()=={i.read_raw()}')
            except pyvisa.errors.VisaIOError:
                print('read_raw() failed')

            i.write_raw(TEST)
            print(f'rt={rt} ei={ei} ', end='')
            try:
                print(f'read_bytes=={i.read_bytes(-1)}')
            except pyvisa.errors.VisaIOError:
                print('read_bytes() failed')

        i.write_raw(TEST)
        print(f'read_bytes(LEN)=={i.read_bytes(len(TEST))}')  # Works

        i.close()```

MatthieuDartiailh, 2023-04-17:
Spending some more time thinking about it (Serial resources are weird), I have some extra questions and suggestions:
- could you test on a machine on which you could install NI, Keysight or R&S VISA ? debugging possible pyvisa-py issues separately would be nicer
- as I mentioned serial resources are weird and you can disable any kind of termination check by setting end_input to `SerialTermination.none` (in addition of setting read_termination to None). In your tests you use True and False which are not valid values so there is something fishy there. Could you maybe post the output of logging (you can use pyvisa.log_to_screen to set up a logger handler)

As for read_bytes(-1) it looks like it won't do anything (same as read_bytes(0)), why do you ask ?

spukhafte-sys, 2023-04-17:
> * could you test on a machine on which you could install NI, Keysight or R&S VISA ? debugging possible pyvisa-py issues separately would be nicer

I'll investigate which of the above are available for ARM Linux.

spukhafte-sys, 2023-04-17:
> As for read_bytes(-1) it looks like it won't do anything (same as read_bytes(0)), why do you ask ?

I found and was puzzled by some code that used read_bytes(-1). You're right, it works exactly like read_bytes(0).

spukhafte-sys, 2023-04-17:
> * as I mentioned serial resources are weird and you can disable any kind of termination check by setting end_input to `SerialTermination.none` (in addition of setting read_termination to None). In [...]

My test program sets .end_input correctly now:
```
#!/usr/bin/env python3

# Test write_raw() and read_raw() on a USB loopback

#   python3-pyvisa
#   python3-pyvisa-py

import pyvisa

TEST = b'\xaa\x03\xfe5\x00\xe0'  # NCD Read EEPROM location 0

pyvisa.log_to_screen()

rm = pyvisa.ResourceManager('@py')  # @py selects the pyVisa-py backend

# Discover what's where
for resource in rm.list_resources():
    if resource.split('::')[0].startswith('ASRL'):  # Serial device?
        i = rm.open_resource(resource,
            {'baud_rate': 9600, 'timeout': 1000, 'write_termination': ''})

        # Test default settings
        i.write_raw(TEST)
        print(f'read_bytes(LEN)=={i.read_bytes(len(TEST))}')  # Works

        i.write_raw(TEST)
        try:  # Test default settings
            print(f'     read_raw()=={i.read_raw()}')
        except pyvisa.errors.VisaIOError:
            print('default read_raw() failed')

        for ei in (pyvisa.constants.SerialTermination.termination_char, 
                   pyvisa.constants.SerialTermination.none):
            i.end_input = ei

            for rt in (None, ''):
                i.read_termination = rt

                print(f'rt={str(rt):4} ei={ei} ', end='')

                i.write_raw(TEST)
                try:
                    print(f'   read_raw()=={i.read_raw()}')
                except pyvisa.errors.VisaIOError:
                    print('read_raw() failed')

                i.write_raw(TEST)
                print(f'rt={str(rt):4} ei={ei} ', end='')
                try:
                    print(f'read_bytes(0)=={i.read_bytes(0)}')
                except pyvisa.errors.VisaIOError:
                    print('read_bytes() failed')

        i.write_raw(TEST)
        print(f'read_bytes(LEN)=={i.read_bytes(len(TEST))}')  # Works

        i.close()
```
Here's stdout:
```
$ ./loopback.py 2>log.txt
read_bytes(LEN)==b'\xaa\x03\xfe5\x00\xe0'
     read_raw()==b'\xaa\x03\xfe5\x00\xe0'
rt=None ei=2    read_raw()==b'\xaa\x03\xfe5\x00\xe0'
rt=None ei=2 read_bytes(0)==b''
rt=     ei=2    read_raw()==b'\xaa\x03\xfe5\x00\xe0\xaa\x03\xfe5\x00\xe0'
[... 21 lines cut ...]
2023-04-17 15:19:27,503 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 6 bytes (last status None)
2023-04-17 15:19:27,526 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\xe0'
2023-04-17 15:19:27,528 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 15:19:29,545 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\xe0'
2023-04-17 15:19:29,548 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 15:19:31,572 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\xe0'
2023-04-17 15:19:31,577 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\xe0'
2023-04-17 15:19:31,580 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 15:19:33,605 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\xe0'
2023-04-17 15:19:33,609 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\xe0'
2023-04-17 15:19:33,611 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 15:19:35,636 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\xe0'
2023-04-17 15:19:35,640 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\xe0'
2023-04-17 15:19:35,643 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 15:19:37,665 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\xe0'
2023-04-17 15:19:37,670 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\xe0'
2023-04-17 15:19:37,672 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 6 bytes (last status None)
2023-04-17 15:19:37,678 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - closing
2023-04-17 15:19:37,687 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - is closed
2023-04-17 15:19:37,689 - pyvisa - DEBUG - Closing ResourceManager (session: 1989745)
2023-04-17 15:19:37,690 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - closing
```
Note: the last read_bytes(LEN) leaves five bytes in the input buffer.

Why are there two "ASRL/dev/ttyUSB0::INSTR - closing" log messages?

spukhafte-sys, 2023-04-17:
> > * could you test on a machine on which you could install NI, Keysight or R&S VISA ? debugging possible pyvisa-py issues separately would be nicer
> 
> I'll investigate which of the above are available for ARM Linux.

We'd have to overcome our aversion to x86 processors to use any of the above. None appear to support the 32-bit ARM Linux running on our lab SBC.

spukhafte-sys, 2023-04-17:
Added a \n inside the TEST bytes and got more interesting results:
```
$ ./loopback.py 2>log.txt
read_bytes(LEN)==b'\xaa\x03\xfe5\x00\n\xe0'
     read_raw()==b'\xaa\x03\xfe5\x00\n'
rt=None ei=2    read_raw()==b'\xe0\xaa\x03\xfe5\x00\n'
rt=None ei=2 read_bytes(0)==b''
rt=     ei=2    read_raw()==b'\xe0\xaa\x03\xfe5\x00\n'
rt=     ei=2 read_bytes(0)==b''
rt=None ei=0    read_raw()==b'\xe0\xaa\x03\xfe5\x00\n\xe0\xaa\x03\xfe5\x00\n\xe0\xaa\x03\xfe5\x00\n\xe0'
rt=None ei=0 read_bytes(0)==b''
rt=     ei=0    read_raw()==b'\xaa\x03\xfe5\x00\n\xe0\xaa\x03\xfe5\x00\n\xe0'
rt=     ei=0 read_bytes(0)==b''
```
Summary: read_raw() works as expected if end_input is SerialTermination.none and doesn't care whether .read_termination is None or '' (extra bytes for rt=None and ei=0 is what previous read_bytes() left in buffer).

spukhafte-sys, 2023-04-17:
Added a \n inside the TEST variable and to the test values for .read_termination and got more interesting results:
```
./loopback.py 2>log.txt
read_bytes(LEN)==b'\xaa\x03\xfe5\x00\n\xe0'
     read_raw()==b'\xaa\x03\xfe5\x00\n'
rt=None ei=2    read_raw()==b'\xe0\xaa\x03\xfe5\x00\n'
rt=None ei=2 read_bytes(0)==b''
rt=     ei=2    read_raw()==b'\xe0\xaa\x03\xfe5\x00\n'
rt=     ei=2 read_bytes(0)==b''
rt=\n   ei=2    read_raw()==b'\xe0\xaa\x03\xfe5\x00\n'
rt=\n   ei=2 read_bytes(0)==b''
rt=None ei=0    read_raw()==b'\xe0\xaa\x03\xfe5\x00\n\xe0\xaa\x03\xfe5\x00\n\xe0\xaa\x03\xfe5\x00\n\xe0\xaa\x03\xfe5\x00\n\xe0'
rt=None ei=0 read_bytes(0)==b''
rt=     ei=0    read_raw()==b'\xaa\x03\xfe5\x00\n\xe0\xaa\x03\xfe5\x00\n\xe0'
rt=     ei=0 read_bytes(0)==b''
rt=\n   ei=0    read_raw()==b'\xaa\x03\xfe5\x00\n\xe0\xaa\x03\xfe5\x00\n\xe0'
rt=\n   ei=0 read_bytes(0)==b''
read_bytes(LEN)==b'\xaa\x03\xfe5\x00\n\xe0'
     read_raw()==b'\xaa\x03\xfe5\x00\n\xe0'
```
Summary: read_raw() works as expected if end_input == SerialTermination.none and doesn't care whether .read_termination is (extra bytes for rt=None and ei=0 is what previous read_bytes() left in buffer).

Log output:
```
$ cat log.txt
2023-04-17 16:30:07,859 - pyvisa - DEBUG - SerialSession was correctly imported.
2023-04-17 16:30:07,965 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2023-04-17 16:30:08,347 - pyvisa - WARNING - (<InterfaceType.vicp: 36000>, 'INSTR') is already registered in the ResourceManager. Overwriting with unavailable Please install PyVICP to use this resource type.
2023-04-17 16:30:08,349 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2023-04-17 16:30:08,354 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2023-04-17 16:30:08,356 - pyvisa - DEBUG - Created library wrapper for py
2023-04-17 16:30:08,357 - pyvisa - DEBUG - Created ResourceManager with session 5663363
2023-04-17 16:30:08,434 - pyvisa - WARNING - Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::1027::24577::???::0::RAW
2023-04-17 16:30:10,475 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - opening ...
2023-04-17 16:30:10,481 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - is open with session 1076107
2023-04-17 16:30:10,482 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:10,484 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 7 bytes (last status None)
2023-04-17 16:30:10,495 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 1 bytes (last status <StatusCode.success: 0>)
2023-04-17 16:30:10,499 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:10,501 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 16:30:10,513 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:10,515 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 16:30:10,528 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:10,533 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:10,535 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 16:30:10,545 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:10,549 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:10,551 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 16:30:10,560 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:10,564 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:10,566 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 16:30:12,594 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:12,599 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:12,601 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 16:30:14,625 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:14,630 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:14,632 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 16:30:16,655 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:16,660 - pyvisa - DEBUG - Serial.write b'\xaa\x03\xfe5\x00\n\xe0'
2023-04-17 16:30:16,662 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 7 bytes (last status None)
2023-04-17 16:30:16,682 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-04-17 16:30:18,689 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - closing
2023-04-17 16:30:18,699 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - is closed
2023-04-17 16:30:18,701 - pyvisa - DEBUG - Closing ResourceManager (session: 5663363)
2023-04-17 16:30:18,702 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - closing
```
Loopback test:
```
#!/usr/bin/env python3

[... 33 lines cut ...]
            for rt in (None, '', '\n'):
[... cut ...]
                except pyvisa.errors.VisaIOError:
                    print('read_raw() failed')

                i.write_raw(TEST)
                print(f'rt={rt:4} ei={ei} ', end='')
                try:
                    print(f'read_bytes(0)=={i.read_bytes(0)}')
                except pyvisa.errors.VisaIOError:
                    print('read_bytes() failed')

        i.write_raw(TEST)
        print(f'read_bytes(LEN)=={i.read_bytes(len(TEST))}')  # Works
        print(f'     read_raw()=={i.read_raw()}')

        i.close()
```

MatthieuDartiailh, 2023-04-18:
Am I right in thinking that your issue is actually fixed but deserve to be properly documented ? Would you be able to make such a PR ?

Regarding the closing log message it stems from the resource manager keeping weak references to all resources and attempting to close them when it closes itself. We may be able to do better.

spukhafte-sys, 2023-04-18:
> Am I right in thinking that your issue is actually fixed but deserve to be properly documented ? Would you be able to make such a PR ?

Yes, I'll submit a PR.

MatthieuDartiailh, 2023-04-27:
It seems I misunderstood you since I though you only needed to properly configure the system to get things to work. Re-reading the VISA specs I realize it could not be so since your use case does not fit VISA approach which wants to ensure the message is complete.

I won't merge the PR since it would introduce a divergence in behavior between the VISA specs and pyvisa-py implementation. For your use case of simply discovering the behavior of an instrument the best I can suggest it to set a short timeout and read one byte at a time. In production you should request the right number of bytes since this will be the only way to safely get the whole message.

spukhafte-sys, 2023-04-27:
> I won't merge the PR since it would introduce a divergence in behavior between the VISA specs and pyvisa-py implementation. For your use case of simply discovering the behavior of an instrument the [...]

What is the point of the read_raw() method then? Shouldn't it behave as the current documentation specifies?

MatthieuDartiailh, 2023-04-27:
read_raw is simply meant to provide a way to bypass decoding the read data and stripping possible termination characters. 

It is not meant to bypass the guarantee VISA make that the receive message is either complete or contain as many bytes as requested.

spukhafte-sys, 2023-04-27:
> read_raw is simply meant to provide a way to bypass decoding the read data and stripping possible termination characters.

Sorry to contradict you, but read_raw() is not supposed to strip termination characters. From the docs:

> By default PyVISA assumes, that the instrument will add the termination character at the end of the data block and actually makes sure it reads it to avoid issues. This behavior fits well a number of devices. However some devices omit the termination character, in which cases the operation will timeout. In this situation, first makes sure you can actually read from the instrument by reading the answer using the read_raw function (you may need to call it multiple time), and check that the advertized length of the block match what you get from your instrument (plus the header). If it is so, then you can safely pass expect_termination=False, and PyVISA will not look for a termination character at the end of the message.

MatthieuDartiailh, 2023-04-27:
I meant that read_raw skip both the decoding of the received data and the removal of the termination characters. Sorry for the unclear phrasing.

spukhafte-sys, 2023-04-27:
> I meant that read_raw skip both the decoding of the received data and the removal of the termination characters. Sorry for the unclear phrasing.

Okay, but my patch is necessary to do that. Without my patch, read_raw() would return a timeout error and toss away the data received from the instrument.

MatthieuDartiailh, 2023-04-27:
I will try to clarify my point.

VISA mandates that to succeed a read operation either:
- return the requested number of bytes
- receive a signal indicating the end of the message. This signal can be:
  - the assertion of a hardware line (EOI line on GPIB)
  - a protocol specific signal (packet header on VXI-11 and USB)
  - a termination character

`read_raw` is not meant to relax those conditions (which you seem to expect). It is meant for advanced user for which the decoding of the message and the removal of the termination characters (if any) would be detrimental, for example because of the extra allocations those operations incur.

What you are looking for is a way to read data without any guarantee that you got all the data you expect, and this goes against the VISA specification. Since your usecase is early development exploration, performance should not be a main concern and hence reading a byte at a time should work.

spukhafte-sys, 2023-04-27:
The documentation for read_raw() contradicts your VISA mandate. It anticipates that some instruments will not send a termination character and that multiple calls to read_raw() with timeouts are necessary to assemble a complete instrument response. What else could one infer?

> In this situation, first makes sure you can actually read from the instrument by reading the answer using the read_raw function (you may need to call it multiple time), and check that the advertized length of the block match what you get from your instrument (plus the header).

MatthieuDartiailh, 2023-04-27:
I have a hard time finding the piece you quote. It is not in the read_raw docstring nor in the latest pyvisa documentation. Could you point me to the source of your quote ?

spukhafte-sys, 2023-04-27:
I can't get the version drop-down to work in either Safari or Chrome, but it shows "latest" [here](https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html#reading-binary-values).

MatthieuDartiailh, 2023-04-27:
This applies to the special case of binary values that may contain the termination character in any place hence the need to call multiple times read_raw. It does not go against my earlier comments.

spukhafte-sys, 2023-04-27:
> This applies to the special case of binary values that may contain the termination character in any place hence the need to call multiple times read_raw. It does not go against my earlier comments.

So what is raw about read_raw()?

spukhafte-sys, 2023-04-27:
Is the plan to deprecate read_raw()?

MatthieuDartiailh, 2023-04-28:
I am sorry this appears frustrating to you. As I explained earlier `read_raw` is raw in that it does not translate the received bytes to str. It uses are limited but there is no plan to deprecate it.

spukhafte-sys, 2023-04-28:
Yes, I'm a bit frustrated, but that's not your fault. Overall I'm quite pleased with the improvements made since 1.8. Let me pitch my point another way.

In my tests, the `read_raw` method discards valid data in the input buffer that precedes a timeout. If I read the code correctly, `read_bytes` behaves the same way. We have two read methods with similar behaviour, which means one is somewhat redundant. If `read_raw` didn't throw an exception on timeout, it would distinguish itself from `read_bytes`, broaden its range of uses, provide useful instrument data and restore its backwards-compatibility.

MatthieuDartiailh, 2023-04-29:
At this point, my memory fails me, and I cannot remember why the behavior changed. 

To me, the old behavior looks buggy, but I will have to dive back into the history to figure out what happened. I am quite busy so it may take time but feel free to ping me.