pyvisa/pyvisa #742: Multiple ResourceManager objects are not independent

dkriegner, 2023-04-17:
When using multiple `ResourceManager` objects in a multithreaded application closing one of them will also close/invalidate the other instances. Is this the intended behavior? 

Naively, one would hope that multiple `ResourceManager` are independent. Or is it mandatory that in _one_ python session one uses only _one_ ResourceManager? I could not find this in the documentation.

To Reproduce
--------------
Steps to reproduce the behavior:
1. In one python session, create two socket servers for testing (e.g. by the following code:
```
import socket
s1 = socket.create_server(("localhost", 50004))
s2 = socket.create_server(("localhost", 50005))
```
)
Likely this would be the same for any other device connection.
2. connect to the sockets and after some communication close one of them
```
import threading
import time

import pyvisa

def runner(resource, niter):
    """
    handle device connection and close it at the end
    """
    rm = pyvisa.ResourceManager()
    dev = rm.open_resource(resource)
    for i in range(niter):
        print(f"{resource}: test{i}")
        dev.write(f"test{i}")
        time.sleep(1)
    dev.close()
    rm.close()

t1 = threading.Thread(target=lambda: runner("TCPIP::localhost::50004::SOCKET", 3))
t2 = threading.Thread(target=lambda: runner("TCPIP::localhost::50005::SOCKET", 5))

print("starting threads")
t1.start()
t2.start()

print("joining threads")
t1.join()
print("joined thread 1")
t2.join()
print("joined thread 2")
```

3. The following Exception occurs when running the code from 2.
```
starting threads
joining threads
TCPIP::localhost::50005::SOCKET: test0
TCPIP::localhost::50004::SOCKET: test0
TCPIP::localhost::50005::SOCKET: test1
TCPIP::localhost::50004::SOCKET: test1
TCPIP::localhost::50005::SOCKET: test2
TCPIP::localhost::50004::SOCKET: test2
TCPIP::localhost::50005::SOCKET: test3
joined thread 1
TCPIP::localhost::50005::SOCKET: test4
Exception in thread Thread-2 (<lambda>):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/dk/tmp/test_pyvisa.py", line 22, in <lambda>
    t2 = threading.Thread(target=lambda: runner("TCPIP::localhost::50005::SOCKET",
  File "/home/dk/tmp/test_pyvisa.py", line 14, in runner
    dev.write(f"test{i}")
  File "/home/dk/tmp/python-venv/lib/python3.10/site-packages/pyvisa/resources/messagebased.py", line 197, in write
    count = self.write_raw(message.encode(enco))
  File "/home/dk/tmp/python-venv/lib/python3.10/site-packages/pyvisa/resources/messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]
  File "/home/dk/tmp/python-venv/lib/python3.10/site-packages/pyvisa/resources/resource.py", line 171, in session
    raise errors.InvalidSession()
pyvisa.errors.InvalidSession: Invalid session handle. The resource might be closed.
joined thread 2
```

**Output of `pyvisa-info`**

```
Machine Details:
   Platform ID:    Linux-6.2.10-gentoo-dist-x86_64-12th_Gen_Intel-R-_Core-TM-_i7-1265U-with-glibc2.37
   Processor:      12th Gen Intel(R) Core(TM) i7-1265U

Python:
   Implementation: CPython
   Executable:     XXXX/python-venv/bin/python3
   Version:        3.10.11
   Compiler:       GCC 12.2.1 20230304
   Bits:           64bit
   Build:          Apr 11 2023 19:52:06 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.6.1
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: disabled (zeroconf not installed)
      VICP INSTR:
         Please install PyVICP to use this resource type.
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

MatthieuDartiailh, 2023-04-17:
Currently we create a single resource manager for a given VISA library so in your case the resource manager is indeed shared. The VISA API only offers a `viOpenDefaultRM()` which demands that a single RM exists. We could implement a kind of reference counting on close but that sounds wasteful since the RM is already closed in `__del__`.

dkriegner, 2023-04-17:
Ok, I understand. 
Maybe it would be worth to add a note on the session behavior in the docstring of `ResoureManager.close()`?

PR #744 (2023-04-26): mention session sharing in ResourceManager.close docstring
Add a note in the docstring of ResourceManager.close to hint the user to the fact that this closes all shared resource manager connections.

Closes #742
closed 2023-04-26 by MatthieuDartiailh