pyvisa/pyvisa #748: read_raw() to picture read data length is 116 but real length more than 116

universeviva, 2023-06-12:
Instrument details
DS1ZC251201787

------------------
* Model: DS1ZC251201787
* Communication: USB
* Link to the documentation (if available):

Output of `pyvisa-info`

 Version: 0.6.0
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.5')
      GPIB INTFC: Available via Linux GPIB (b'4.3.5')

-----------------------

MatthieuDartiailh, 2023-06-12:
`read_raw` will stop if it encounters a term character. You may want to consult the documentation with respect to `read_binary_data`. Could you post your test script ?

universeviva, 2023-06-12:
i use read_raw() in the windows enviroments for read picture of oscilloscope complete read all data ( data length 51201) ,but i use read_raw() in the ubuntu enviroments for read picture of oscilloscope complete read all data ( data length 116) .the windows backends is ivi ,ubuntu is py

universeviva, 2023-06-12:
test script:

      self.session.read_termination = None
      self.session.write_termination = None
      self.session.write(instruction)
      bitStr = b''
      code = 200
      content = []
      rawAst = self.session.read_raw()

universeviva, 2023-06-12:
i guess the reathon maybe is backends .so i change pyvisa backends py to ivi in the ubuntu  but use ivi not found resource 

pyvis-info detail :

Machine Details:
   Platform ID:    Linux-5.15.0-73-generic-x86_64-with-glibc2.35
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.10.6
   Compiler:       GCC 11.3.0
   Bits:           64bit
   Build:          May 29 2023 11:10:38 (#main)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.23.3.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 24118016
         Spec. Version: 7340032
   py:
      Version: 0.6.0
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.5')
      GPIB INTFC: Available via Linux GPIB (b'4.3.5')

universeviva, 2023-06-14:
The instruments offers GPIB,USB and Lan ，use LAN  of oscilloscope read picture data complete read all data ( data length 51201) in the ubuntu but USB is not。
My instruments is  RIGOL  DS1104Z

universeviva, 2023-06-15:
I really appreciate your help.

i wil try

universeviva, 2023-06-15:
i have a another question.why can't I detect resources when I modify the configuration py to ini 
pyvisa-info output：
Machine Details:
Platform ID: Linux-5.15.0-73-generic-x86_64-with-glibc2.35
Processor: x86_64

Python:
Implementation: CPython
Executable: /usr/bin/python3
Version: 3.10.6
Compiler: GCC 11.3.0
Bits: 64bit
Build: May 29 2023 11:10:38 (#main)
Unicode: UCS4

PyVISA Version: 1.12.0

Backends:
ivi:
Version: 1.12.0 (bundled with PyVISA)
https://github.com/pyvisa/pyvisa/issues/1: /usr/lib/x86_64-linux-gnu/libvisa.so.23.3.0:
found by: auto
bitness: 64
Vendor: National Instruments
Impl. Version: 24118016
Spec. Version: 7340032
py:
Version: 0.6.0
ASRL INSTR: Available via PySerial (3.5)
USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
TCPIP INSTR: Available
GPIB INSTR: Available via Linux GPIB (b'4.3.5')
GPIB INTFC: Available via Linux GPIB (b'4.3.5')

when i install ivi  ican't find resource

universeviva, 2023-06-15:
when i use backends of py that find resources