pyvisa/pyvisa #749: Why can't I detect resources when I modify the configuration py to ini

universeviva, 2023-06-12:
Instrument details
普源数字示波器DS1104Z PLUS
------------------
* Model: 普源数字示波器DS1104Z PLUS
* Communication: USB
* Link to the documentation (if available):

Output of `pyvisa-info`
[image]
Machine Details:
   Platform ID:    Linux-5.15.0-73-generic-x86_64-with-glibc2.35
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.10.6
   Compiler:       GCC 11.3.0
   Bits:           64bit
   Build:          May 29 2023 11:10:38 (#main)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.23.3.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 24118016
         Spec. Version: 7340032
   py:
      Version: 0.6.0
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.5')
      GPIB INTFC: Available via Linux GPIB (b'4.3.5')

question describe:
 default backend py :
  1.i modify the backends to ivi ,but use ivi to find resource is not found. i use the pyvisa-shell  and list is not found resouce  (( 0) ASRL1::INSTR)
 2. i use pyvisa-shell -b py and list found resources (( 0) ASRL/dev/ttyS0::INSTR  ( 1) USB0::6833::1230::DS1ZC251201787::0::INSTR)
 why change the 

-----------------------

universeviva, 2023-06-13:
system is ubuntu 22.04

MatthieuDartiailh, 2023-06-15:
I have honestly no idea why NI fails to detect the device. Since PyVISA fully defers to the NI implementation when using the IVI backend, you may have more luck asking on a NI forum.

universeviva, 2023-06-15:
Thank you very much MatthieuDartiailh