pyvisa/pyvisa #750: pyvisa.ResourceManager unable to find my instrument

0yixuanli, 2023-06-17:
To Reproduce
--------------
Steps to reproduce the behavior:
1. First I installed pyvisa successfully, then I tried to connect to my instrument, but it only output "()", as below:

C:\Users\ASUS>python
Python 3.6.8 (tags/v3.6.8:3c6b436a57, Dec 23 2018, 23:31:17) [MSC v.1916 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyvisa
>>> rm=pyvisa.ResourceManager('C:/Windows/system32/visa32.dll')
>>> print(rm.list_resources())
()

2. I tried to assign the location of visa library, but still, it output "()" but within longer time, as below:
>>> rm=pyvisa.ResourceManager('C:/Windows/system32/visa32.dll')
>>> print(rm.list_resources())
()

4. I am sure that I have connected to the NI instrument, because it is shown in NI Expert.
5. Many thanks for your help!

**Output of `pyvisa-info`**
C:\Users\ASUS>python -m visa info
C:\Users\ASUS\AppData\Local\Programs\Python\Python36-32\lib\site-packages\visa.py:23: FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.

The reason for the deprecation is the possible conflict with the visa package provided by the https://github.com/visa-sdk/visa-python which can result in hard to debug situations.
  FutureWarning,
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      Intel64 Family 6 Model 61 Stepping 4, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\ASUS\AppData\Local\Programs\Python\Python36-32\python.exe
   Version:        3.6.8
   Compiler:       MSC v.1916 32 bit (Intel)
   Bits:           32bit
   Build:          Dec 23 2018 23:31:17 (#tags/v3.6.8:3c6b436a57)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR: Available via Linux GPIB (None)
      GPIB INTFC: Available via Linux GPIB (None)

arr-ee, 2023-06-17:
Hi,

First off, you are using quite old versions of python and pyvisa (and pyvisa-py, although this is not as relevant).

Next, could you please follow https://pyvisa.readthedocs.io/en/latest/faq/contributing.html#report-issues and post the logging output?

Finally, just to make sure, could you confirm that `C:/Windows/system32/visa32.dll` actually exists?

0yixuanli, 2023-06-19:
> Hi,
>
> First off, you are using quite old versions of python and pyvisa (and pyvisa-py, although this is not as relevant).
>
> Next, could you please follow [...]

Yes, thanks for your reply. I revised the logging output using pyvisa-info in my original report. BTW, I checked that the visa32.dll does exist in this path. 
I used python3.7.9 at first and then I updated it to py3.10 and finally py3.6, but they all can't fix my problems.

arr-ee, 2023-06-19:
> I revised the logging output using pyvisa-info in my original report.

Apologies for the confusion, I was referring to adding `pyvisa.log_to_screen()` to the beginning of your test script (== before creating a ResourceManager) to get more information about what’s going on.

> I checked that the visa32.dll does exist in this path.

good to know, thanks!

> BTW I used python3.7.9 at first and then I updated it to py3.10 and finally py3.6, but they all can't fix my problems. I installed pyvisa and pyvisa-info using pip install -U directly.

There is no reason to downgrade, especially to 3.6. pyvisa 1.12.0 only supports python >= 3.7, which is why you can’t install it on 3.6.

I would suggest going back to 3.10 unless you have other software that requires older Python version.

0yixuanli, 2023-06-19:
> > I revised the logging output using pyvisa-info in my original report.
> 
> Apologies for the confusion, I was referring to adding `pyvisa.log_to_screen()` to the beginning of your test script (== before creating a ResourceManager) to get more information about what’s going on.
Thanks for your suggestion, and know I get output as below:
>>> import pyvisa
>>> pyvisa.log_to_screen()
>>> rm = pyvisa.ResourceManager()
2023-06-19 16:20:39,473 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2023-06-19 16:20:39,474 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2023-06-19 16:20:39,501 - pyvisa - DEBUG - No user defined configuration
2023-06-19 16:20:39,537 - pyvisa - DEBUG - Automatically found library files: [None, None, None, None, None]
2023-06-19 16:20:39,538 - pyvisa - DEBUG - No user defined library files
2023-06-19 16:20:39,538 - pyvisa - DEBUG - Did not find IVI binary
2023-06-19 16:20:39,549 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2023-06-19 16:20:39,551 - pyvisa - DEBUG - USBSession and USBRawSession were not imported No module named 'usb'.
2023-06-19 16:20:39,699 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2023-06-19 16:20:39,754 - pyvisa - DEBUG - GPIBSession was correctly imported.
2023-06-19 16:20:39,754 - pyvisa - DEBUG - pyvisa-py is available.
2023-06-19 16:20:39,755 - pyvisa - DEBUG - Created library wrapper for py
2023-06-19 16:20:39,755 - pyvisa - DEBUG - Created ResourceManager with session 6736161
>>> print(rm.list_resources())
2023-06-19 16:21:01,522 - pyvisa - DEBUG - GPIB board 0 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,534 - pyvisa - DEBUG - GPIB board 1 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,550 - pyvisa - DEBUG - GPIB board 2 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,567 - pyvisa - DEBUG - GPIB board 3 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,584 - pyvisa - DEBUG - GPIB board 4 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,597 - pyvisa - DEBUG - GPIB board 5 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,610 - pyvisa - DEBUG - GPIB board 6 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,624 - pyvisa - DEBUG - GPIB board 7 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,638 - pyvisa - DEBUG - GPIB board 8 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,652 - pyvisa - DEBUG - GPIB board 9 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,677 - pyvisa - DEBUG - GPIB board 10 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,692 - pyvisa - DEBUG - GPIB board 11 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,707 - pyvisa - DEBUG - GPIB board 12 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,727 - pyvisa - DEBUG - GPIB board 13 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,748 - pyvisa - DEBUG - GPIB board 14 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,769 - pyvisa - DEBUG - GPIB board 15 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,781 - pyvisa - DEBUG - GPIB board 0 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,790 - pyvisa - DEBUG - GPIB board 1 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,798 - pyvisa - DEBUG - GPIB board 2 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,808 - pyvisa - DEBUG - GPIB board 3 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,825 - pyvisa - DEBUG - GPIB board 4 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,840 - pyvisa - DEBUG - GPIB board 5 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,859 - pyvisa - DEBUG - GPIB board 6 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,874 - pyvisa - DEBUG - GPIB board 7 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,889 - pyvisa - DEBUG - GPIB board 8 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,905 - pyvisa - DEBUG - GPIB board 9 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,925 - pyvisa - DEBUG - GPIB board 10 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:01,947 - pyvisa - DEBUG - GPIB board 11 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
[... cut ...]
2023-06-19 16:21:01,982 - pyvisa - DEBUG - GPIB board 13 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:02,003 - pyvisa - DEBUG - GPIB board 14 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
2023-06-19 16:21:02,025 - pyvisa - DEBUG - GPIB board 15 error in _find_boards(): GpibError('ask() error: Errno -535560155, Unknown error')
()
> 
> I would suggest going back to 3.10 unless you have other software that requires older Python version.
I get it and now am using python3.10.11.

0yixuanli, 2023-06-19:
Besides, if I specified visa32.dll, also using _pyvisa.log_to_screen()_, then the output will be different:
>>> import pyvisa
>>> pyvisa.log_to_screen()
>>> rm = pyvisa.ResourceManager('C:/Windows/system32/visa32.dll')
2023-06-19 16:40:51,006 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2023-06-19 16:40:51,007 - pyvisa - DEBUG - Created library wrapper for C:/Windows/system32/visa32.dll
2023-06-19 16:40:51,053 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x01D64460>',) -> 0
2023-06-19 16:40:51,054 - pyvisa - DEBUG - Created ResourceManager with session 27244880
>>> print(rm.list_resources())
2023-06-19 16:41:11,892 - pyvisa - DEBUG - viFindRsrc(27244880, '?*::INSTR', '<ViObject object at 0x03553970>', 'c_ulong(0)', <ctypes.c_char_Array_256 object at 0x03553BF8>) -> -1073807343
()

arr-ee, 2023-06-19:
Oh, you were one step ahead of me.

> 2023-06-19 16:41:11,892 - pyvisa - DEBUG - viFindRsrc(27244880, '?*::INSTR', '<ViObject object at 0x03553970>', 'c_ulong(0)', <ctypes.c_char_Array_256 object at 0x03553BF8>) -> -1073807343

I am not terribly familiar with native VISA implementations, but as far as I understand you're getting code 1073807343 in return to list request, which is `VISA: (Hex 0xBFFF0011) Insufficient location information or the device or resource is not present in the system.` This is a bit odd to me since I thought this error occurs when one tries to open a specific resource.

I wonder if this might be due to an issue with VISA installation — which might also explain why it's not being detected by `pyvisa` — but I doubt I can help further. Maybe Matthieu has more ideas.

MatthieuDartiailh, 2023-06-19:
I agree with @arr-ee that the fact that the VISA library is not detected is suspicious. So if you can indeed re-install VISA it would be nice.

0yixuanli, 2023-07-01:
I have tried to reinstall these softwares multiple times. However, this didn't fix the problem, so I guessed some fundamental files missing and reinstalled my Windows system. Luckily, this solved my problem. Thanks for your help!

MatthieuDartiailh, 2023-07-01:
That was extreme but happy you got things to work.

closed 2023-07-01 by MatthieuDartiailh