pyvisa/pyvisa #752: Interfacing with pyvisa-mock

ataras2, 2023-06-18:
I'm working on a project involving motors and it would be really useful to mock motor connections to verify GUI things before plugging in to a large system. I'm using [pyvisa-mock](https://github.com/microsoft/pyvisa-mock) from Microsoft, but it isn't actively supported. When creating a resource manager while specifying `visa_library`, the `pyvisa` code in this repo throws an error in `get_wrapper_class` of `highlevel.py`, since it tries to import `pyvisa_mock` instead of `pyvisa-mock`.

Currently, I've cloned a copy of `pyvisa` and changed that line by removing the `_` in 
`pkg = cast(PyVISAModule, import_module("pyvisa_" + backend_name))`, 
but this is not a clean fix if I want to share my code further. Is there something obvious I'm missing here?

To Reproduce
--------------
Steps to reproduce the behavior:
1. Clone [pyvisa-mock](https://github.com/microsoft/pyvisa-mock)
2. `pip install .`
3. Try to run a simple script:
```
from collections import defaultdict
from visa_mock.base.base_mocker import BaseMocker, scpi
from visa_mock.base.register import register_resource
from pyvisa import ResourceManager

class MockerChannel(BaseMocker):
    """
    A mocker channel for a multi channel voltage source.
    Voltages are zero by default
    """

    def __init__(self, call_delay: float = 0.0):
        super().__init__(call_delay=call_delay)
        self._voltage = 0

    # Lets define handler functions.

    @scpi(":VOLTage <voltage>")
    def _set_voltage(self, voltage: float) -> None:
        self._voltage = voltage

    @scpi(":VOLTage?")
    def _get_voltage(self) -> float:
        return self._voltage

class Mocker(BaseMocker):
    """
    The main mocker class.
    """

    def __init__(self, call_delay: float = 0.0):
        super().__init__(call_delay=call_delay)
        self._channels = defaultdict(MockerChannel)

    @scpi("*IDN?")
    def idn(self) -> str:
        """
        'vendor', 'model', 'serial', 'firmware'
        """
        return "Mocker,testing,00000,0.01"

    @scpi(":INSTRument:CHAnnel<channel>")
    def _get_channel(self, channel: int) -> MockerChannel:
        return self._channels[channel]

register_resource("MOCK0::mock1::INSTR", Mocker())

rc = ResourceManager(visa_library="@mock")
res = rc.open_resource("MOCK0::mock1::INSTR")
res.write(":INSTR:CHANNEL1:VOLT 2.3")
reply = res.query(":INSTR:CHA1:VOLT?")  # This should return '2.3'
reply = res.query(
    ":instrument:channel1:voltage?"
)  # We can either use the short form or the long form

```

**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Linux-5.15.0-71-generic-x86_64-with-glibc2.35
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/adam/miniconda3/envs/fresh_motors/bin/python
   Version:        3.9.16
   Compiler:       GCC 11.2.0
   Bits:           64bit
   Build:          May 15 2023 23:46:34 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      Binary library: Not found
   mock:
      Could not instantiate backend
      -> Wrapper not found: No package named pyvisa_mock
   py:
      Version: 0.6.3
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      VICP INSTR:
         Please install PyVICP to use this resource type.
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

arr-ee, 2023-06-18:
Any reason why re-exporting `visa_mock` under a module named as you want is not an option? That’s what `pyvisa-mock` is doing with `pyvisa-mock` module.

ataras2, 2023-06-18:
Thanks for the quick response. I hadn't thought of that but am not sure I understand completely, could you please elaborate further on how I would do this?

arr-ee, 2023-06-18:
https://github.com/microsoft/pyvisa-mock/blob/master/pyvisa-mock/__init__.py

Same thing, create a directory `pyvisa_mock`, and make a `__init__.py` along the lines of one above. I might be missing something but I think it should do the trick.

ataras2, 2023-06-19:
I've had a go with putting this in my own package (https://github.com/ataras2/newport-motors-py). After `pip install .`, it doesn't work (trying e.g. `pytest .`). Thoughts?

arr-ee, 2023-06-19:
If you try importing [this module](https://github.com/ataras2/newport-motors-py/blob/main/newport_motors/pyvisa_mock/__init__.py) in REPL, you'll notice its name is `newport_motors.pyvisa_mock`. 

However, as you pointed out earlier, when `pyvisa` resolves backend module name, it expects it to be the top level one, i.e. without `newport_motors`. Fix that, and it'll work*

\*there is another small issue in your code that prevents it from working, but I'm sure it's just a typo that you'll find on your own.

ataras2, 2023-06-21:
Managed to set this up, thanks for the help and patience! 
For anyone coming back to this thread, solution is to put that `__init__.py` in its own folder and make sure you change the `setup.py` accordingly.

closed 2023-06-21 by ataras2

MatthieuDartiailh, 2023-06-21:
For people interested in pyvisa-sim, we could maybe leverage some parts of pyvisa-mock (in particular, they seem to have a scpi command matcher, which could be very interesting).