pyvisa/pyvisa #754: [COM] Communication issue with Tektronix Oscilloscope using pyvisa

ebenezer-usih, 2023-06-20:
Instrument details
------------------
* Model: TEKTRONIX,MSO54,C014153,CF:91.1CT FV:1.12.5.5575
* Communication: USB

When using python to send query commands to my oscilloscope, sometimes, it returns a value, and other times, it does not (since I set the timeout to be None). It seems to always get stuck in this part of my code 'r = scope.query('*opc?') '. What do I do? I got the script from the tektronix Github repository.

I do not have this problem while working with my power supply unit and source measure unit.

Below is my python script:

import time # std module
import pyvisa as visa # http://github.com/hgrecco/pyvisa
import matplotlib.pyplot as plt # http://matplotlib.org/
import numpy as np # http://www.numpy.org/

visa_address = 'USB0::0x0699::0x0522::C014153::0::INSTR'

rm = visa.ResourceManager()
scope = rm.open_resource(visa_address)
scope.timeout = None # ms
scope.encoding = 'latin_1'
scope.read_termination = '\n'
scope.write_termination = '\n' #None
scope.write('*cls') # clear ESR

print(scope.query('*idn?'))

input("""
ACTION:
Connect probe to oscilloscope Channel 1 and the probe compensation signal.

Press Enter to continue...
""")

scope.write('*rst') # reset
t1 = time.perf_counter()
r = scope.query('*opc?') # sync
t2 = time.perf_counter()
print('reset time: {}'.format(t2 - t1))

scope.write('autoset EXECUTE') # autoset
t3 = time.perf_counter()
r = scope.query('*opc?') # sync
t4 = time.perf_counter()
print('autoset time: {} s'.format(t4 - t3))

# io config
scope.write('header 0')
scope.write('data:encdg SRIBINARY')
scope.write('data:source CH1') # channel
scope.write('data:start 1') # first sample
record = int(scope.query('horizontal:recordlength?'))
scope.write('data:stop {}'.format(record)) # last sample
scope.write('wfmoutpre:byt_n 1') # 1 byte per sample

# acq config
scope.write('acquire:state 0') # stop
scope.write('acquire:stopafter SEQUENCE') # single
scope.write('acquire:state 1') # run
t5 = time.perf_counter()
r = scope.query('*opc?') # sync
t6 = time.perf_counter()
print('acquire time: {} s'.format(t6 - t5))

# data query
t7 = time.perf_counter()
bin_wave = scope.query_binary_values('curve?', datatype='b', container=np.array)
t8 = time.perf_counter()
print('transfer time: {} s'.format(t8 - t7))

# retrieve scaling factors
tscale = float(scope.query('wfmoutpre:xincr?'))
tstart = float(scope.query('wfmoutpre:xzero?'))
vscale = float(scope.query('wfmoutpre:ymult?')) # volts / level
voff = float(scope.query('wfmoutpre:yzero?')) # reference voltage
vpos = float(scope.query('wfmoutpre:yoff?')) # reference position (level)

# error checking
r = int(scope.query('*esr?'))
print('event status register: 0b{:08b}'.format(r))
r = scope.query('allev?').strip()
print('all event messages: {}'.format(r))

scope.close()
rm.close()

# create scaled vectors
# horizontal (time)
total_time = tscale * record
tstop = tstart + total_time
scaled_time = np.linspace(tstart, tstop, num=record, endpoint=False)
# vertical (voltage)
unscaled_wave = np.array(bin_wave, dtype='double') # data type conversion
scaled_wave = (unscaled_wave - vpos) * vscale + voff

# plotting
plt.plot(scaled_time, scaled_wave)
plt.title('channel 1') # plot label
plt.xlabel('time (seconds)') # x label
plt.ylabel('voltage (volts)') # y label
print("look for plot window...")
plt.show()

MatthieuDartiailh, 2023-06-21:
Given the number of OPC commands in your script, could you tell us which one gives you a timeout error ?

arr-ee, 2023-06-21:
1. Please follow the [guide for reporting issues](https://pyvisa.readthedocs.io/en/latest/faq/contributing.html#report-issues) — omitting it often makes it hard if not impossible to help. 
1. > since I set the timeout to be None

    There are very few reasons to do so, and I doubt there is one here. It does make debugging more difficult though.

    Removing the timeout will point to the exact statement on which responses cease, and will allow further debugging.
1. Can you please confirm that basic communication works, e.g. `query('*IDN?')` works as expected?

ebenezer-usih, 2023-06-21:
Matthieu, It freezes at the very first instance of the OPC command, between t1 and t2, for the calculation of reset time. What's confusing me is that it works rarely, after a very long delay, most of the time, it just freezes there.

arr-ee, I apologize for not following the guide for reporting issues. Like I stated above, it does not go ahead to print the reset time as stated in the code, meaning it gets stuck at the OPC command between t1 and t2. And yes, query('*IDN?') works.

I logged pyvisa to the console as directed by the issue reporting guide, and this is the output:

2023-06-21 10:21:22,177 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2023-06-21 10:21:22,177 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2023-06-21 10:21:22,177 - pyvisa - DEBUG - No user defined configuration
2023-06-21 10:21:22,192 - pyvisa - DEBUG - Automatically found library files: [None, 'C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa64.dll', 'C:\\WINDOWS\\system32\\visa64.dll']
2023-06-21 10:21:22,192 - pyvisa - DEBUG - No user defined library files
2023-06-21 10:21:22,192 - pyvisa - DEBUG - The IVI implementation available
2023-06-21 10:21:22,192 - pyvisa - DEBUG - No user defined configuration
2023-06-21 10:21:22,192 - pyvisa - DEBUG - Automatically found library files: [None, 'C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa64.dll', 'C:\\WINDOWS\\system32\\visa64.dll']
2023-06-21 10:21:22,202 - pyvisa - DEBUG - No user defined library files
2023-06-21 10:21:22,265 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2023-06-21 10:21:22,265 - pyvisa - DEBUG - Created library wrapper for C:\WINDOWS\system32\visa32.dll
2023-06-21 10:21:22,281 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x00000186E562E740>',) -> 0
2023-06-21 10:21:22,281 - pyvisa - DEBUG - Created ResourceManager with session 1024
2023-06-21 10:21:22,281 - pyvisa - DEBUG - viParseRsrcEx(1024, 'USB0::0x0699::0x0522::C014153::0::INSTR', 'c_ushort(7)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x00000186E56E9BC0>, <ctypes.c_char_Array_256 object at 0x00000186E56E9C40>, <ctypes.c_char_Array_256 object at 0x00000186E56E9EC0>) -> 0
2023-06-21 10:21:22,281 - pyvisa - DEBUG - USB0::0x0699::0x0522::C014153::0::INSTR - opening ...
2023-06-21 10:21:22,322 - pyvisa - DEBUG - viOpen(1024, 'USB0::0x0699::0x0522::C014153::0::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x00000186E562E740>') -> 0
2023-06-21 10:21:22,322 - pyvisa - DEBUG - USB0::0x0699::0x0522::C014153::0::INSTR - is open with session 1
2023-06-21 10:21:22,322 - pyvisa - DEBUG - viSetAttribute(1, 1073676314, 4294967295) -> 0
2023-06-21 10:21:22,322 - pyvisa - DEBUG - viSetAttribute(1, <ResourceAttribute.termchar: 1073676312>, 10) -> 0
2023-06-21 10:21:22,322 - pyvisa - DEBUG - viSetAttribute(1, <ResourceAttribute.termchar_enabled: 1073676344>, 1) -> 0
2023-06-21 10:21:22,338 - pyvisa - DEBUG - viWrite(1, b'*cls\n', 5, 'c_ulong(5)') -> 0
2023-06-21 10:21:22,338 - pyvisa - DEBUG - viWrite(1, b'*idn?\n', 6, 'c_ulong(6)') -> 0
2023-06-21 10:21:22,338 - pyvisa - DEBUG - USB0::0x0699::0x0522::C014153::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-06-21 10:21:22,338 - pyvisa - DEBUG - viRead(1, <ctypes.c_char_Array_20480 object at 0x00000186E562E740>, 20480, 'c_ulong(49)') -> 0
TEKTRONIX,MSO54,C014153,CF:91.1CT FV:1.12.5.5575

ACTION:
Connect probe to oscilloscope Channel 1 and the probe compensation signal.

Press Enter to continue...

2023-06-21 10:21:29,510 - pyvisa - DEBUG - viWrite(1, b'*rst\n', 5, 'c_ulong(5)') -> 0
2023-06-21 10:21:29,510 - pyvisa - DEBUG - viWrite(1, b'*opc?\n', 6, 'c_ulong(6)') -> 0
2023-06-21 10:21:29,510 - pyvisa - DEBUG - USB0::0x0699::0x0522::C014153::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

arr-ee, 2023-06-21:
(for posterity: https://github.com/tektronix/Programmatic-Control-Examples/blob/master/Examples/Oscilloscopes/MidrangeScopes/src/SimplePlotExample/mdo_simple_plot.py appears to be the script used)

arr-ee, 2023-06-21:
Unfortunately, USB implementation quality can vary from instrument to instrument quite a bit. LXI tends to be a bit more consistent, potentially in part due to certification.

Glad you got things working.

closed 2023-06-21 by ebenezer-usih