pyvisa/pyvisa #756: [COM] Communication issue with HP3458A using pyvisa: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

arad96, 2023-06-22:
Instrument details
------------------
* Model: HP3458A
* Communication: GPIB
* Link to the documentation (if available): https://xdevs.com/doc/HP_Agilent_Keysight/HP%203458A%20Operating,%20Programming%20&%20Configuration.pdf

Output of `pyvisa-info`
-----------------------

```
$ pyvisa-info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 85 Stepping 4, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\alangebe\venvs\venv_hp_gpib\Scripts\python.exe
   Version:        3.8.5
   Compiler:       MSC v.1926 32 bit (Intel)
   Bits:           32bit
   Build:          Jul 20 2020 15:43:08 (#tags/v3.8.5:580fbb0)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 524390
```
## Problem Statement 
I want to automate data collection from my HP voltmeter. I tried querying the ID and I got a time out error. I followed the troubleshooting steps and confirmed that `multimeter.write()` works.
When I run `multimeter.read()` I get the error message pasted below.
I ran 
```
while True:
    print(my_instrument.read_bytes(1))

```
and I can see the proper data in the first 8 bytes "b'HP3458A" but the read_bytes command never times out and continuously spits out what is on the inst display. 

I have tried the terminators specified in the inst documentation and tried other common terminators but could not get the read to work properly.
ex.)
```
multimeter = rm.open_resource('GPIB0::22::INSTR', write_termination='\n', read_termination='\n')     # setting terminators 
.
.
.
print(multimeter.read_bytes(100, break_on_termchar=multimeter.read_termination))     # reads 100 bytes but never stops at the terminator
```
**I cannot get multimeter.query() to work becasuse a time out error occurs during the read() command.**
read_bytes() verifies that we communicate with the instrument properly the method does not stop at the specified terminators.

Any advice on debugging this further?

Thank you,

--ALF

## Code Segment
```
import pyvisa as visa

visa.log_to_screen()
[... 73 lines cut ...]
Buffer content: bytearray(b'')
Traceback (most recent call last):
  File "C:/Users/alangebe/PycharmProjects/pythonProject/main.py", line 26, in <module>
    main()
  File "C:/Users/alangebe/PycharmProjects/pythonProject/main.py", line 19, in main
    print(multimeter.read())
  File "C:\Users\alangebe\venvs\venv_hp_gpib\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\alangebe\venvs\venv_hp_gpib\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\alangebe\venvs\venv_hp_gpib\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\alangebe\venvs\venv_hp_gpib\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\alangebe\venvs\venv_hp_gpib\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
2023-06-21 20:42:40,948 - pyvisa - DEBUG - Closing ResourceManager (session: 47772280)
2023-06-21 20:42:40,948 - pyvisa - DEBUG - GPIB0::22::INSTR - closing
2023-06-21 20:42:40,948 - pyvisa - DEBUG - viDisableEvent(46649760, <EventType.all_enabled: 1073709055>, <EventMechanism.all: 65535>) -> 1073676291
2023-06-21 20:42:40,948 - pyvisa - DEBUG - viDiscardEvents(46649760, <EventType.all_enabled: 1073709055>, <EventMechanism.all: 65535>) -> 1073676292
2023-06-21 20:42:40,949 - pyvisa - DEBUG - viClose(46649760,) -> 0
2023-06-21 20:42:40,949 - pyvisa - DEBUG - GPIB0::22::INSTR - is closed
2023-06-21 20:42:40,950 - pyvisa - DEBUG - viClose(47772280,) -> 0
```

arr-ee, 2023-06-22:
I'm afraid we'll have to request a demo unit from Keysight to reproduce this issue, no way around it ;)

Seriously though, thank you for collecting all the data, this is very useful.

I think what's happening is a corner case caused by VISA library, 3458's peculiar implementation of SCPI, and possibly your GPIB adapter.

My suspicion is that `read_bytes` does not terminate because VISA is not getting a EOI and/or another signal from the instrument to indicate that transfer is over, and because of that we ignore termination characters: https://github.com/pyvisa/pyvisa/blob/071fa6d5a8e7703f1b38d4c0b0cccd0952a5972f/pyvisa/resources/messagebased.py#L373-L375

While I suspect that above is not wrong, can you try dropping the `OUTPUT 722;` part from the command you're sending? As far as I understand, that's part of the BASIC program (that is still featured prominently even in the latest manual, and is not explicitly mentioned anywhere) and is not something that should be sent to the meter. I checked a couple of known-working scripts talking to 3458A (not via pyvisa but that's irrelevant), and they're sending commands as I would expect, without BASIC bits: https://github.com/tin-/teckit/blob/master/devices/hp3458a.py#L247-L261

Reading TiN's code (and manual) further, on page 4-20, section `Output Termination`, we'll see what appears to be a confirmation of my hypothesis wrt `EOI`: by default it's not being set, and meter just streams readings. You can send `END ALWAYS` to make it behave like one would expect, ending each message with EOI — TiN does that as well: https://github.com/tin-/teckit/blob/957b4f1ef9eccd965c09900a15c16f1dbc9f0c8b/devices/hp3458a.py#L259

Or you can use `read_raw` and do message splitting on your end.

arad96, 2023-06-22:
Haha this is the ancient 31 y/o version not the keysight but still a beast :)
Thank you for helping out
When I drop the BASIC command `multimeter.write("ID?;")` I get the same output. Thank you 

When I add `multimeter.write("END ALWAYS;")` before reading everything works as expected! 
Thank you very much. the visa terminators and instrument terminator signals both need to be present. 
You Rock

closed 2023-06-22 by arad96