pyvisa/pyvisa #757: VI_ERROR_TMO timeout error when using `pyvisa-sim` backend

shareefj, 2023-06-23:
I'm starting to use `pyvisa-sim` to test an application and can't get even a basic command to work.  I've seen some related issues in the `pyvisa-sim` repo but given they were opened in 2016 I don't hold out much hope of them being answered.

I've got a simple YAML file as follows:

```yaml
spec: 1.1

devices:
  DS1102E:
    eom:
      USB INSTR:
        q: "\r\n"
        r: "\n"
    dialogues:
      - q: "*IDN?"
        r: "RIGOL DS1102E sim"
    error: ERROR

resources:
  USB0::1234::5678::123456789:
    device: DS1102E
```

Yet when I try to run a test I get:

```
DEBUG    pyvisa:devices.py:214 Writing into device input buffer: b'*'
DEBUG    pyvisa:devices.py:214 Writing into device input buffer: b'I'
DEBUG    pyvisa:devices.py:214 Writing into device input buffer: b'D'
DEBUG    pyvisa:devices.py:214 Writing into device input buffer: b'N'
DEBUG    pyvisa:devices.py:214 Writing into device input buffer: b'?'
DEBUG    pyvisa:messagebased.py:436 USB0::1234::5678::123456789::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
DEBUG    pyvisa:messagebased.py:445 USB0::1234::5678::123456789::0::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
=========================================================================================================== short test summary info ============================================================================================================
FAILED test_simulation.py::test_ds1102e_sim - pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

What could cause a timeout when reading a simulation backend?

MatthieuDartiailh, 2023-06-23:
My best guess is that you did need set the `write_termination` on your resource. As a consequence you do not send `\r\n` which the simulated instrument expects to signal the end of a command. As a consequence the command is not processed and no response is sent.

MatthieuDartiailh, 2023-06-23:
Could you post your full script and what the log after setting `write_termination` ?

shareefj, 2023-06-23:
OK, so a bit of heavy logging has produced this.  It looks like setting `write_termination` doesn't do anything.  Should I be seeing the `"\r\n"` being printed out when each byte is written to the input buffer of the simulated instrument?

```
INFO     pyvisa_sim.usb:usb.py:71 Starting write: len=5 data=b'*IDN?'
DEBUG    pyvisa:devices.py:214 Writing into device input buffer: b'*'
DEBUG    pyvisa:devices.py:214 Writing into device input buffer: b'I'
DEBUG    pyvisa:devices.py:214 Writing into device input buffer: b'D'
DEBUG    pyvisa:devices.py:214 Writing into device input buffer: b'N'
DEBUG    pyvisa:devices.py:214 Writing into device input buffer: b'?'
DEBUG    pyvisa:messagebased.py:436 USB0::1234::5678::123456789::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
INFO     pyvisa_sim.usb:usb.py:46 Starting read: end_char=10 enabled=1 timeout=2.0
DEBUG    pyvisa:messagebased.py:445 USB0::1234::5678::123456789::0::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
=========================================================================================================== short test summary info ============================================================================================================
FAILED test_simulation.py::test_ds1102e_sim - pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

However, if I force the command to be `"*IDN?\r\n"` then the command is written successfully and as long as I set a matching `read_termination = "\n"`, then it succeeds, or at least I get an ERROR rather than the correct response.

```
2023-06-23 13:36:18 [    INFO] Got scope=[<pydex_ivi.scope.IviScope object at 0x7f5c659a3c40>] (test_simulation.py:10)
2023-06-23 13:36:18 [    INFO] type(instrument)=<class 'pyvisa.resources.usb.USBInstrument'> (test_simulation.py:14)
2023-06-23 13:36:18 [    INFO] write_termination=
 (test_simulation.py:15)
2023-06-23 13:36:18 [    INFO] scope=<pydex_ivi.scope.IviScope object at 0x7f5c659a3c40> (test_simulation.py:16)
2023-06-23 13:36:18 [    INFO] Starting write: len=7 data=b'*IDN?\r\n' (usb.py:71)
2023-06-23 13:36:18 [   DEBUG] Writing into device input buffer: b'*' (devices.py:214)
2023-06-23 13:36:18 [   DEBUG] Writing into device input buffer: b'I' (devices.py:214)
2023-06-23 13:36:18 [   DEBUG] Writing into device input buffer: b'D' (devices.py:214)
2023-06-23 13:36:18 [   DEBUG] Writing into device input buffer: b'N' (devices.py:214)
2023-06-23 13:36:18 [   DEBUG] Writing into device input buffer: b'?' (devices.py:214)
2023-06-23 13:36:18 [   DEBUG] Writing into device input buffer: b'\r' (devices.py:214)
2023-06-23 13:36:18 [   DEBUG] Writing into device input buffer: b'\n' (devices.py:214)
2023-06-23 13:36:18 [   DEBUG] USB0::1234::5678::123456789::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>) (messagebased.py:436)
2023-06-23 13:36:18 [    INFO] Starting read: end_char=10 enabled=1 timeout=2.0 (usb.py:46)
2023-06-23 13:36:18 [    INFO] char=bytearray(b'E') (usb.py:57)
2023-06-23 13:36:18 [    INFO] char=bytearray(b'R') (usb.py:57)
2023-06-23 13:36:18 [    INFO] char=bytearray(b'R') (usb.py:57)
2023-06-23 13:36:18 [    INFO] char=bytearray(b'O') (usb.py:57)
2023-06-23 13:36:18 [    INFO] char=bytearray(b'R') (usb.py:57)
2023-06-23 13:36:18 [    INFO] char=bytearray(b'\n') (usb.py:57)
2023-06-23 13:36:18 [    INFO] res=ERROR (test_simulation.py:18)
PASSED  
```

I'm trying to use a modified `python-ivi` so my test looks like:

```python
def test_ds1102e_sim(request, ivi_scope):
    log.info(f"Got scope={ivi_scope}")
    scope = ivi_scope[0]
    scope.instrument.write_termination = "\r\n"
    scope.instrument.read_termination = "\n"
    log.info(f"type(instrument)={type(scope.instrument)}")
    log.info(f"write_termination={scope.instrument.write_termination}")
    log.info(f"scope={scope}")
    res = scope.query("*IDN?\r\n")
    log.info(f"res={res}")
```

shareefj, 2023-06-23:
@arr-ee I think you'll find it is.  I'm using `pyvisa` underneath `python-ivi` and am using `pyvisa-sim` as the backend.

arr-ee, 2023-06-23:
Unless you post a full repro I doubt we’ll find much at all.

arr-ee, 2023-06-23:
Case in point:

pyvisa & pyvisa-sim from pip.

```python
import pyvisa

rm = pyvisa.ResourceManager('test.yml@sim')

print(rm.list_resources())

i = rm.open_resource('USB0::1234::5678::123456789')
i.read_termination = '\n'
i.write_termination = '\r\n'

print(i.query('*IDN?'))
```

output:

```
('USB0::1234::5678::123456789::0::INSTR',)
RIGOL DS1102E sim
```

shareefj, 2023-06-23:
Hmmm, okay.  So here's a testcase.  From what I've understood so far from the `python-ivi` code was that it just uses `pyvisa` underneath.

Setting the `read_termination` fixed the hanging response so I assumed I was at least setting the attributes on the correct `pyvisa` resource.

https://github.com/idex-biometrics/python-ivi/tree/feature/pyvisa-sim/tests

shareefj, 2023-06-23:
Right, this looks like `python-ivi` calls `write_raw` instead of `write`, so the termination has no effect.

shareefj, 2023-06-23:
Thanks for the help here.  I've now updated the `pyvisa` wrapper within `python-ivi` and it works.  I think there's going to be a few more fixes in this area but at least I know where to look now.  Cheers.

closed 2023-06-23 by shareefj