pyvisa/pyvisa #758: ThorLabs PM100USB not listed by pyvisa-shell

m-kru, 2023-07-07:
I have ThorLabs PM100USB meter that is correctly listed by `lsusb`:
```
Bus 001 Device 032: ID 1313:8072 ThorLabs PM100USB Power and Energy Meter Interface
```
How when I execute list command in `pyvisa-shell` no device is listed.
Does anyone know what might be wrong?

afischer-sweepme, 2023-07-07:
Hi, the PM100 can be used with two different drivers, one for using USBTMC/VISA runtime and one is the newer TLPM driver to communicate via dll. There is a Thorlabs tool called 'Driver Switcher' that can be used to change the driver. We had the same problem when writing our SweepMe! driver for the PM100 (https://sweep-me.net/device/237/Thorlabs_PM100) in order to find the USBTMC resource.

m-kru, 2023-07-07:
I have found the DriverSwitcher application, changed the driver, but it works with neither of them. None device is listed.

lui-rojas, 2023-07-21:
Which Version of the Thorlabs Software and which version of the NI-Visa Software do you have install on your computer?

jllanfranchi, 2024-04-07:
## Summary

The most common reason why this happens for the PM100D (and other USBTMC devices) is that the user doesn't have permissions on the USB node. (There are also cases where an older or particular Linux kernel might not support USBTMC, but I expect that would not be the case for most users.)

## References

I learned about the solution to this reading instructions for the USBTMC kernel driver [here](https://github.com/dpenkler/linux-usbtmc/blob/master/README.md) and from discussion in the pyusb github on an issue where langid isn't readable for a USB device [here](https://github.com/pyusb/pyusb/issues/139#issuecomment-627325009).

There's also a similar answer to a similar issue [#609] in [this comment](https://github.com/pyvisa/pyvisa/issues/609#issuecomment-871484403).

## Install pyusb module

First, all of the below is predicated on having the `pyusb` Python module installed. So make sure you have that (`import usb` should succeed). If not, run `conda install pyusb` or `pip install pyusb` or, if you are using your Ubuntu/Debian system Python installation, `sudo apt install python3-usb` works; basically, whatever you do to install python modules in your Python environment, do that for pyusb.

## Run pyvisa with debug logging messages enabled

It was helpful to run pyvisa with debug-level logging enabled to help track down what was going on. If the solution I suggest doesn't help you, the debugging messages might help point you in the right direction.

To get the debug messages, `import logging` at the top of your script/module/notebook/etc. Then before you do pyvisa stuff, run

```python
logging.basicConfig(level=logging.DEBUG)
```

From thereon, debug messages will be emitted on stderr (from ALL python modules you call that use the `logging` module).

A good thing to try is the following:

```python
import logging
import pyvisa

logging.basicConfig(level=logging.DEBUG)
rm = pyvisa.ResourceManager("@py")
rm.list_resources()
```

which, when I _**do NOT have correct permissions**_ on the usbtmc devices, yields

```
DEBUG:pyvisa:SerialSession was correctly imported.                                                                                                                                             
DEBUG:pyvisa:USBSession and USBRawSession were correctly imported.                                                                                                                             
DEBUG:pyvisa:TCPIPSession was correctly imported.                                                                                                                                              
DEBUG:pyvisa:GPIBSession was not imported No module named 'gpib'.                                                                                                                              
DEBUG:pyvisa:Created library wrapper for py                                                                                                                                                    
DEBUG:pyvisa:Created ResourceManager with session 5021915
WARNING:pyvisa:Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::4883::32888::???::0::INSTR
WARNING:pyvisa:Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::2391::45848::???::0::INSTR
WARNING:pyvisa:Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::1027::24593::???::0::RAW
DEBUG:asyncio:Using selector: EpollSelector

('ASRL/dev/ttyUSB3::INSTR',
 'ASRL/dev/ttyUSB2::INSTR',
 'ASRL/dev/ttyUSB1::INSTR',
 'ASRL/dev/ttyUSB0::INSTR')
```

where the smoking gun is one or more lines like the following:

```
WARNING:pyvisa:Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::4883::32888::???::0::INSTR
WARNING:pyvisa:Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::2391::45848::???::0::INSTR
```

There are connected USB devices that advertise themselves as instruments ("INSTR"), but the question marks are populated where serial number should be because you don't have permission to read the serial number (or do much else, for that matter) on those devices.

Note that I am not worried about the `USB0::1027::24593::???::0::RAW` message, as that device does not advertise itself as INSTR (it says "RAW"), so it is nothing that we want to use pyvisa to control anyhow (it might be a printer, webcam, or some other peripheral).

When I run the above Python code & I _**do have correct permissions**_ (following my solution below), I get this output instead:

[... 44 lines cut ...]

where the returned string for the ThorLabs PM100D is

```
Bus 001 Device 016: ID 1313:8078 ThorLabs PM100D Compact Power and Energy Meter Console
```

and `ID 1313:8078` is the key bit to look at, where the first number is the `idVendor` and the second is `idProduct`.

Finally, you can make the rules apply unplugging/re-plugging-in the device by running [per guidance on stackexchange here](https://unix.stackexchange.com/a/39371):

```bash
sudo udevadm control --reload-rules
sudo udevadm trigger
```

### Add your user to the group specified

Add your user to the `dialout` group--if that's what you pick--or use another group, like `plugdev`, as mentioned [in this similar answer on a similar issue](https://github.com/pyvisa/pyvisa/issues/609#issuecomment-871484403); do note that comment sets permissions to 0770, but really  [... cut ...]  be found in many places but command-line instructions which are fairly universal (across many linux distros) can be found [here](https://linuxize.com/post/how-to-add-user-to-group-in-linux/). To summarize these:

```bash
sudo usermod -a -G groupname username
```

and logout and back in for a GUI (Gnome, KDE, Unity, etc.) session or start a new terminal such that the group change takes effect. If you're using a terminal, check by running the `groups` command that the group is among those listed.

arr-ee, 2024-04-07:
@jllanfranchi that’s a good writeup, thank you! I’ve been debugging a similar issue with usb devices recently, and it’s even possible to surface the permissions error after some poking at pyusb. I think there was an issue to provide better diagnostics for permissions errors, or we should open one.

One note: pyvisa-py does not use kernel’s usbtmc driver, so the rules helping you are the last two ones.

jllanfranchi, 2024-04-07:
You're welcome, and thanks for your feedback! Since I only used `/dev/usbtmc1` as if it were a simple TTY/serial port device, I thought that was exclusively a TTY/serial port thing (though I did have cognitive dissonance from the name).

I'd actually go further, and say that your advice e.g. on issue #791 and in the documentation which refers to [this link](https://www.xmodulo.com/change-usb-device-permission-linux.html) to check permissions is _counter-productive_ because I thought (assumed) that was referring to /dev/usbtmc1, and you can have all permissions on /dev/usbtmc1, be able to read from/write to /dev/usbtmc1 via minicom, and yet pyvisa will still not list the corresponding device! This felt like a dead end to me, & looks like it is for other users as well.

So the question is what is the "thing" that we need to look at to see if we have permissions set correctly? (I do not understand enough to know what that "thing" is; I only know /dev/usbtmc1 is NOT it.)

Ah, I think I found it! If I run `lsusb`, I get for the PM100D

```
Bus 001 Device 016: ID 1313:8078 ThorLabs PM100D Compact Power and Energy Meter Console
```

So it's not /dev/usbtmc1 that we need to check permissions on, it's actually

```
$ ls -l /dev/bus/usb/001/016
crw-rw----. 1 root dialout 189, 15 Apr  7 03:18 /dev/bus/usb/001/016
```

where 001 in the path comes from `Bus 001` in the `lsusb` listing and 016 in the path comes from `Device 016` in the `lsusb` listing. If the permissions on _that_ file are not correct, then pyvisa will not list the corresponding device.

So now I can bypass all the udev and adding users to groups complexity for simple debugging purposes:

```bash
sudo chmod 666 /dev/bus/usb/001/016
```

then pyvisa picks up the instrument, even if udev rules are completely wrong! Now the user can move on to digging into udev rules, adding users to appropriate groups, etc. so that permissions stick when unplugged/re-plugged and we don't give everyone permissions. But at least we know pyvisa can see and work with our device; it's not a dead end any longer.

arr-ee, 2024-04-07:
> to check permissions is counter-productive because I thought (assumed) that was referring to /dev/usbtmc1

I agree that that advice is imprecise at best — checking permissions on the tmc device won’t help, but adding a bus/device-specific rule as per link does do the trick. Since then I spent more time fighting USB issues, and ended up with the same chain of checks you outlined.

Would you care to open a PR adding a more detailed explanation to the docs? That’d be very useful, and I can try adding permissions check linking to the updated docs — or feel free to tackle that along the way.

Thanks!

jllanfranchi, 2024-04-07:
I'd be happy to try to re-work the docs to add this info. I see the following section: https://pyvisa.readthedocs.io/projects/pyvisa-py/en/latest/installation.html#usb-resources-usb-instr-raw is relevant, so that's where I'd presumably start. Anywhere else?

And not sure if I can hack the code to do this, but I feel it's possible so I will leave that door open & let you know if I get anywhere with that.

arr-ee, 2024-04-07:
Either there or in a new entry in https://pyvisa.readthedocs.io/projects/pyvisa-py/en/latest/faq.html would be fine. Given the level of detail, maybe FAQ + link to it from the installation page?