pyvisa/pyvisa #761: Memory leak in IVIVisaLibrary.read_asynchronously

tobinjones, 2023-07-20:
When using the IVIVisaLibrary backend, each time the `read_asynchronously()` method is called, an entry is added to the `async_read_jobs` attribute. There seems to be no way to clean these up.

Here is the function in question:
https://github.com/pyvisa/pyvisa/blob/071fa6d5a8e7703f1b38d4c0b0cccd0952a5972f/pyvisa/ctwrapper/highlevel.py#L310C23-L310C23

I am interested in this because I'm looking into implementing async reads per #403 

I would like to know if there is an intended way to use this method that I'm not understanding? Is this API considered stable? or could we change it?

To Reproduce
--------------

```python
from pyvisa import ResourceManager, constants, resources

rm = ResourceManager()
inst: resources.MessageBasedResource = rm.open_resource("??")

inst.visalib.enable_event(
    inst.session, constants.VI_EVENT_IO_COMPLETION, constants.VI_QUEUE
)

for n in range(3):
    inst.write("*IDN?")
    buf, job_id, status = inst.visalib.read_asynchronously(inst.session, inst.chunk_size)
    ev_type, ctx, status = inst.visalib.wait_on_event(
        inst.session, constants.VI_EVENT_IO_COMPLETION, 1000
    )
    inst.visalib.close(ctx)

# There are 3 entries here, including buffers.
print(inst.visalib._async_read_jobs)
```python

MatthieuDartiailh, 2023-07-20:
The trouble here is that we never know that nobody else will need the result of an async read. This API is stable but is unlikely to see any heavy use (otherwise somebody else would have complained) and could be deprecated. In the meantime we could add a method `get_buffer_from_id_and_pop` that would clean up the list.

tobinjones, 2023-07-20:
Ok good to know. Thanks! My feeling is that having the low-level API try to do this bookkeeping might be unnecessary complexity.

So far, `get_buffer_from_id` does not seem particularly useful in implementing a higher level async API. I suggest that perhaps the `get_buffer_from_id` could be marked as deprecated, and the signature of `read_asynchronously` could be changed to something like:

```python
    def read_asynchronously(
        self, session: typing.VISASession, count: int, retain_buffer: bool=True
    ) -> Tuple[SupportsBytes, types.ViJobId, constants.StatusCode]:
    ...
    if(retain_buffer):
        self._async_read_jobs.append((job_id, buffer))  # type: ignore
    ...
 ```

Which allows the behavior to be turned off. Later, `retain_buffer` could default to `False` and/or go away completely. Just a thought, and I might be missing a great reason to leave that functionality there!

closed 2026-10-13 by MatthieuDartiailh