pyvisa/pyvisa #764: [COM] Communication issue with Rigol MSO5374 using USB

jdrosent, 2023-08-10:
Instrument details
------------------
* Model: Rigol MSO5074 (hacked upgrade to MSO5374
* Communication: USB
* Link to the documentation: https://int.rigol.com/Images/MSO5000ProgrammingGuideEN_tcm7-4051.pdf

Output of `pyvisa-info`
-----------------------
Python:
   Implementation: CPython
   Executable:     C:\Users\XXXXXXX\Documents\Projects\OscilloscopeMeasurements\venv\Scripts\python.exe
   Version:        3.8.8
   Compiler:       MSC v.1928 64 bit (AMD64)
   Bits:           64bit
   Build:          Feb 19 2021 13:18:16 (#tags/v3.8.8:024d805)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1107765328
         Spec. Version: 7340544
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1107765328
         Spec. Version: 7340544
   py:
      Version: 0.7.0
      USB INSTR: Available via PyUSB (1.2.1). Backend: N/A
      USB RAW: Available via PyUSB (1.2.1). Backend: N/A
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

```
import pyvisa as visa
import numpy as np
import matplotlib.pyplot as plt

visa.log_to_screen()   # print debug output to screen

rm = visa.ResourceManager()
print(rm.list_resources())

scope = rm.open_resource('USB0::0x1AB1::0x0515::MS5A250801116::0::INSTR')

# verify the connection was made
scope.query("*IDN?") # this command is successful and returns:
                     # 'RIGOL TECHNOLOGIES,MSO5074,MS5A250801116,00.01.03.03.00\n'

# turn on signal generator
scope.write(":OUTP2 ON")

# Test Case 1 (Successful): acquiring data from the scope's screen--this works
# MSO5000ProgrammingGuideEN_tcm7-4051.pdf p.262
scope.write(":WAV:SOUR CHAN2")
scope.write(":WAV:MODE NORM")
scope.write(":WAV:FORM BYTE")
data = scope.query(":WAV:DATA?")
arr = bytes(data, 'utf-8')
data = np.frombuffer(arr,'B')
plt.plot(data)
plt.show()

# Test Case 2 (Fails): acquiring data from the scope's internal memory--this fails
scope.chunksize = 1024000
scope.timeout = 20000
scope.write(":WAV:SOUR CHAN2")
scope.write(":WAV:MODE RAW")
scope.write(":WAV:FORM BYTE")
scope.write(":WAV:POIN 1000")
scope.write(":ACQ:MDEP AUTO")
scope.write(":STOP")
rawdata = scope.query(":WAV:DATA?") # fails here!
arr = bytes(rawdata, 'utf-8')
data = np.frombuffer(arr,'B')
plt.plot(data)
plt.show()
```

Debug output for when Test Case 1 is successful:
```
data = scope.query(":WAV:DATA?")
2023-08-10 16:12:01,387 - pyvisa - DEBUG - viWrite(4, b':WAV:DATA?\r\n', 12, 'c_ulong(12)') -> 0
2023-08-10 16:12:01,388 - pyvisa - DEBUG - USB0::0x1AB1::0x0515::MS5A250801116::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-08-10 16:12:01,456 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001996DB363C0>, 20480, 'c_ulong(1012)') -> 0
```

Debug output for when Test Case 2 fails:
```
rawdata = scope.query(":WAV:DATA?")
2023-08-10 16:12:22,605 - pyvisa - DEBUG - viWrite(4, b':WAV:DATA?\r\n', 12, 'c_ulong(12)') -> 0
2023-08-10 16:12:22,606 - pyvisa - DEBUG - USB0::0x1AB1::0x0515::MS5A250801116::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-08-10 16:12:22,656 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001996DB36440>, 20480, 'c_ulong(8180)') -> -1073807360
2023-08-10 16:12:22,656 - pyvisa - DEBUG - USB0::0x1AB1::0x0515::MS5A250801116::0::INSTR - exception while reading: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
Buffer content: bytearray(b'')
2023-08-10 16:12:22,657 - pyvisa - DEBUG - USB0::0x1AB1::0x0515::MS5A250801116::0::INSTR - closing
2023-08-10 16:12:22,657 - pyvisa - DEBUG - viDisableEvent(3, <EventType.all_enabled: 1073709055>, <EventMechanism.all: 65535>) -> 0
2023-08-10 16:12:22,658 - pyvisa - DEBUG - viDiscardEvents(3, <EventType.all_enabled: 1073709055>, <EventMechanism.all: 65535>) -> 1073676292
2023-08-10 16:12:22,661 - pyvisa - DEBUG - viClose(3,) -> 0
2023-08-10 16:12:22,662 - pyvisa - DEBUG - USB0::0x1AB1::0x0515::MS5A250801116::0::INSTR - is closed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 648, in query
    return self.read()
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
```

If I then try to query the oscilloscope again I get the following error:
```
scope.query("*IDN?") # trying to query scope again
2023-08-10 16:19:39,629 - pyvisa - DEBUG - viWrite(3, b'*IDN?\r\n', 7, 'c_ulong(7)') -> 0
2023-08-10 16:19:39,630 - pyvisa - DEBUG - USB0::0x1AB1::0x0515::MS5A250801116::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-08-10 16:19:39,631 - pyvisa - DEBUG - viRead(3, <ctypes.c_char_Array_20480 object at 0x000001EAE9DB17C0>, 20480, 'c_ulong(0)') -> -1073807298
2023-08-10 16:19:39,632 - pyvisa - DEBUG - USB0::0x1AB1::0x0515::MS5A250801116::0::INSTR - exception while reading: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.
Buffer content: bytearray(b'')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 648, in query
    return self.read()
  File "C:\Users\Documents\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\Projects\OscilloscopeMeasurements\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

```

jdrosent, 2023-08-11:
Issue was solved by uninstalling and re-installing VISA drivers.

closed 2023-08-11 by jdrosent

MatthieuDartiailh, 2023-08-11:
Thanks for letting us know !