pyvisa/pyvisa #768: PyVISA stuck when logging to full fixed size queue

SDeriazSercalo, 2023-09-01:
Hello !

I found this issue that could turn a debugging session into quite a headache

When implementing a custom logging class with fixed size queue, pyVISA gets stuck if the queue ever gets full

Minimal working example :

```python
from queue import Queue
import logging
from pyvisa import ResourceManager, log_to_screen

ENABLE_ISSUE = False # Test the issue here

if ENABLE_ISSUE:
    uiQueue = Queue(10)
else:
    uiQueue = Queue()

class LogHandler(logging.Handler):
    def __init__(self, queue):
        super().__init__()
        self.queue = queue

    def emit(self, record : logging.LogRecord):
        self.queue.put(record)

def main():
    log_to_screen()
    logHandler = LogHandler(uiQueue)
    logging.getLogger().addHandler(logHandler)

    rm = ResourceManager() # Stuck here

if __name__ == '__main__':
    main()
```

If this is expected behavior, i believe some kind of warning should be displayed (avoiding the usual logger route of course)

pyvisa-info :

```
Machine Details:
   Platform ID:    Windows-10-10.0.19045-SP0
   Processor:      Intel64 Family 6 Model 165 Stepping 5, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\sebastien\AppData\Local\miniconda3\envs\burnin\python.exe
   Version:        3.10.12
   Compiler:       MSC v.1936 64 bit (AMD64)
   Bits:           64bit
   Build:          Jun 23 2023 22:34:57 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 22020096
         Spec. Version: 5244928
      #2: C:\windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 22020096
         Spec. Version: 5244928
   py:
      Version: 0.7.0
      ASRL INSTR: Available via PySerial (3.5)
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: ok
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```

MatthieuDartiailh, 2023-09-01:
PyVISA does nothing special with logging and simply rely on Python standard library. As a consequence t is up to you to adjust your handler code to prevent such a behavior. Dealing with every possible handler behavior within PyVISA is not reasonable.

SDeriazSercalo, 2023-09-01:
Thank you for the answer, that sounds good to me. Hopefully this issue can help somebody in the future who might encounter a similar problem

closed 2023-09-01 by SDeriazSercalo