pyvisa/pyvisa #769: [COM] Communication issue with keysight using pyvisa

bardagu, 2023-09-06:
Hi I’m getting this error -pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

but it’s not happening every time I try to control the device. It can works well and then this error pops up. 
Someone knows how to solve this?
Thanks in advance

MatthieuDartiailh, 2023-09-06:
Can you post the output of `pyvisa-info` ?

bardgu, 2023-09-07:
Exception in Tkinter callback
Traceback (most recent call last):
  File "X:\General\Characterization\Python\XXXXX\Python39\lib\tkinter\__init__.py", line 1884, in __call__
    return self.func(*args)
  File "C:\Users\bench\Desktop\PyCHAZ\GUI\MainProject.py", line 230, in <lambda>
    command=lambda: self.run_click(),
  File "C:\Users\bench\Desktop\PyCHAZ\GUI\MainProject.py", line 569, in run_click
    flag = self.project.mainproject(self.ip_name,
  File "C:\Users\bench\Desktop\PyCHAZ\XXXXX\XXXX\XXXXX", line 11, in mainproject
    flag = IP_Mapping[ip_name].test_map[test_name](proj)
  File "C:\Users\bench\Desktop\PyCHAZ\XXXXX\iPLL.py", line 37, in tosc_calibration
    flag = TOSC_Calibration_test.test(proj, my_ftdi, dcps, scope, dmm1, dmm2, current_dmm)
  File "C:\Users\bench\Desktop\PyCHAZ\XXXXX\TOSC_Calibration.py", line 92, in test
    global_control.power_on(dcps, dmm1, dmm2,current_dmm, scope)
  File "C:\Users\bench\Desktop\PyCHAZ\WD_Projects\MihirMp7\IPs\iPLL\global_control.py", line 72, in power_on
    dmm1.config.set_vdc('AUTO')
  File "C:\Users\bench\Desktop\PyCHAZ\Equipment\DMM.py", line 73, in set_vdc
    self._rm.query(f'MEASure:VOLTage:DC? {new_range}')
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\resources\messagebased.py", line 642, in query
    self.write(message)
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\resources\messagebased.py", line 197, in write
    count = self.write_raw(message.encode(enco))
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\resources\messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2796, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_CONN_LOST (-1073807194): The connection for the given session has been lost.

Thanks for your support.

arr-ee, 2023-09-07:
@bardgu this is not what was asked for.

Also, could you please use `code` formatting when pasting output next time? Thanks.

arr-ee, 2023-09-07:
https://pyvisa.readthedocs.io/en/latest/faq/contributing.html#report-issues has more information on reporting errors.