pyvisa/pyvisa #771: [COM] Communication issue with Pyvisa using Python

bardgu, 2023-09-14:
Im getting this error - the only way to solve this issue is to recomple my project.  by that itclean all resource_manager (open sessions)

my question is - is there any option to clear all open sources using pyvisa ? and not to recomple my project very time im getting this message? 

by the way am also attach the connection to my scope

```
resource_manager = pyvisa.ResourceManager()

class Scope:
    def __init__(self, ip: str, usbaddress: str = ''):
        self.ip = ip
        self.usbaddress = usbaddress

        if self.usbaddress == '':
            self.resource = f'TCPIP0::{ip}::INSTR'
        else:
            self.resource = f'USB0::{usbaddress}::0::INSTR'
        self._rm = resource_manager.open_resource(self.resource)
        self._rm.baud_rate = 9600
        self._rm.timeout = 30000

        self.id = self._rm.query('*IDN?')

        self.config = Config(self._rm, self)
        self.screen = Screen(self._rm, self)
        self.markers = Markers(self._rm, self)
        self.trigger = Trigger(self._rm, self)
        self.save_file = Save_Files(self._rm)
        self.measurements = Measurements(self._rm, self)
        self.jitter = Jitter(self._rm, self)
        self.functions = Functions(self._rm, self)
```

Instrument details
------------------
* Model: Osciloscope MSO 
* Communication: Ethernet 
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------
Exception in Tkinter callback
Traceback (most recent call last):
  File "X:\General\Characterization\Python\CHAZ_Environment\Python39\lib\tkinter\__init__.py", line 1884, in __call__
    return self.func(*args)
  File "C:\Users\bench\Desktop\PyCHAZ\GUI\MainProject.py", line 230, in <lambda>
    command=lambda: self.run_click(),
  File "C:\Users\bench\Desktop\PyCHAZ\GUI\MainProject.py", line 576, in run_click
    flag = self.project.mainproject(self.ip_name,
  File "C:\Users\bench\Desktop\PyCHAZ\WD_Projects\MihirMp7\MihirMp7.py", line 11, in mainproject
    flag = IP_Mapping[ip_name].test_map[test_name](proj)
  File "C:\Users\bench\Desktop\PyCHAZ\WD_Projects\MihirMp7\IPs\iPLL\iPLL.py", line 48, in active_and_leakage_current
    my_ftdi, dcps1, scope, dmm1, dmm2,dmm3,current_dmm = global_control.connect(proj)
  File "C:\Users\bench\Desktop\PyCHAZ\WD_Projects\MihirMp7\IPs\iPLL\global_control.py", line 18, in connect
    scope = Scope(proj.station_dict['Scope_Add'],'')
  File "C:\Users\bench\Desktop\PyCHAZ\Equipment\Scope.py", line 36, in __init__
    self._rm = resource_manager.open_resource(self.resource)
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\highlevel.py", line 3284, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\resources\resource.py", line 282, in open
    self.session, status = self._resource_manager.open_bare_resource(
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\highlevel.py", line 3209, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1851, in open
    ret = library.viOpen(
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\bench\Desktop\Pychaz_Guy\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_ALLOC (-1073807300): Insufficient system resources to perform necessary memory allocation.

MatthieuDartiailh, 2023-09-14:
I don't really have enough context to answer. My only suggestion is to ensure you close resources you do not need anymore.

bardgu, 2023-09-14:
Hi Matthieu, thanks for swift response.
if Im ussing `resource_manager.close()` im facing with another issue on the next time I try to open_resources
do you know a way to close resources or even just to clear the stack of all opened resources?

Thanks.

MatthieuDartiailh, 2023-09-14:
I meant the resource itself (so your _rm which has a close method) not the global resource manager

LaughingMaann, 2025-06-09:
assuming it's faulting at the else logic when opening a resource manager at the TCP comms
`        else:
            self.resource = f'USB0::{usbaddress}::0::INSTR'`
what i did to fix this issue is, was using the name that NIMAX assigned it. meaning, if you open nimax at the left pane, go to My System" > Devices and Interfaces > Network Devices > there should be a TCP address, port no and socket naming convention like this 

name  = "TCPIP0::420.429.0.69::10000::SOCKET"
^then open like 
inst = rm.open_resource(name)

in short, you cannot use the return output of 

resources= rm.list_resources()
^say if you print this and it's the 0th index or element

`for i, ree in enumerate(resources):
    print(f"idx: {i} = {ree}")`

idx: 0 = TCPIP0::MAP420::INST0::INSTR
idx: 1 = ASRL1::INSTR
idx: 2 = ASRL4::INSTR
idx: 3 = GPIB0::5::INSTR

^meaning, you cannot do inst = rm.open_resource(resources[0])

because it throws an error -1073807300. you have to use the NIMAX name directly, like so

name  = "TCPIP0::420.429.0.69::10000::SOCKET"
rm.open_resource(name)

such taht name = TCPIP::hostname::port::SOCKET
^and you can find the exact name from the nimax gui