pyvisa/pyvisa #772: Not handling NULL terminator

jonyscathe, 2023-09-19:
I am attempting to communicate over Ethernet with a Keithley KI4200A.
The Keithley gives the option of None, LF, CR or CR/LF line endings but also expects (and adds) a null terminator (\0).
I can't work out any settings that allow me to easily read off data. As soon as there is a `\0` in the read_termination I no longer get any data back. 

To Reproduce
--------------

With the Keithley set to None for line endings I can write successfully but reading using read() or query() returns an empty string. Reading using read_bytes does work, but only if I set the read_termination to ''.

```
>>> from pyvisa import ResourceManager
>>> 
>>> rm = ResourceManager()
>>> instrument = rm.open_resource("TCPIP0::169.254.208.172::1225::SOCKET")
>>> instrument.write_termination = "\0"
>>> instrument.read_termination = "\0"
>>> instrument.query('*IDN?')
''
>>> instrument.read_termination = ""
>>> instrument.write('*IDN?')
6
>>> instrument.read_bytes(43)
b'KEITHLEY INSTRUMENTS,KI4200A,1466085,V1.12\x00'
```

With the Keithley set to LFfor line endings I can write successfully, reading using read() or query() returns an empty string unless I don't include the \0 in read termination, but then I get a \0 in each subsequent response unless I do a clear() in-between each instruction.

```
>>> from pyvisa import ResourceManager
>>> 
>>> rm = ResourceManager()
>>> instrument = rm.open_resource("TCPIP0::169.254.208.172::1225::SOCKET")
>>> instrument.write_termination = "\n\0"
>>> instrument.read_termination = "\n0"
>>> instrument.query('*IDN?')
''
>>> instrument.read_termination = "\n"
>>> instrument.query('*IDN?')
'KEITHLEY INSTRUMENTS,KI4200A,1466085,V1.12'
>>> instrument.query('*IDN?')
'\x00KEITHLEY INSTRUMENTS,KI4200A,1466085,V1.12'
>>> instrument.query('*IDN?')
'\x00KEITHLEY INSTRUMENTS,KI4200A,1466085,V1.12'
>>> instrument.clear()
>>> instrument.query('*IDN?')   
'KEITHLEY INSTRUMENTS,KI4200A,1466085,V1.12'
>>> instrument.clear()
>>> instrument.query('*IDN?')   
'KEITHLEY INSTRUMENTS,KI4200A,1466085,V1.12'
```

I am running:
pyvisa = '1.13.0'
pyvisa-py = '0.7.0'
python = '3.11.5'
Running this from the python-slim docker container.

I guess with the clear() in-between each instruction I do have a way forward, but it is a bit clunky

MatthieuDartiailh, 2023-09-19:
Did you try setting read_termination to NULL ?Or \n\0 ?

jonyscathe, 2023-09-19:
Yes I did.
Whenever I had read_termination set to '\n\0" or "\0" I only ever received an empty string from either query() or read().

That is essentially the problem I am having.
In both examples above the first thing I try is having read_termination set correctly and an example of an empty response.
Then I show the only way I have been able to get data out - either read_bytes() in the case where the instrument termination is just '\0' or using read() when the instrument termination is "\n\0" but setting read_termination to just "\n" at least allows some data to come back.

MatthieuDartiailh, 2023-09-19:
I missed it but you are running pyvisa-py. Just so that I get a clear picture could you post the output of `pyvisa-info`.

MatthieuDartiailh, 2023-09-19:
Could you edit line 1149 of tcpip.py in pyvisa-py which reads `        term_byte = common.int_to_byte(term_char) if term_char else b""` so that it reads `        term_byte = common.int_to_byte(term_char) if term_char is not None else b""` ?

It should address your issue, hopefully.

jonyscathe, 2023-09-20:
Hi,
Thanks for your help, that does look like a likely solution. I can get access to the Keithley tomorrow so will try that out then.

And my pyvisa-info is:
```
Machine Details:
   Platform ID:    Linux-5.15.0-83-generic-x86_64-with-glibc2.36
   Processor:      

Python:
   Implementation: CPython
   Executable:     /root/.cache/pypoetry/virtualenvs/test-switch-control-qEQXemH4-py3.11/bin/python
   Version:        3.11.5
   Compiler:       GCC 12.2.0
   Bits:           64bit
   Build:          Sep  7 2023 12:46:15 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.0
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available 
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```

jonyscathe, 2023-09-22:
Can confirm that your 1 line fix does the trick

MatthieuDartiailh, 2023-09-22:
Great ! Could you make a PR ?

MatthieuDartiailh, 2023-10-12:
Closed by #https://github.com/pyvisa/pyvisa-py/issues/393

closed 2023-10-12 by MatthieuDartiailh