pyvisa/pyvisa #775: Unresolved attribute reference 'query' for class 'Resource'

Fiskelord, 2023-09-29:
I am experiencing an issue where my IDE can't find the function "query". Look at the code in the picture below:

[image]

The code works fine, no problems what so ever, but the IDE is confused about it, and auto complete doesn't work either.

I have tried with both PyCharm Professional 2023.2.1 and VS Code 1.82.2 with the Python extension from Microsoft installed, version 2022.10.1, and both are equally confused about it, VS Code doesn't mark it as an error though, just doesn't auto complete.

This is the only library i have this issue with, maybe there is some form of documentation missing for that specific function?

To Reproduce
--------------
Steps to reproduce the behavior:
Create a python script, type the code from the picture above, and see the IDE being weird about it.

pyvisa-info
--------------
```
Machine Details:
   Platform ID:    Windows-10-10.0.22621-SP0
   Processor:      Intel64 Family 6 Model 151 Stepping 2, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\jonas\AppData\Local\Programs\Python\Python311\python.exe
   Version:        3.11.4
   Compiler:       MSC v.1934 64 bit (AMD64)
   Bits:           64bit
   Build:          Jun  7 2023 05:45:37 (#tags/v3.11.4:d2340ef)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1107765328
         Spec. Version: 7340544
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1107765328
         Spec. Version: 7340544```

arr-ee, 2023-09-29:
> I am experiencing an issue where my IDE can't find the function "query".

This is because there is no function `query` in class `Resource`.

`query` is [defined in `MessageBasedResource` subclass](https://github.com/pyvisa/pyvisa/blob/071fa6d5a8e7703f1b38d4c0b0cccd0952a5972f/pyvisa/resources/messagebased.py#L624C9-L624C14), which interface-specific subclasses (like `GPIBInstrument` that you're actually using) inherit from.

I suspect that both pycharm and pylance (vscode) rely on type annotations here, and type annotations currently [don't mention specific subclasses](https://github.com/pyvisa/pyvisa/blob/071fa6d5a8e7703f1b38d4c0b0cccd0952a5972f/pyvisa/highlevel.py#L3218).

As a workaround you can try annotating your code, something along these lines:

```python
# More general `MessageBasedResource` should also work, and will also fit usb/lan instruments.
# Don't forget to import the class!
instr: GPIBInstrument = rm.open_resource ... <snip>
```

Fiskelord, 2023-10-05:
Thanks a lot for your help, @arr-ee, it did indeed work, although PyCharm is now complaining about ```instr``` being a ```Resource``` and not a ```GPIBInstrument``` as is defined.

Is it possible to fix the type annotations? I would be happy to do it and submit a PR, but it would be nice to know if someone has tried and failed for whatever reason, before trying myself ;)

arr-ee, 2023-10-05:
I briefly looked into that some months ago, and found no easy way of doing so.

The function will need to become generic, and I am not sure how to easily do that without making the user explicitly specify result type hints (and even that might not be trivial.) IIRC we can’t leverage `resource_pyclass` parameter as it is right now since we can’t use actual values for type hints? Would love to be corrected on this.

For now I can only suggest slapping a `# type: ignore` or a more specific version of same onto offending line; I do want a better solution but I’m not sure it will come soon.

MatthieuDartiailh, 2023-10-13:
This is a hard issue since the class is selected based on the resource name (so by value). 

We could use overload to make something better when a specific class is provided though. We could have an annotation of the type `def open_resource(resource_name: str, ..., resource_pyclass: Type[T]) -> T:` with `T` a type var bound to `Resource`.

bytelinker, 2024-04-17:
Hello,
is there any news on this issue, please?

Regards
Bytelinker

arr-ee, 2024-04-17:
@bytelinker you can see proposed workarounds, and note there is no PRs linking to this issue.

hmms, 2026-08-21:
Hi, I'd like to work on this using the overload-based approach @MatthieuDartiailh sketched above (a `TypeVar` bound to `Resource`, with `open_resource` overloaded so that passing an explicit `resource_pyclass=SomeClass` narrows the return type to `SomeClass`, and the no-argument/None case keeps returning the base `Resource`). Will open a PR shortly.