pyvisa/pyvisa #777: [COM] Communication issue with LeCroy DSO WS3054 using TCP/IP

AbdulghffarSh, 2023-10-13:
Hi everyone I hope you can help me with this issue, I am new in this field, I checked every documentation relate to this issue, whenever I add any query except the `instrument.query('*IDN?')` I receive the same output **WARNING : CURRENT REMOTE CONTROL INTERFACE IS TCPIP** the IDN query working fine and I receive the output fine

```
import pyvisa as visa

oscilloscope = visa.ResourceManager('@py')
instrument = oscilloscope.open_resource('TCPIP0::10.40.0.21::inst0::INSTR')
trigger_settings = instrument.query('TRIGger:MAIN?')
print("Trigger Settings:", trigger_settings)
```

Instrument details
------------------
Instrument Model	WS3054
Manufacturer	LECROY
Serial Number	LCRY3702N14196
Description	LeCroy DSO WS3054 [LCRY3702N14196]
LXI Extended Features	
LXI Version	1.4 LXI Device Specification 2011
Host Name	Compact.local.
TCP/IP Address	10.40.0.21
Firmware Revision	7.7.2
Instrument Address String	TCPIP0::10.40.0.21::inst0::INSTR

Output of `pyvisa-info`
-----------------------
Trigger Settings: WARNING : CURRENT REMOTE CONTROL INTERFACE IS TCPIP
[image]

Thank you

MatthieuDartiailh, 2023-10-13:
Your instrument is configured to communicate through the VICP protocol which is specific to LeCroy. PyVISA supports it when using the PyVISA-py backend (and you should use the resource name VICP::10.40.0.21::INSTR). Alternatively configured your instrument to use the VXI-11 protocol and then you current code should work.

AbdulghffarSh, 2023-10-13:
> Your instrument is configured to communicate through the VICP protocol which is specific to LeCroy. PyVISA supports it when using the PyVISA-py backend (and you should use the resource name [...]

thank you for your fast reply, I tried it already and getting timeout error for every query except the IDN?

> pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2023-10-13:
What did you try exactly ? could you post your code ?

AbdulghffarSh, 2023-10-13:
> What did you try exactly ? could you post your code ?
so I am not very aware of these devices and I got a task to export the measurements using python I saw the Manuel and got nothing I tried some simple queries like:

```
import pyvisa as visa

oscilloscope = visa.ResourceManager('@py')
instrument = oscilloscope.open_resource('VICP::10.40.0.21::INSTR')
print(instrument.query('OUTPut?'))
```
**output:** pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

the IDN command works fine:` print(instrument.query('*IDN?')) `

**output:** *IDN LECROY,WS3054,LCRY3702N14196,7.7.2

MatthieuDartiailh, 2023-10-13:
The manual at https://cdn.teledynelecroy.com/files/manuals/maui-remote-control-and-automation-manual.pdf does not list the output query. Could you try ``C1:CPL?``

MatthieuDartiailh, 2023-10-13:
So it looks like things are working. Do you need help with something else ?

AbdulghffarSh, 2023-10-15:
I greatly appreciate your assistance. I'm currently working on exporting an XLSX file containing readings from all the channels on the device. If you have any resources or recommendations for further understanding SCPI queries, please don't hesitate to share. Thank you.

closed 2023-10-15 by AbdulghffarSh

nine-V, 2024-12-12:
Hello, when I use tcpip to communicate between LabVIEW and the Lecroy oscilloscope, this warning also appears: WARNING : CURRENT REMOTE CONTROL INTERFACE IS TCPIP;
 and an error occurs: The format of the CMR? response is not as expected. Make sure that the Command Header is not being reprogrammed from the value configured in the Default Instrument Setup VI, where it is configured to OFF with the command CHDR OFF. (Default Instrument Setup.VI is called from Initialize.vi.) 
Do you know how to solve this problem? Thank you very much!
[image]