pyvisa/pyvisa #779: [COM] Communication issue with Prologix GPIB-USB using pyvisa

ebaum1211, 2023-10-20:
Instrument details
------------------
* Communication: Prologix GPIB-USB cable to Keithley 2400 SourceMeter

Hello everyone!

I know there have been a handful of similar threads, but the ones most similar to mine have been closed a while and/or had already gotten lengthy. My current test setup is running Pymeasure which uses pyvisa to communicate with a Keithley 2400 SourceMeter via a Prologix GPIB-USB cable. I have had this setup for many months working fine, and it just started having issues about a week ago (perhaps due to the newest version of NI-VISA??). NI-VISA is used as the backend for pyvisa. The computer, pyvisa, and NI-MAX all recognize the cable as ASRL::3 (COM3). There should be a GPIB portion to this name as well, but it doesn't show up. What is interesting is that using the serial.write library, we can communicate with the Keithley through the Prologix cable just fine. My goal is to use the Pymeasure wrapper though, so I need to get pyvisa working again. You will see below in the output of pyvisa-info that I am missing a gpib library. I checked both Windows/System32 and Windows/SysWOW64 and neither have a GPIB dll. Not sure how this happened, especially since it has worked for so long and just stopped working a week ago. Any suggestions are welcome! Thanks in advance! 

Output of `pyvisa-info`
-----------------------
UserWarning: GPIB library not found. Please manually load it using _load_lib(f
ilename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:                                                      
   Platform ID:    Windows-10-10.0.19045-SP0                          
   Processor:      Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\41EBaum\Anaconda3\python.exe
   Version:        3.9.13
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Aug 25 2022 23:51:50 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 24119296
         Spec. Version: 7340032
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 24119296
         Spec. Version: 7340032
   py:
      Version: 0.7.0
      ASRL INSTR: Available via PySerial (3.5)
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         gpib_ctypes is installed but could not locate the gpib library.
         Please manually load it using:
           gpib_ctypes.gpib.gpib._load_lib(filename)
         before importing pyvisa.
      GPIB INTFC:
         gpib_ctypes is installed but could not locate the gpib library.
         Please manually load it using:
           gpib_ctypes.gpib.gpib._load_lib(filename)
         before importing pyvisa.

LongnoseRob, 2023-10-21:
Unfortunately the prologix adapter requires the use of the `PrologixAdapter`, see the [documentaion](https://pymeasure.readthedocs.io/en/latest/api/adapters.html#prologix-adapter).

With that the automagic device detecion also will not work.

ebaum1211, 2023-10-31:
So I have VISA communication working again between the Prologix and the Keithley, even though my NI-VISA problem isn't quite solved. Using pyvisa-py for the backend instead worked. 

`adapter = pymeasure.adapters.PrologixAdapter('ASRL3::INSTR', visa_library="@py")`
 

I still get these warnings even though communication works again:

UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
warnings.warn(message)
0 : USB0::0x0699::0x0408::C012230::INSTR : TEKTRONIX,MDO3024,C012230,CF:91.1CT FV:v1.10
1 : ASRL3::INSTR : Visa IO Error: check connections
VI_ERROR_RSRC_BUSY (-1073807246): The resource is valid, but VISA cannot currently access it.
2 : ASRL7::INSTR : Not known
3 : ASRL9::INSTR : Not known
4 : ASRL10::INSTR : Visa IO Error: check connections
VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
5 : ASRL11::INSTR : Not known
('USB0::0x0699::0x0408::C012230::INSTR', 'ASRL3::INSTR', 'ASRL7::INSTR', 'ASRL9::INSTR', 'ASRL10::INSTR', 'ASRL11::INSTR')

 
I am still unsure why the NI-VISA backend worked for many months and no longer does. I am still confused about the above issues and warnings, too. I will post back if I find any answers to these questions, and I welcome any input as to why this might have happened.

MatthieuDartiailh, 2023-10-31:
Your prologix adapter does not require a GPIB library to be installed. The adapter mocks GPIB internally and only requires a serial link which is why it works even though no GPIB library could be found. 
Thta does not explain why you cannot get things to work with NI though.

ebaum1211, 2023-11-07:
Yeah I think that is the only question I have left. NI-VISA was fine for many months and then just stopped working it seems. Has pyvisa been tested with NI-VISA recently (on or after October 11th, 2023)? Maybe it is on the NI side, maybe within pymeasure, maybe pyvisa, or maybe it is an issue on my own personal laptop. Please let me know if you find anything or if you think you have an idea of what may have broke my NI-VISA backend functionality. Thanks everyone! P.S. my experiments have been running fine since switching to pyvisa-py