pyvisa/pyvisa #782: [query_binary_values] Timeout Error

rpj17-iNSANE, 2023-11-06:
Instrument details
------------------
* Model: Power Meter
* Communication: GPIB

Output of `pyvisa-info`
-----------------------
UserWarning: GPIB library not found. Please
 manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)

Machine Details:
   Platform ID:    Windows-10-10.0.19043-SP0
   Processor:      Intel64 Family 6 Model 140 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Version:        3.12.0
   Compiler:       MSC v.1935 64 bit (AMD64)
   Bits:           64bit
   
PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928
   py:
      Version: 0.7.1
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: ok
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         gpib_ctypes is installed but could not locate the gpib library.
         Please manually load it using:
           gpib_ctypes.gpib.gpib._load_lib(filename)
         before importing pyvisa.
      GPIB INTFC:
         gpib_ctypes is installed but could not locate the gpib library.
         Please manually load it using:
           gpib_ctypes.gpib.gpib._load_lib(filename)
         before importing pyvisa.

Hey there, 

This is my code,

import pyvisa
rm = pyvisa.ResourceManager()  

instr = rm.open_resource('GPIB0::15::INSTR')  
instr.read_termination = '\r\n'
instr.write_termination = '\r\n'
instr_details = instr.query('*IDN?')

data = instr.query_binary_values(f'LOGG? 0,1', datatype='d', is_big_endian=True, expect_termination=False)
print(data)

Whenever I run the above code and try to print data, I get a timeout error as below,

Traceback (most recent call last):
  line 24, in <module>
    data = instr.query_binary_values(f'LOGG? 0,1', datatype='d', is_big_endian=True, expect_termination=False)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File, line 751, in query_binary_values
    return self.read_binary_values(
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File, line 575, in read_binary_values
    block = self._read_raw(chunk_size)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File, line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File, line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File, line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File, line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

However when I use the instrument command 'LOGG? 0,1' in NI Max it works, also when I use the read_raw() method it works. But the read_raw method outputs values in IEEE floating point format, thats the reason I wanted to use the binary query command.

Looking foward for your response!!

MatthieuDartiailh, 2023-11-06:
My first question will be the following: are you sure your instrument is sending back f64 (i.e. double) ? If your instrument is sending f32 you will get half the amount of data which would explain the issue. If the instrument uses f32 you should use a datatype of ``"f"`` instead of ``"d"``.

rpj17-iNSANE, 2023-11-07:
Hey Matthieu, 
Thanks for the quick response. Yes, did try using the datatype 'f' isntead of 'd', but still did get the same error as above.

rpj17-iNSANE, 2023-11-07:
Hi Matthieu,
Sorry to disturb. Any updates ??

arr-ee, 2023-11-07:
As dedicated as Matthieu is, expecting a response within hours, especially during nighttime in Europe, might be optimistic.

In the meantime could you post output with logging enabled, ideally for both binary and raw read?

MatthieuDartiailh, 2023-11-07:
Thanks @arr-ee. 

Indeed having the result you get using read_raw or read_bytes(1) till you get a time out it would be helpful.

rpj17-iNSANE, 2023-11-07:
Output:
b'#3204\x00\x00\xdc\xc3\x00\x00\xdc\xc2\x00\x00\xdc\xc2\x00\x00\xdc\xc2\x00\x00\xdc\xc2'

This is the output I get when I use the read_raw method, the binary data is just the binary format of this, like 23 33 32 30 32 00 00 DC C3 00 00 DC C2 and so on. If I were to use the data from the read_raw method, I have to perform some operations and calculations to convert this data into decimal data. 
I could do that, but I feel using the binary values method could save me code space and my time.

MatthieuDartiailh, 2023-11-07:
First thing I can say is that the instrument is not sending double precision floating point number since the header specify the block is made of 204 bytes which cannot be divided by 8. You mentioned that using f did not work though. Could you maybe double check both with expect_termination=True and expect_termination=False ?

Could you also share the exact model of the instrument so that I can look at the manual ?

rpj17-iNSANE, 2023-11-08:
hey Matthieu,
I tried using both the expect_termination=True and expect_termination=False paramenters one by one; expect_termination=False seems to work fine for now. 
Now, the problem is, the values that are outputed are not as expected values.

instr.query_binary_values('LOGG? 0,1', datatype='f', is_big_endian=True, expect_termination=False)
This is command I used, 
first time when I ran the command with is_big_endian=True, I got [-110, -110, -110, -110, -110, ..........];
second time with is_big_endian=False, I got [7.984355, 7.12923, 7.732467, 7.2389472, 7.238748, ..........]
Here, the values in both the cases don't match the ideal values [-0.02685546875, -0.02685546875, -0.02685546875, -0.02685546875, -0.02685546875, ...........]; these values were obtained by manual calculation.

And, regarding the [Instrument Manual](https://downloads.santec.com/files/downloadfile/f61e25c4-1736-42a6-9ac4-4d48b1253e8f), sorry, a direct link for the manual isn't available.

MatthieuDartiailh, 2023-11-11:
Could you share part of the response so that we can double-check the calculation ?

Does the instrument display the result, and could you also check the conversion that way ?

rpj17-santec, 2025-04-21:
Hi @MatthieuDartiailh and team,

I'm reopening this issue which I had initially raised about a year ago. Back then, I believed the issue was resolved after some testing with `expect_termination=False`, but the timeout exception still persists intermittently, and I'm continuing to face problems retrieving binary logging data reliably from my power meter.

### Instrument Details:
- Instrument Model: Santec [MPM-210H-v1.0.pdf](https://github.com/user-attachments/files/19828509/MPM-210H-M-E-v1.0.pdf)
- Communication: GPIB / LAN

------------------

### Environment Details:
Machine Details:
   Platform ID:    Windows-10-10.0.19045-SP0
   Processor:      Intel64 Family 6 Model 186 Stepping 2, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\rpj\AppData\Local\Programs\Python\Python312\python.exe
   Version:        3.12.2
   Compiler:       MSC v.1937 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Feb  6 2024 21:26:36 (#tags/v3.12.2:6abddd9)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928

   py:
      Version: 0.7.1
      ASRL INSTR: Available via PySerial (3.5)
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         gpib_ctypes is installed but could not locate the gpib library.
         Please manually load it using:
           gpib_ctypes.gpib.gpib._load_lib(filename)
         before importing pyvisa.
      GPIB INTFC:
         gpib_ctypes is installed but could not locate the gpib library.
         Please manually load it using:
           gpib_ctypes.gpib.gpib._load_lib(filename)
         before importing pyvisa.

------------------

### Issue Summary:
Using the following command:
```python
data = mpm.query_binary_values('LOGG? 0,1', datatype='f', is_big_endian=True)
```

I receive the following timeout exception:
```makefile
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```
This happens despite the fact that:

- The same command works fine using `read_raw()` (though it requires manual parsing).
- The command also runs successfully in NI MAX without any timeout.
- I have tried switching datatype='d' to 'f' and 'i'.
- I have also tried switching `is_big_endian` and `expect_termination`.
- Sample output from `read_raw()`
  ```bash
    b'#3204\x00\x00\xdc\xc3\x00\x00\xdc\xc2\x00\x00\xdc\xc2...'
  ```

------------------

### Request:

Could you please help me debug this further or suggest a reliable workaround? I’d be happy to provide additional samples, logs, or test code.

------------------

Thanks again for your help and continued support!

MatthieuDartiailh, 2025-04-25:
Could you clarify what you mean by intermittent ? Does it sometimes work as expected ?

From the manual, I would expect the datatype to be "f" so on that we agree. When you say that read_raw works, did you check you get the expected amount of data (namely 204 bytes in your above example) ? (Same question for NI)

Also since you have both the NI and pyvisa-py backend installed can you specify which one you use ?

rpj17-santec, 2025-04-29:
Hello @MatthieuDartiailh ,

Thanks for the  response. 

First, to answer your questions,
> Could you clarify what you mean by intermittent ? Does it sometimes work as expected ?

I meant that with different `is_big_endian` and `expect_termination` settings, `query_binary_values()` sometimes completes without timeout, but the returned data is incorrect and unusable.

>From the manual, I would expect the datatype to be "f" so on that we agree. When you say that read_raw works, did you check you get the expected amount of data (namely 204 bytes in your above example) ? (Same question for NI)

Yes, I verified that all 204 bytes are correctly received.

> Also since you have both the NI and pyvisa-py backend installed can you specify which one you use ?

I am using the NI backend.

---

In addition, after some more investigation, I was able to find a workaround that seems to solve the issue for my setup. Here’s a brief summary of my findings:

##  Issue Summary
**1. VISA Status Handling Differences**

- LAN: Always returns StatusCode.success_max_count_read
  → Causes _read_raw() to loop unnecessarily, as it interprets this status as needing more reads. This eventually results in a timeout.

- GPIB: Returns StatusCode.success after a complete read
  → The loop exits correctly, and the read completes as expected.

------

**2. Extra Termination Character in LAN Responses**

- LAN: Includes an unexpected `\n` after the binary block header (e.g., after b'#')

- GPIB: Termination only appears at the end of the response.

**Impact**

  For LAN, I had to add +1 to the expected size to account for the extra newline character during block size parsing.

------

**3. Binary Block Misinterpretation**

- After _read_raw(), read_bytes() relies on the block header (e.g., b'#3404...') to determine how many bytes to expect.

- The extra newline in LAN responses causes this to miscalculate the remaining data length.
→ As a result, read_bytes() triggers an additional VISA read, which times out because there's nothing more to read.

------

### Workaround / Fix
- I used the write('command') followed by read_bytes(expected_size) for fetching logging data over LAN.

- Then I passed the read_bytes() response to a utility function like pyvisa.util.from_ieee_block() to convert the binary data into readable decimal format.

This workaround works reliably for LAN without triggering timeouts or misinterpreting the binary block.

Thanks again!

closed 2025-07-01 by rpj17-santec