pyvisa/pyvisa #783: [COM] Communication issue with EPA-1200H using GPIB

deerpasta, 2023-11-13:
Hi 
Currently I am working on a tool what measure RF, for which I will be using a EPA-1200H programmable step attenator. 
When I thought I had completed this tool, I realized that I couldn't control some of the attenuators.
however, when I use VI-VISA to test the attenuators, I can control all of them as well.
I discovered through the status indicator lights that the issue is occurring in the open_resource() part,it lights up when it's controlled.
same code,same model name,same address,same VISA resource name,only identification is different.
and no error message
[image]
[image]

may someone help me?
```python
import  pyvisa,time
from time import sleep
class   GPIB():
    def __init__(self) -> None:
        command=[None]*103
        command[100]='A876B54321'~~~~        command[0]	='AB87654321'
        self.Command=command
        self.gpib_address = "GPIB0::28::INSTR"
        try:
            self.visa_object = pyvisa.ResourceManager()
            self.connetion = self.visa_object.open_resource(self.gpib_address)
        except Exception as e:
            print(f"Error: {e}")    
            
    def Set_attenuation(self,Value:int):
        try:    
            self.connetion.write(self.Command[Value])
        except Exception as e:
            print(f"Error: {e}")    
    def over(self):
        try:    
            self.connetion.write(self.Command[0])
        except Exception as e:
            print(f"Error: {e}")      
        

attenuator=GPIB()
sleep(1)
attenuator.Set_attenuation(67)
print("0")
sleep(1)
attenuator.over()
print('1')
sleep(1)
``` 

Instrument details
------------------
* Model: EPA-1200H
* Communication: GPIB
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------

arr-ee, 2023-11-13:
I’m not sure what problem you are describing — “couldn’t control” is vague, and lack of any information related to your pyvisa setup does not help either.

> same code,same model name,same address,same VISA resource name,only identification is different.

Does this mean you are swapping instruments on the same address, with one working and not another?

deerpasta, 2023-11-13:
yes.
please let me know any information I can provide

deerpasta, 2023-11-15:
" links to documentation"
May you mean log file??

[gpib-log-work.log](https://github.com/pyvisa/pyvisa/files/13361160/gpib-log-work.log)
[gpib-log-don't-work.log](https://github.com/pyvisa/pyvisa/files/13361161/gpib-log-don.t-work.log)

EldarPolitkin, 2023-11-19:
Can you add the documentation for this device that list the commands that you need to send?

deerpasta, 2023-11-20:
> From logs, it _seems_ like we are getting a single character back that gets parsed as a response (unless I’m misunderstanding the log):
> 
> > 2023-11-15 14:40:09,875 - pyvisa - DEBUG - viRead(75714536, <ctypes.c_char_Array_20480 object at 0x119A47B0>, 20480, 'c_ulong(1)') -> 0

hmm......
I had try to test at NI MAX again
" query("*IDN?") "  => EPA-1200H V1.0

other command like A876B54321(set 100 dB).6br161(Meaningless string)
all will respond with "E."
----------------------
I have the user manual obtained from the distributor, but it's in Chinese, and it doesn't include debugging instructions, so I haven't provided it directly,sorry

[EPA-1200B GPIB Programming Guide.pdf](https://github.com/pyvisa/pyvisa/files/13407859/EPA-1200B.GPIB.Programming.Guide.pdf)

deerpasta, 2023-11-27:
thank you guys
the issue arises from the termination character,don't work version does not accept the termination character.
When I changed the "write_termination" parameter, it started functioning correctly.

closed 2023-11-27 by deerpasta