pyvisa/pyvisa #784: [COM] Communication issue with XXX using XXX

universeviva, 2023-11-16:
I have a question about pyinstaller 

i use the Pyinstaller to pack the pyvisa-py in order to get a executable program in ubuntu.

i exe the programer that can detect network instrument but not detect usb instruement.

system environment: ubuntu 

pack command :pyinstaller --distpath /opt/ate-conn-uni-standard --name conn-stand --collect-all=sanic --hidden-import=pyvisa_py --hidden-import=pyusb  -F main.py

I look forward to your reply. Thank you very much

LongnoseRob, 2023-11-16:
Have you read the docu on pyinstaller esp [the troubleshooting guide](https://pyinstaller.org/en/stable/when-things-go-wrong.html#finding-out-what-went-wrong)?
Would it be possible to share the logfiles?

I assume usb instruments are not detected due to missing pyusb or its lower-lvel dependencies..
Could also be a linux user-rights issue..

universeviva, 2023-11-17:
i understand thank you

universeviva, 2023-11-17:
thank you i see

closed 2023-11-17 by universeviva