pyvisa/pyvisa #789: [COM] Communication issue with CASSTK SA-14212 using TCPIP (works only if exact number of bytes to read is specified)

TonyBoothNZ, 2023-12-01:
When communicating with the CASSTK SA-14212 (an obscure fibre-bragg grating demodulator from http://www.casstk.com) there is an error when reading if I use `read()` or `read_bytes(n)` where `n` is more than the actual number of bytes returned.

Here is an example program with its output (the correct response would be the 8-letter string `SA-14212`):

```
import pyvisa as visa

HOST = '192.168.0.15'
PORT = 7088
message = '*IDN?'

visa.log_to_screen()

resourceManager = visa.ResourceManager() 
dev = 'TCPIP0::' + HOST + '::' + str(PORT) + '::SOCKET'
session = resourceManager.open_resource(dev)

session.read_termination = '\0'
session.write_termination = '\0'
session.write(message)

data = session.read_bytes(9)
print(data.decode())

session.close()
```
Output (error case)
-----------
```
2023-12-02 12:36:28,497 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2023-12-02 12:36:28,497 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2023-12-02 12:36:28,507 - pyvisa - DEBUG - No user defined configuration
2023-12-02 12:36:28,508 - pyvisa - DEBUG - Automatically found library files: [None, 'C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa64.dll', 'C:\\WINDOWS\\system32\\visa64.dll']
2023-12-02 12:36:28,508 - pyvisa - DEBUG - No user defined library files
2023-12-02 12:36:28,518 - pyvisa - DEBUG - The IVI implementation available
2023-12-02 12:36:28,518 - pyvisa - DEBUG - No user defined configuration
2023-12-02 12:36:28,518 - pyvisa - DEBUG - Automatically found library files: [None, 'C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa64.dll', 'C:\\WINDOWS\\system32\\visa64.dll']
2023-12-02 12:36:28,528 - pyvisa - DEBUG - No user defined library files
2023-12-02 12:36:28,528 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2023-12-02 12:36:28,528 - pyvisa - DEBUG - Created library wrapper for C:\WINDOWS\system32\visa32.dll
2023-12-02 12:36:28,579 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x000002584692FD40>',) -> 0
2023-12-02 12:36:28,589 - pyvisa - DEBUG - Created ResourceManager with session 4097
2023-12-02 12:36:28,589 - pyvisa - DEBUG - viParseRsrcEx(4097, 'TCPIP0::192.168.0.15::7088::SOCKET', 'c_ushort(6)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x00000258469844C0>, <ctypes.c_char_Array_256 object at 0x00000258469840C0>, <ctypes.c_char_Array_256 object at 0x0000025846984540>) -> 0
2023-12-02 12:36:28,589 - pyvisa - DEBUG - TCPIP0::192.168.0.15::7088::SOCKET - opening ...
2023-12-02 12:36:28,589 - pyvisa - DEBUG - viOpen(4097, 'TCPIP0::192.168.0.15::7088::SOCKET', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x000002584692FD40>') -> 0
2023-12-02 12:36:28,602 - pyvisa - DEBUG - TCPIP0::192.168.0.15::7088::SOCKET - is open with session 1
2023-12-02 12:36:28,602 - pyvisa - DEBUG - viSetAttribute(1, <ResourceAttribute.termchar: 1073676312>, 0) -> 0
2023-12-02 12:36:28,612 - pyvisa - DEBUG - viSetAttribute(1, <ResourceAttribute.termchar_enabled: 1073676344>, 1) -> 0
2023-12-02 12:36:28,612 - pyvisa - DEBUG - viWrite(1, b'*IDN?\x00', 6, 'c_ulong(6)') -> 0
2023-12-02 12:36:28,612 - pyvisa - DEBUG - TCPIP0::192.168.0.15::7088::SOCKET - reading 9 bytes (last status None)
2023-12-02 12:36:28,612 - pyvisa - DEBUG - viRead(1, <ctypes.c_char_Array_9 object at 0x000002584692FD40>, 9, 'c_ulong(8)') -> -1073807298
2023-12-02 12:36:28,612 - pyvisa - DEBUG - TCPIP0::192.168.0.15::7088::SOCKET - exception while reading: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.
Buffer content: bytearray(b'')
Traceback (most recent call last):
  File "C:\temp\junk3.py", line 17, in <module>
    data = session.read_bytes(9)
  File "C:\[edited]\Python\Python39\site-packages\pyvisa\resources\messagebased.py", line 370, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "C:\[edited]\Python\Python39\site-packages\pyvisa\ctwrapper\functions.py", line 2336, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\[edited]\Python\Python39\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\[edited]\Python\Python39\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.
```
It makes no difference if the termination characters are changed to `''` or `'\n'` (I think the actual setting is no termination character at all).

And here is the output with `data = session.read_bytes(9)` changed to `data = session.read_bytes(8)`:

[... 73 lines cut ...]
   Version:        3.9.2
   Compiler:       MSC v.1928 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Feb 19 2021 13:44:55 (#tags/v3.9.2:1a79785)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928

LongnoseRob, 2023-12-02:
1. It seems that these devices use a different protocol, not based on SCPI
2. with the help of an online translator CN-EN, I was able to fin this:
[English SA&FT documentation](http://sc.casstk.com/sa/help/#/README)

TonyBoothNZ, 2023-12-02:
LongnoseRob wrote:
> 1. It seems that these devices use a different protocol, not based on SCPI
> 
>     2. with the help of an online translator CN-EN, I was able to fin this:
>        [English SA&FT documentation](http://sc.casstk.com/sa/help/#/README)

Thank you for that.  Unfortunately that documentation, which we have seen, is for a newer version of the device.  It does not work at all for our device.  Ours is circa 2017 and the *XXX?-style commands work fine using python sockets (both UDP and  TCP, in fact), just not nicely with PyVISA.