pyvisa/pyvisa #791: [COM] Communication issue with Keysight DSOX4104A oscilloscope using USB

paniash, 2023-12-11:
Instrument details
------------------
* Model: Keysight InfiniiVision DSOX 4104A Oscilloscope
* Communication: USB
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------

```
Machine Details:
   Platform ID:    Linux-6.6.5-arch1-1-x86_64-with-glibc2.38
   Processor:

Python:
   Implementation: CPython
   Executable:     /home/ashish/pyvisa/bin/python
   Version:        3.11.6
   Compiler:       GCC 13.2.1 20230801
   Architecture:   ('x86', 64)
   Build:          Nov 14 2023 09:36:21 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.1
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```

When trying to list the visa address of the attached instrument, I only get `ASRL<Num>::INSTR` addresses and not the oscilloscope. The interface of the oscilloscope lists a USB type address i.e. `USB0::2391::6048::MY61260137::INSTR`.
On running `lsusb`, I can confirm that the oscilloscope is connected to my laptop displaying the line:
```
Bus 003 Device 014: ID 0957:17a0 Agilent Technologies, Inc. DSO-X 4104A
```

arr-ee, 2023-12-11:
Can you check permissions on the device file? See https://pyvisa.readthedocs.io/projects/pyvisa-py/en/latest/installation.html#usb-resources-usb-instr-raw for details.

paniash, 2023-12-11:
Thanks for the quick response! By connect do you mean as a general usb interface? I can see it in the output of `lsusb`.

paniash, 2023-12-11:
> Can you check permissions on the device file? See https://pyvisa.readthedocs.io/projects/pyvisa-py/en/latest/installation.html#usb-resources-usb-instr-raw for details.

Ah, I'll have to see if the permissions are right. Let me check that!

paniash, 2023-12-11:
> > Can you check permissions on the device file? See https://pyvisa.readthedocs.io/projects/pyvisa-py/en/latest/installation.html#usb-resources-usb-instr-raw for details.
> 
> Ah, I'll have to see if the permissions are right. Let me check that!

@arr-ee Unfortunately, doesn't fix the issue.

MatthieuDartiailh, 2023-12-11:
Can you open the VISA resource using the known resource name ? Sometimes, we cannot list a resource but still connect to it nonetheless.

paniash, 2023-12-11:
@MatthieuDartiailh My apologies but I'm very new to the VISA interface. Could you please tell me what exactly I need to open when you refer to VISA resource? Thanks.

MatthieuDartiailh, 2023-12-11:
What do you get when running the following script ?
```python
import pyvisa

rm = pyvisa.ResourceManager()
print(rm.list_resources())
osc = rm.open_resource("USB0::2391::6048::MY61260137::INSTR")
print(osc.query("*IDN?")
```

paniash, 2023-12-11:
> What do you get when running the following script ?
> 
> ```python
> import pyvisa
> 
> rm = pyvisa.ResourceManager()
> print(rm.list_resources())
> osc = rm.open_resource("USB0::2391::6048::MY61260137::INSTR")
> print(osc.query("*IDN?")
> ```

```
('TCPIP::192.168.20.142::INSTR', 'TCPIP::192.168.20.142::hislip0,4880::INSTR')
Traceback (most recent call last):
  File "/home/ashish/script.py", line 5, in <module>
    osc = rm.open_resource("USB0::2391::6048::MY61260137::INSTR")
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ashish/visa/lib/python3.11/site-packages/pyvisa/highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/ashish/visa/lib/python3.11/site-packages/pyvisa/resources/resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ashish/visa/lib/python3.11/site-packages/pyvisa/highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ashish/visa/lib/python3.11/site-packages/pyvisa_py/highlevel.py", line 167, in open
    sess = cls(session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ashish/visa/lib/python3.11/site-packages/pyvisa_py/sessions.py", line 321, in __init__
    self.after_parsing()
  File "/home/ashish/visa/lib/python3.11/site-packages/pyvisa_py/usb.py", line 81, in after_parsing
    self.interface = self._intf_cls(
                     ^^^^^^^^^^^^^^^
  File "/home/ashish/visa/lib/python3.11/site-packages/pyvisa_py/protocols/usbtmc.py", line 307, in __init__
    super(USBTMC, self).__init__(vendor, product, serial_number, **kwargs)
  File "/home/ashish/visa/lib/python3.11/site-packages/pyvisa_py/protocols/usbtmc.py", line 199, in __init__
    raise ValueError("No device found.")
ValueError: No device found.
```

MatthieuDartiailh, 2023-12-11:
Some instruments support enabling only a single interface at a time. Could you check your device configuration to see if USB is enabled ?

paniash, 2023-12-11:
> Some instruments support enabling only a single interface at a time. Could you check your device configuration to see if USB is enabled ?

Seems to be so.

[image]

MatthieuDartiailh, 2023-12-18:
I must say I do not a clear idea of what is going on. Could you maybe play with the `find_tmc_devices` function in `pyvisa_py.protocols.usbtmc` and see what you can find ? 

I would in particular look at the devices identified by `usb.core.find` to understand why none is seen as a USBTMC device.

DavidLutton, 2023-12-24:
We had USB legacy mode enabled on a DSOX3104A which made it non responsive over USB despite appearing in NI MAX.
Check and if enabled, turn it off.

Sponge5, 2024-01-05:
Hi, my scope is different (DSOX1102A), but my issue is the same. I have done all steps in this thread, including disabling `Compatibility Mode` on the scope and still have the same issues. Here is output from `find_tmc_devices`:

```
DEVICE ID 2a8d:1787 on Bus 003 Address 017 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
  bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x2a8d
 idProduct              : 0x1787
 bcdDevice              :  0x100 Device 1.0
 iManufacturer          :    0x1 Error Accessing String
 iProduct               :    0x2 Error Accessing String
 iSerialNumber          :    0x3 Error Accessing String
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 0 mA ====================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x27 (39 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0
   bmAttributes         :   0xc0 Self Powered
   bMaxPower            :    0x0 (0 mA)
    INTERFACE 0: Application Specific ======================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x3
     bInterfaceClass    :   0xfe Application Specific
     bInterfaceSubClass :    0x3
     bInterfaceProtocol :    0x1
     iInterface         :    0x4 Error Accessing String
      ENDPOINT 0x2: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :   0xff
      ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :   0xff
      ENDPOINT 0x83: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x2 (2 bytes)
       bInterval        :    0x1
```

Hope it gives you some information.

Here's `pyvisa-info`:

```
Machine Details:
   Platform ID:    Linux-6.6.9-arch1-1-x86_64-with-glibc2.38
   Processor:

Python:
   Implementation: CPython
   Executable:     /home/procioa/Other/python/keysightosc/bin/python
   Version:        3.11.6
   Compiler:       GCC 13.2.1 20230801
   Architecture:   ('x86', 64)
   Build:          Nov 14 2023 09:36:21 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.1
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial.tools'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```

paniash, 2024-01-05:
@MatthieuDartiailh @DavidLutton So I was able to get the DSOX4104A to work over ethernet so I didn't quite bother figuring out the issue over USB. I'll look over it and see if disabling the legacy mode fixes it.

jllanfranchi, 2024-04-07:
@panish it's possible that the discussion over on Issue #758 and specifically my comment [here](https://github.com/pyvisa/pyvisa/issues/758#issuecomment-2041328096) & subsequent discussion may be helpful for USB test equipment 

Specifically, when plugged in with that device, if you get for `lsusb`

```
Bus 003 Device 014: ID 0957:17a0 Agilent Technologies, Inc. DSO-X 4104A
```

then check permissions on the file

```bash
ls -l /dev/bus/usb/003/014
```

You can temporarily change the permissions on that file to test that pyvisa works with your equipment:

```bash
sudo chmod 666 /dev/bus/usb/003/014
```

and then try to list resources again. If that works, then work on fixing up your udev rules similarly as I've done in my comment on that issue.