pyvisa/pyvisa #794: Latest NI-VISA use nivisa64.dll instead of visa64.dll

kimbyungnam, 2024-01-11:
## Environment
- NI-VISA : NI-VISA 2023 Q4
- python : python 3.10.11
- OS : Windows 10 64bit

## 
In [How to Use NI-VISA with C in Visual Studio](https://knowledge.ni.com/KnowledgeArticleDetails?id=kA03q000000x8VUCAY&l=ko-KR), for 64 bit nivisa64.lib is used
[image]

##
nivisa64.lib use nivisa64.dll
```
dumpbin /HEADERS "C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Lib_x64\msc\nivisa64.lib"
```
[image]

##
I check the below code works
```
rm = ResourceManager("C:/WINDOWS/System32/nivisa64.dll')
```

##
**Output of `pyvisa-info`**

```
Machine Details:
   Platform ID:    Windows-10-10.0.19045-SP0
   Processor:      Intel64 Family 6 Model 167 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\qwer\Desktop\visa\venv\Scripts\python.exe
   Version:        3.10.11
   Compiler:       MSC v.1929 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Apr  5 2023 00:38:17 (#tags/v3.10.11:7d4cc5a)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Could not get more info:
            VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.
      #2: C:\windows\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Could not get more info:
            VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.
```

MatthieuDartiailh, 2024-01-11:
Usually the visa.dll acts as a common entry point for all implementers. You can directly get to the implementer dll but it is usually more compex and of little value. 

In you case, it looks like your installation is broken. Do you have another version of visa installed ?

fradaxx, 2024-03-13:
> ## Environment
> * NI-VISA : NI-VISA 2023 Q4
> * python : python 3.10.11
> * OS : Windows 10 64bit
>
> In [How to Use NI-VISA with C in Visual [...]

I experimented the same issue with my installation of NI-VISA on a Windows 11 Pro-based PC.
I have installed only NI-VISA version 2024 Q1 and the instruction

`rm = pyvisa.ResourceManager()`

without any argument, always exits with the error 

```
Exception "unhandled pyvisa.errors.VisaIOError"
VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.
File: C:\eric_env\Lib\site-packages\pyvisa\highlevel.py, Line: 251
```

The same problem arise also if I explicitly point to the visa64.dll library in c:\windows\system32 folder (my Python version is 3.12.2 64-bit) with the instruction

`rm = pyvisa.ResourceManager(r'C:\Windows\System32\visa64.dll')`

However, if I point explicitly to the nivisa64.dll library in the same folder, the problem doesn't arise and the script can continue to execute.

MatthieuDartiailh, 2024-03-15:
To me this is reminiscent of the need to use `os.add_dll_directory` in some circumstances. Out of curiosity if you add `"C:\Windows\System32"` does it fix your issue ?

fradaxx, 2024-03-15:
@MatthieuDartiailh thank you for your suggestion; I just tried to run this script on my PC:

```
import pyvisa
import os

os.add_dll_directory('c:/windows/system32')
rm = pyvisa.ResourceManager()
```

Unfortunately I obtain again the VI_ERROR_LIBRARY_NFOUND error:

```
Exception "unhandled pyvisa.errors.VisaIOError"
VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.
File: C:\eric_env\Lib\site-packages\pyvisa\highlevel.py, Line: 251
```

Cheers

MatthieuDartiailh, 2024-03-15:
Thanks for trying. I know it is a pain but could you try uninstalling NI, checking there is no visa.dll leftover and re-installing ?

As I mentioned I have seen such errors in corrupt installation before.

fradaxx, 2024-03-15:
@MatthieuDartiailh You're right: uninstalling and reinstalling any NI software is truly a painful process!
Unfortunately at the moment I don't have time to do that on my laptop. However next week I'll have the opportunity to do some tries on other PCs at work. I'll be back with some results ASAP.

Cheers