pyvisa/pyvisa #795: Unable to communicate properly with instrument.

wlgrd, 2024-01-11:
I have a Keysight 2450 SMU which I communicate with over USB.
This _has_ been working flawlessly before, but I don't know what component might have been updated that is breaking the behavior. 

To Reproduce
--------------
1. Running this script:
 
```
import pyvisa as visa
rm = visa.ResourceManager('@py')
print(rm.list_resources())
instr = rm.open_resource("USB0::1510::9296::04500460::0::INSTR")
print(instr.query('*IDN?'))
print(instr.query('MEAS:VOLT?'))
```

I am getting this exception thrown during `list_resources()`

```
Exception in thread zeroconf-ServiceBrowser-_hislip._tcp-3059708:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "src/zeroconf/_services/browser.py", line 601, in zeroconf._services.browser.ServiceBrowser.run
  File "src/zeroconf/_services/browser.py", line 470, in zeroconf._services.browser._ServiceBrowserBase._fire_service_state_changed_event
  File "src/zeroconf/_services/browser.py", line 480, in zeroconf._services.browser._ServiceBrowserBase._fire_service_state_changed_event
  File "src/zeroconf/_services/__init__.py", line 56, in zeroconf._services.Signal.fire
  File "src/zeroconf/_services/browser.py", line 204, in zeroconf._services.browser._on_change_dispatcher
  File "/home/wilgaard/.local/lib/python3.10/site-packages/pyvisa_py/tcpip.py", line 1420, in add_service
    properties[key.decode()] = val.decode()
AttributeError: 'NoneType' object has no attribute 'decode'
('ASRL/dev/ttyS4::INSTR', 'ASRL/dev/ttyS0::INSTR', 'ASRL/dev/ttyACM0::INSTR', 'ASRL/dev/ttyACM1::INSTR', 'USB0::1510::9296::04500460::0::INSTR')
```

Using `pyvisa-shell`'s `list` command also gives the same exception but is able to list the devices. The real problem first arises when I try queering for `MEAS:VOLT?`, getting "TSP Syntax error at line 1: function arguments expected near '?'" on the instrument display. I have never had problems with these **SCPI** commands before. `*IDN?` works as expected.

```
(open) query *IDN?
Response: KEITHLEY INSTRUMENTS,MODEL 2450,04500460,1.7.5b
```

**Output of `pyvisa-info`**
Machine Details:
   Platform ID:    Linux-5.15.0-88-generic-x86_64-with-glibc2.31
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/local/bin/python3.10
   Version:        3.10.7
   Compiler:       GCC 9.4.0
   Architecture:   ('x86', 64)
   Build:          Nov 21 2023 09:57:23 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.1
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available 
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

Hopefully this is some silly, stupid thing but I hope someone can help out with this one.

wlgrd, 2024-01-11:
Wow, it was that simple... I guess some power outage must have done it as I have power cycled it _a lot_. Thank you very much!

closed 2024-01-11 by wlgrd

MatthieuDartiailh, 2024-01-11:
Happy to help