pyvisa/pyvisa #796: [COM] Communication issue with Keysight B1500A (Parameter Analyzer) using PyVISA

NabilaTasnim, 2024-01-12:
My code is:

```
import pyvisa
import time
visa_address = 'GPIB0::8::INSTR'
rm = pyvisa.ResourceManager()
b1500 = rm.open_resource(visa_address)
try:
 
    b1500.write(':SOURce1:VOLTage:LEVel 3.0')  # Set SMU1:HR to 3V
    b1500.write(':SOURce4:VOLTage:LEVel 0.0')  # Set SMU4:HR to 0V

    time.sleep(3)

    b1500.write(':MEASure1:CURRent?')
    response_smu1 = b1500.read()
    

    b1500.write(':MEASure4:CURRent?')
    response_smu4 = b1500.read()
    
    print(f"Current on SMU1:HR: {response_smu1} A")
    print(f"Current on SMU4:HR: {response_smu4} A")

finally:
    b1500.close()
```

I am trying to measure the current flowing through a device and the device is connected in between SMU1:HR and SMU4:HR channels of B1500 (parameter analyzer). I have tried to use various methods like b1500.query(':MEASure4:CURRent?'), b1500.write(':MEASure4:CURRent?') and b1500.read_raw(), b1500.write(':MEASure4:CURRent?') and b1500.read_bytes(), b1500.write(':MEASure4:CURRent?') and b1500.read('\n')etc. I have also provided a very large time for timeout. But still, I am getting the following error - 

Traceback (most recent call last):
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\resistor_test.py", line 23, in <module>
    current_smu1 = (b1500.read('\n'))
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 489, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Can anyone please help to solve this issue?

Instrument details
------------------
* Model: Keysight B1500A
* Communication: GPIB
* Link to the documentation (if available):

Output of `pyvisa-info`
[image]

LongnoseRob, 2024-01-12:
I would recommend to check the following:- does the device switch to remote status?- query the ID of the device- are the 3V set?- is it necessary to enable the outputs?- does the device indicate errors which might need a reset?Robert

 

    

    

pyvisa-info.jpg (view on web)

NabilaTasnim, 2024-01-12:
I have tried to check the things you have suggested. But still, I am getting errors. I tried to check whether there is any system error but it is also showing timeout error. My updated code and error message are - 

```
import pyvisa
import time

visa_address = 'GPIB0::8::INSTR'

# Create a VISA resource manager and open a connection to the instrument
rm = pyvisa.ResourceManager()
b1500 = rm.open_resource(visa_address, timeout = 9000)
idn = b1500.query('*IDN?')
print(f'Connected to: {idn}')
b1500.write("*RST")

b1500.read_termination = '\n'
b1500.write_termination = '\n'
b1500.write('SYSTem:REMote')

try:
    
    b1500.write("OUTP ON")
    b1500.write(":SOURce1:FUNC VOLT")
    b1500.write(":SOURce4:FUNC VOLT")
    b1500.write(':SOURce1:VOLTage:LEVel 3.0')  # Set SMU1:HR to 3V
    b1500.write(':SOURce4:VOLTage:LEVel 0.0')  # Set SMU4:HR to 0V
 
    b1500.write("*OPC")
    b1500.write("*WAI")

    b1500.query("SYST:ERR?")
    
    #Measure the current on SMU1:HR
    b1500.write(':MEASure1:CURRent?')
    b1500.write("*OPC")
    b1500.write("*WAI")
    current_smu1 = b1500.read()
finally:
    # Close the connection
    b1500.close()
```

I could not understand what the problem is. Can you please help? @LongnoseRob

NabilaTasnim, 2024-01-12:
Dear Robert,

Thank you for your suggestion. I have tried to check the things you have
suggested. But still, I am getting errors. I tried to check whether there
is any system error but it is also showing timeout error. My updated code
and error message are -

import pyvisa
import time

visa_address = 'GPIB0::8::INSTR'

# Create a VISA resource manager and open a connection to the instrument
rm = pyvisa.ResourceManager()
b1500 = rm.open_resource(visa_address, timeout = 9000)
idn = b1500.query('*IDN?')
print(f'Connected to: {idn}')
b1500.write("*RST")

b1500.read_termination = '\n'
b1500.write_termination = '\n'
b1500.write('SYSTem:REMote')

try:

    b1500.write("OUTP ON")
    b1500.write(":SOURce1:FUNC VOLT")
    b1500.write(":SOURce4:FUNC VOLT")
    b1500.write(':SOURce1:VOLTage:LEVel 3.0')  # Set SMU1:HR to 3V
    b1500.write(':SOURce4:VOLTage:LEVel 0.0')  # Set SMU4:HR to 0V

    b1500.write("*OPC")
    b1500.write("*WAI")

    b1500.query("SYST:ERR?")

    #Measure the current on SMU1:HR
    b1500.write(':MEASure1:CURRent?')
    b1500.write("*OPC")
    b1500.write("*WAI")
    current_smu1 = b1500.read()
finally:
    # Close the connection
    b1500.close()

error message:
Connected to: Agilent Technologies,B1500A,0,A.06.01.2016.0401

Traceback (most recent call last):
  File
"C:\Users\b1500\PycharmProjects\TestProjectNabila\BasicMeasurement.py",
line 34, in <module>
    b1500.query("SYST:ERR?")
  File
"C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\resources\messagebased.py",
line 648, in query
    return self.read()
  File
"C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\resources\messagebased.py",
line 486, in read
    message = self._read_raw().decode(enco)
  File
"C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\resources\messagebased.py",
line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File
"C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\ctwrapper\functions.py",
line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File
"C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py",
line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File
"C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\highlevel.py",
line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired
before operation completed.

Process finished with exit code 1

I could not understand what the problem was. Can you please help me to
solve this issue?

afischer-sweepme, 2024-01-12:
I guess you need to setup the B1500 before communication is possible. Otherwise, the message is not processed and you are running into a timeout error.

I have summarized some necessary steps here:
https://sweep-me.net/device/23/Agilent_B1500

It is also described in the programming guide:
https://www.keysight.com/us/en/assets/9018-01851/programming-guides/9018-01851.pdf
page 1-3

NabilaTasnim, 2024-01-13:
@afischer-sweepme I tried to turn off the system controller as per your suggestion, but the problem is I could not make it happen. 

[image]

It is saying to use card vendor's tool, but I am not sure how to do this. Can you please help?

afischer-sweepme, 2024-01-13:
I don't know what it means, but sound like there is another tool installed that might have access to your PCI card. Have a look at the card whether you can see the name of the vendor or check whether you find any tool on your computer that sound like "GPIB configuration" or the Keysight IO Libraries Suite . Also check that there is no other program running at the same time that might access the GPIB card as system controller, e.g EasyEXPERT. Additionally, check the Windows system tray whether there is any hidden application running in the background.

As long as the system controller is active, the GPIB interface of the B1500 is not processing external commands. This mode is needed when the B1500 is used to control other external equipment.

If you can not find further help in the programming or user manual, I suggest that you contact the Keysight support. I am relatively sure that it is not a pyvisa problem and I guess they know this issue.

If you find a solution, I would be happy to know it. Sorry that I cannot help here more.

NabilaTasnim, 2024-01-14:
I have turned off the system controller using Hardware Configuration Utility, but if I do this then no instrument is found in GPIB port, also the Python code throws an error. Can you please help here? @LongnoseRob @afischer-sweepme 

[image]

The error message of the Python code - 

Traceback (most recent call last):
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\BasicMeasurement.py", line 10, in <module>
    b1500 = rm.open_resource(visa_address, timeout = 60000)
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\highlevel.py", line 3284, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\resources\resource.py", line 282, in open
    self.session, status = self._resource_manager.open_bare_resource(
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\highlevel.py", line 3209, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1851, in open
    ret = library.viOpen(
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\b1500\PycharmProjects\TestProjectNabila\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_NSYS_CNTLR (-1073807263): The interface associated with this session is not the system controller.

Process finished with exit code 1

afischer-sweepme, 2024-01-14:
Do you actually run the python script on the pc that is built-in with the B1500 or do you run your python script from an external computer?
If it is the first case, then this will not work via GPIB as it is not possible to send a command via the GPIB interface and receive it at the same one.

NabilaTasnim, 2024-01-16:
I am running the Python script from an external PC. But if turn off system controller, then B1500 is not found in GPIB port and Python script throws the above error while trying to connect to B1500. Can you please help to solve this issue? @afischer-sweepme @LongnoseRob

afischer-sweepme, 2024-01-16:
Have you restarted EasyEXPERT and closed it again to only see the dialog "Start EasyEXPERT"? This step is needed to make the B1500 ready for remote control from an external computer. You could further try to restart the instrument. If you already did this, I suggest to contact the Keysight support. I have no further ideas at the moment.

afischer-sweepme, 2024-01-16:
You need to start EasyEXPERT on the B1500 and then close it again on the B1500. A small dialog window with "Start EasyEXPERT" should remain on the screen of the B1500. Thi is needed to put the B1500 in a state where it can process external GPIB communication. Please have a look at the links in my first response where this procedure is explained as well.

NabilaTasnim, 2024-01-16:
Sorry if I am asking too silly questions but I am very new to B1500 and PyVISA.  The following screenshot is taken from your link. 
[image]

Should all of steps mentioned in 'communication' part be done on the B1500 or external computer?

afischer-sweepme, 2024-01-16:
All these steps are done on the B1500. The first 2 steps are needed to switch off the "system controller". If this is already done, you only need to do the last step, by making sure that you see the "Start EasyEXPERT" dialog.

NabilaTasnim, 2024-01-24:
At first, I wrote the following code to control B1500A. But it was throwing a timeout error and on the screen of B1500A, an error - "Undefined GPIB commands" was showing. So, I rewrote the code using the "command reference" section of the attached file and the code was working and I got the output. **Why didn't the previous code work for B1500A?** @afischer-sweepme @LongnoseRob @twam @canol 

My previous code:      
 ```
        import time
        import pyvisa
        visa_address = 'GPIB0::8::INSTR'

        # Create a VISA resource manager and open a connection to the instrument
        rm = pyvisa.ResourceManager()
        b1500 = rm.open_resource(visa_address)
        b1500.write("*RST")
        b1500.read_termination = '\r'
        b1500.write_termination = '\n'
        try:
            # Configure SMU1:HR to 3V and SMU4:HR to 0V
            b1500.write(":SOURce1:FUNC VOLT")
            b1500.write(":SOURce4:FUNC VOLT")
            b1500.write(':SOURce1:VOLTage:LEVel 3.0')  # Set SMU1:HR to 3V
            b1500.write(':SOURce4:VOLTage:LEVel 0.0')  # Set SMU4:HR to 0V

            # Apply the voltages for 3 seconds
            time.sleep(3)

            #Measure the current on SMU1:HR
            b1500.write(':MEASure1:CURRent?')
            current_smu1 = (b1500.read_raw())

            # Measure the current on SMU4:HR
            b1500.write(':MEASure4:CURRent?')
            current_smu4 = (b1500.read_raw())

            # Print the measured currents
            print(f"Current on SMU1:HR: {current_smu1} A")
            print(f"Current on SMU4:HR: {current_smu4} A")
```

My new code:
```
import pyvisa
import time

# Create a resource manager
rm = pyvisa.ResourceManager()

# Open a connection to the B1500A
# Replace 'GPIB0::8::INSTR' with the actual address of your B1500A
b1500a = rm.open_resource('GPIB0::8::INSTR')

# Set the timeout to 20 seconds
b1500a.timeout = 20000

# Reset the B1500A to its default settings
b1500a.write('*RST')

# Enable the channels for SMU1 and SMU2
b1500a.write('CN 1,2')

# Set SMU1 to force 0V
b1500a.write('DV 1,0,0')

# Set SMU2 to force 3V
b1500a.write('DV 2,0,3')

# Wait for 3 seconds
time.sleep(3)

# Start the measurement
b1500a.write('XE')

# Read the current measurement from SMU2
current = b1500a.query('TI 2,0')

# Print the measured current
print(f'The current flowing through the device is {current} A')

# Force 0 V on SMU1 and SMU2 to stop the voltage output
b1500a.write('DZ 1')
b1500a.write('DZ 2')

# Disable the channels
b1500a.write('CL 1,2')
```