pyvisa/pyvisa #801: Traceback on rm.open_resource with TCPIP resource

schlatterbeck, 2024-02-02:
When opening a TCPIP resource (both with an IP address or a DNS name) I'm getting a traceback
Accessing the same resource with librevisa works.

To Reproduce
--------------
```
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyvisa
>>> pyvisa.__version__
'1.14.1'
>>> rm = pyvisa.ResourceManager()
>>> inst = rm.open_resource('TCPIP::10.23.5.46')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ralf/.virtualenvs/visa/lib/python3.11/site-packages/pyvisa/highlevel.py", line 3255, in open_resource
    info = self.resource_info(resource_name, extended=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ralf/.virtualenvs/visa/lib/python3.11/site-packages/pyvisa/highlevel.py", line 3182, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ralf/.virtualenvs/visa/lib/python3.11/site-packages/pyvisa/ctwrapper/functions.py", line 2076, in parse_resource_extended
    ret = library.viParseRsrcEx(
          ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'IVIVisaLibrary' object has no attribute 'viParseRsrcEx'. Did you mean: 'viParseRsrc'?
```

**Output of `pyvisa-info`**
```
<!--- Output of the pyvisa-info command (to run in a command prompt) -->
Machine Details:
   Platform ID:    Linux-6.1.0-13-amd64-x86_64-with-glibc2.36
   Processor:      

Python:
   Implementation: CPython
   Executable:     /home/ralf/.virtualenvs/visa/bin/python3
   Version:        3.11.2
   Compiler:       GCC 12.2.0
   Architecture:   ('x86', 64)
   Build:          Mar 13 2023 12:18:29 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: /usr/local/lib/libvisa.so.0.0.0:
         found by: auto
         architecture:
            ('x86', 64)
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.7.1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available 
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```

MatthieuDartiailh, 2024-02-02:
From the output of `pyvisa-info` it looks like your implementation of VISA is somewhat broken. I suggest you reinstall it. You can also try to use the pyvisa-py backend.

schlatterbeck, 2024-02-02:
The pyvisa-py backend works.
Can you be more specific about the brokenness? This is a fresh install into a virtual env using pip just some minutes before the bug report. I'm getting the same error with the Debian python package (debian bookworm).

MatthieuDartiailh, 2024-02-02:
Yes it does and indeed librevisa does not support parseRscEx, I remember opening a ticket at one point but I did not follow up.

schlatterbeck, 2024-02-02:
Can I somehow force what backend is used to avoid using a pre-installed library? I need this lib for a binary I'm trying to replace with a python-based approach...

MatthieuDartiailh, 2024-02-02:
Yes by passing "@py" when creating the resource manager.

schlatterbeck, 2024-02-02:
And '@py' is the default when no visa library is installed? Is this documented somewhere?
My impression of what I had read is that without arguments to the resource manager a default compiled library is used and with '@py' the python implementation is used. So there isn't a default compiled binary lib and python is the default?

Looks like I can pass a library path to the ResourceManager constructor, at least when passing my installed lib I'm getting the same error message when trying to open_resource. Is there some documentation on this? I'm working with docstrings currently but it's hard to figure out the usage from that...

MatthieuDartiailh, 2024-02-02:
It is documented here https://pyvisa.readthedocs.io/en/latest/introduction/configuring.html