pyvisa/pyvisa #805: [COM] query_ascii error with Agillient Infiniium DSA90000A series Oscilloscope using pyvisa

harrywjs, 2024-03-03:
Hi, I'm learning how to use Python and Pyvisa to connect equipment to my main computer.

Recently, I have connected a few equipment including VNA and a signal generator, and it was successfully done.

However, now I'm struggling to connect the Agilent Oscilloscope, and it is not working well.
I want to save the Waveform data to my main computer, so I used as,

    def save_waveform_ascii(osc, file_index):
        vna_directory = 'D:\\Otest\\'
        filename = f'{file_index:04}.csv'
        full_path = vna_directory + filename
        try:
            osc.write(":WAVeform:SOURce CHANnel1")
            osc.write(":WAVeform:FORMat ASCII")
            osc.write(":DIGitize CHANnel1")
            osc.query("*OPC?")  # Wait for the operation to complete
            data = osc.query_ascii_values('WAVeform:DATA?', container=np.ndarray)
            np.savetxt(full_path, data, delimiter=',')
            print(f"Data saved to {full_path}")
        except visa.VisaIOError as e:
            print(f"Error saving waveform data: {e}")

When I ran the code for setup and capture, it is working well including 'CLS, RST, IDN'.
However, when I want to read data by "osc.query_ascii_values('WAVeform:DATA?)'", it always gives me an error as "self.read()" with Timeout error.

I already solved the same problem for Agilient E8362C VNA earlier, but I can't find the issue in this time.
For the VNA, I used the code as

    s_param_data = pna.query_ascii_values('CALC1:DATA? SDATA', container=np.ndarray)
    np.savetxt(full_path, s_param_data, delimiter=',')

I would greatly appreciate any insights, advice, or guidance you could offer on this matter. 

Instrument details
------------------
* Model: Agilient infiniium DSO91304A
* Communication: TCPIP
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Windows-10-10.0.19045-SP0
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\sdsw\AppData\Local\Programs\Python\Python312\python.exe
   Version:        3.12.2
   Compiler:       MSC v.1937 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Feb  6 2024 21:26:36 (#tags/v3.12.2:6abddd9)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 25165824
         Spec. Version: 7340544
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 25165824
         Spec. Version: 7340544
   py:
      Version: 0.7.1
      ASRL INSTR: Available via PySerial (3.5)
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

MatthieuDartiailh, 2024-03-04:
There is nothing obviously wrong with your code. However since ASCII transfer is slow and you are using the default timeout you may not manage to get the full waveform within the imparted time. You can try to increase the timeout or to read only a fixed number of bytes from the response using `read_bytes` to see if the instrument is answering anything.

harrywjs, 2024-03-05:
I solved the issue today. The version of equipment firmware might be the problem. After updating the software, the communication is completely done. The older version didn't handle the TCPIP "instrument", only can connect the "TCPIP Socket". Because of this, the communication rate maybe become lower. Thanks for your help!

closed 2024-03-05 by harrywjs