pyvisa/pyvisa #806: Linux IVI backend confusion

tnixa, 2024-03-05:
I have a Windows 11 laptop with python 3.11.8 and pyvisa 1.14.1 and I did not install any backends. I am able to communicate with instruments fine with this setup. 

I have a Linux RHEL8 laptop with python 3.11.5 and pyvisa 1.14.1 and I did not install any backends. I am not able to communicate with any instruments with this setup as I am getting an error saying `ValueError: Could not locate a VISA implementation. Install either the IVI binary or pyvisa-py.`

Do I need to manually install IVI backends with Linux? The pyvisa documentation says `Currently there are two backends available: The one included in pyvisa, which uses the IVI library (include NI-VISA, Keysight VISA, R&S VISA, tekVISA etc.),` I read this as pyvisa includes the IVI library which contains NI-VISA, Keysight VISA, R&S VISA, tekVISA etc. Or is that only true for Windows?

Here's my error and config on my RHEL8 laptop...

```
(.venv) pwrtest:test2$ pyvisa-info
Machine Details:
   Platform ID:    Linux-4.18.0-513.5.1.el8_9.x86_64-x86_64-with-glibc2.28
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/pwrtest/PycharmProjects/test2/.venv/bin/python
   Version:        3.11.5
   Compiler:       GCC 8.5.0 20210514 (Red Hat 8.5.0-20)
   Architecture:   ('x86', 64)
   Build:          Sep 22 2023 15:34:29 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
```

```
PyDev console: starting.
Python 3.11.5 (main, Sep 22 2023, 15:34:29) [GCC 8.5.0 20210514 (Red Hat 8.5.0-20)] on linux
import pyvisa
import pyvisa
rm = pyvisa.ResourceManager()
Traceback (most recent call last):
  File "/opt/pycharm-community-2023.3.4/plugins/python-ce/helpers/pydev/pydevconsole.py", line 364, in runcode
    coro = func()
           ^^^^^^
  File "<input>", line 1, in <module>
  File "/home/pwrtest/PycharmProjects/test2/.venv/lib/python3.11/site-packages/pyvisa/highlevel.py", line 2992, in __new__
    visa_library = open_visa_library(visa_library)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pwrtest/PycharmProjects/test2/.venv/lib/python3.11/site-packages/pyvisa/highlevel.py", line 2899, in open_visa_library
    wrapper = _get_default_wrapper()
              ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pwrtest/PycharmProjects/test2/.venv/lib/python3.11/site-packages/pyvisa/highlevel.py", line 2858, in _get_default_wrapper
    raise ValueError(
ValueError: Could not locate a VISA implementation. Install either the IVI binary or pyvisa-py.
```

arr-ee, 2024-03-05:
> I read this as pyvisa includes the IVI library which contains NI-VISA, Keysight VISA, R&S VISA, tekVISA etc.

This is not what the quote meant. VISA implementations listed are _supported_ by the IVI library that comes with pyvisa, but are not _provided_ by pyvisa — as they are separate proprietary software packages distributed by TE vendors.

Your Windows machine likely had NI VISA or other VISA implementation installed at some point, and that’s what pyvisa is using — output of pyvisa-info from the Windows box would go a long way to confirming this.

tnixa, 2024-03-06:
@arr-ee Thank you for the explanation! My windows does show NI packages under IVI library.

```
PS C:\Users\TerrenceNixa\Box\tnixa\_power\PyCharmProjects\Daro_3600w> pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.22631-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\TerrenceNixa\AppData\Local\Programs\Python\Python311\python.exe
   Version:        3.11.8
   Compiler:       MSC v.1937 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Feb  6 2024 22:03:32 (#tags/v3.11.8:db85d51)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 24118528
         Spec. Version: 7340032
      #2: C:\windows\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 24118528
         Spec. Version: 7340032
```

I just tried to install the NI-VISA for Linux package but its not showing up under ivi still. I used the instructions here:
https://www.ni.com/docs/en-US/bundle/ni-platform-on-linux-desktop/page/installing-ni-products-red-hat-enterprise-linux.html 

```
Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
```

tnixa, 2024-03-08:
> On Linux it may be easier to use Pyvisa-py as a backend.

OK, I'll try pyvisa-py. I'll close this issue but it would be nice if something could be added to the documentation to help differentiate _included_ vs _supported_ for the IVI library.

closed 2024-03-08 by tnixa