pyvisa/pyvisa #807: Extra DLL paths from .pyvisarc are never removed from DLL search path

prakhub, 2024-03-15:
# Problem Description
When specifying extra DLL paths in the `.pyvisarc`, the DLL paths are added to the DLL search path each time a new `ResourceManager` is instantiated, but never removed. Because of this, at some point the DLL search path will become too long for windows to handle, which will result in a WinError 206.

# Steps to Reproduce
1. Have some extra DLL paths in the `.pyvisarc`, for example the Keysight IO libs:<br> `dll_extra_paths: C:\Program Files (x86)\Keysight\IO Libraries Suite\bin`
2. Open and close many `ResourceManager` instances after each other
```
import pyvisa

for i in range(1000):
    print(i)
    rm = pyvisa.ResourceManager()
    rm.close()
```
On my system, at `i = 297` this will result in
```
FileNotFoundError: [WinError 206] The filename or extension is too long:  'C:\\Program Files (x86)\\Keysight\\IO Libraries Suite\\bin'
```

# Possible Fix
In my opinion the issue is here:
https://github.com/pyvisa/pyvisa/blob/6bca99233dcf66742db54d6d410c9840efb18cb5/pyvisa/util.py#L150
The call to `os.add_dll_directory` returns a `os._AddedDllDirectory` object, which needs to be closed according to the python docs for `os`:
>"Remove the directory by calling close() on the returned object or using it in a [with](https://docs.python.org/3/reference/compound_stmts.html#with) statement."
However, the  `os._AddedDllDirectory` is simply discarded here.

I am not yet sure about a fix, do we need to keep track of the `os._AddedDllDirectory` until the `ResourceManger` instance is closed?
Or could we return the path to the DLL library as a string, and later construct a `LibraryPath` object from the path in `get_library_paths`?

arr-ee, 2024-03-15:
We should keep track of what we've added already, and short-circuit if we try adding the same thing again. That function is not limited to (even if it's unlikely to be called outside of) creation of `ResourceManager`.

_Ideally_ we would do some eager configuration instead so that we have a single place we execute and store results of all the pathfinding etc, but it's a bit of an undertaking.

MatthieuDartiailh, 2024-03-15:
Do you want to make a PR @arr-ee ?

I agree we should track what we added to avoid duplication and ideally also use `atexit` to ensure we close the objects before exiting.

PR #811 (2024-09-17): Prevent duplicate paths from being added to the DLL paths
This PR adds very rudimentary tracking of dll paths we already added in order to prevent exceeding path length limit if we create `ResourceManager`s in a loop.

- [x] Closes #807
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [x] The change is fully covered by automated unit tests
- [ ] Documented in docs/ as appropriate
- [x] Added an entry to the CHANGES file
closed 2024-09-17 by MatthieuDartiailh