pyvisa/pyvisa #808: Support for multi-character read terminations?

neutrinonerd3333, 2024-03-15:
I'm working with an instrument that only provides responses terminated with CRLF (`'\r\n'`). I see that VISA doesn't support multi-character read terminations, instead treating the final character, e.g. LF, as a termchar. I also see that `pyvisa` provides some minimal support on top of that by raising a warning if a read yields a string that ends with the termchar but not the full termination string:

https://github.com/pyvisa/pyvisa/blob/6bca99233dcf66742db54d6d410c9840efb18cb5/pyvisa/resources/messagebased.py#L493-L497

In my case, the instrument actually returns multi-line responses (with LF as delimiter) terminated with a final CRLF, so the only workaround would be to do multiple VISA reads until we get a CRLF and concatenate the responses. I do plan on implementing this, but I am wondering: would it make sense for `pyvisa` to provide this kind of added support for multi-character read terminations? I think it would be natural to implement this workaround at the `pyvisa` level rather than in client code, but there's of course an argument to be made to keep the library more a wrapper around VISA routines rather than to introduce additional features/complexity/abstraction.

DavidLutton, 2024-03-17:
For the case of multiple reads to collect a response.
I did this last week for a power meter with known number of reads to collect calibration data from it:
``` python
inst.write('SI')   # Command for calibration data, follow with read() 31 times
calibration_data = [inst.read().strip() for i in range(1,32)]
```

neutrinonerd3333, 2024-03-21:
Just to be clear — I'm aware that one _can_ just call `MessageBasedResource.read()` multiple times to read a multi-line response. As I wrote, I'm opening this issue to ask whether it makes sense to _include_ support in `pyvisa` to handle the situation where a single logical response (terminated by a multi-char string) must be read by VISA as a sequence of individual reads because the final character of the termination string occurs in the body of the response.