pyvisa/pyvisa #809: 'ascii' codec can't decode byte 0xa1 in position 6: ordinal not in range(128) - Owon xdm1241

somervda, 2024-03-27:
I get the "'ascii' codec can't decode byte 0xa1 in position 6: ordinal not in range(128)" error on the Owon xdm1241 multimeter when performing this query

self._xdm1241.query('MEAS1:SHOW?')

This only occurs on the Temp and Resistance range where the meter returns a special character for Ohms (\u03A9) and Degrees (\u00B0)

Because they are greater than 128 I'm guessing it is not being interpreted as ascii codes

Full python traceback is

ERROR:    Exception in ASGI application
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/uvicorn/protocols/http/httptools_impl.py", line 411, in run_asgi
    result = await app(  # type: ignore[func-returns-value]
  File "/home/pi/.local/lib/python3.9/site-packages/uvicorn/middleware/proxy_headers.py", line 69, in __call__
    return await self.app(scope, receive, send)
  File "/home/pi/.local/lib/python3.9/site-packages/fastapi/applications.py", line 1054, in __call__
    await super().__call__(scope, receive, send)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/applications.py", line 123, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/middleware/errors.py", line 186, in __call__
    raise exc
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/middleware/errors.py", line 164, in __call__
    await self.app(scope, receive, _send)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/middleware/cors.py", line 83, in __call__
    await self.app(scope, receive, send)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/middleware/exceptions.py", line 62, in __call__
    await wrap_app_handling_exceptions(self.app, conn)(scope, receive, send)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/_exception_handler.py", line 64, in wrapped_app
    raise exc
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
    await app(scope, receive, sender)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/routing.py", line 758, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/routing.py", line 778, in app
    await route.handle(scope, receive, send)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/routing.py", line 299, in handle
    await self.app(scope, receive, send)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/routing.py", line 79, in app
    await wrap_app_handling_exceptions(app, request)(scope, receive, send)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/_exception_handler.py", line 64, in wrapped_app
    raise exc
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
    await app(scope, receive, sender)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/routing.py", line 74, in app
    response = await func(request)
  File "/home/pi/.local/lib/python3.9/site-packages/fastapi/routing.py", line 278, in app
    raw_response = await run_endpoint_function(
  File "/home/pi/.local/lib/python3.9/site-packages/fastapi/routing.py", line 193, in run_endpoint_function
    return await run_in_threadpool(dependant.call, **values)
  File "/home/pi/.local/lib/python3.9/site-packages/starlette/concurrency.py", line 42, in run_in_threadpool
    return await anyio.to_thread.run_sync(func, *args)
  File "/home/pi/.local/lib/python3.9/site-packages/anyio/to_thread.py", line 56, in run_sync
    return await get_async_backend().run_sync_in_worker_thread(
  File "/home/pi/.local/lib/python3.9/site-packages/anyio/_backends/_asyncio.py", line 2144, in run_sync_in_worker_thread
    return await future
  File "/home/pi/.local/lib/python3.9/site-packages/anyio/_backends/_asyncio.py", line 851, in run
    result = context.run(func, *args)
  File "/home/pi/scpipi/web.py", line 50, in xdm1241measure
    return xdm1241.measure()
  File "/home/pi/scpipi/xdm1241.py", line 175, in measure
    result = self._xdm1241.query('MEAS1:SHOW?')
  File "/home/pi/.local/lib/python3.9/site-packages/pyvisa/resources/messagebased.py", line 647, in query
    return self.read()
  File "/home/pi/.local/lib/python3.9/site-packages/pyvisa/resources/messagebased.py", line 485, in read
    message = self._read_raw().decode(enco)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xa1 in position 6: ordinal not in range(128)

To Reproduce
--------------
Steps to reproduce the behavior:
1. Connect to Owon xdm1241 multimeter with pyvisa
2. Set multimeter to Ohms or Resistance
3. Do a query('MEAS1:SHOW?')
4. See error above

arr-ee, 2024-03-27:
I think the culprit in the trace you showed is a different UTF-8 character, but your diagnosis sounds right to me.

You can change encoding used by the resource via [`encoding` property](https://pyvisa.readthedocs.io/en/latest/api/resources.html#pyvisa.resources.MessageBasedResource.encoding), try setting that to `utf-8`?

somervda, 2024-03-28:
Thanks - that got me on the right track. utf-8 had the same error but using euc_cn encoding worked.
Here are a couple of the results using this code

xpm1241.encoding="euc_cn"
print("xpm1241 reading:",xpm1241.query('MEAS1:SHOW?'))

xpm1241 reading: 20.771℃
xpm1241 reading: 2.1725kΩ

closed 2024-03-28 by somervda