pyvisa/pyvisa #810: Dev environment setup instructions are outdated

arr-ee, 2024-03-27:
Stumbled upon this while rebuilding my dev env working on something else: https://pyvisa.readthedocs.io/en/latest/faq/contributing.html#contribute-code references `setup.py`, which does not exist anymore.

We also seem to be lacking `pytest` in `dev-requirements.txt`.

Could also use this opportunity to move/copy that part to a [`CONRTRIBUTING` file](https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/setting-guidelines-for-repository-contributors#adding-a-contributing-file)

arr-ee, 2024-03-27:
+ update checklist in the PR template

MatthieuDartiailh, 2024-04-01:
In addition to #814 I subscribed to pre-commit.ci to avoid delayed updates to pre-commit configuration

closed 2024-04-01 by MatthieuDartiailh

arr-ee, 2024-04-01:
Awesome, thank you!

PR #814 (2024-04-01): Address #810 and issues seen in #812

- [x] Closes #810