pyvisa/pyvisa #813: error: unpack_from requires a buffer of at least 4 bytes

0xDEADBEEF-ASAP, 2024-03-31:
I got this error on one machine but not in the other. Both have the same software (freash arch install and other install)_ both machines are little endian.

Any ideas what to check?

MatthieuDartiailh, 2024-03-31:
Please post the output of pyvisa-info, the model of the instrument you are trying to address, the transport layer you use (tcpip, usb, etc), and the complete traceback of the error.

0xDEADBEEF-ASAP, 2024-03-31:
Hi! Here is a video of whole process : https://github.com/0xDEADBEEF-ASAP/MISC-1/blob/main/pyvisa.mp4

Layer: USB (pyusb, pyvis, pyvisa-py) installed with pip3 in venv

pyvisa-info:
'''
 Machine Details:
   Platform ID:    Linux-6.8.2-arch2-1-x86_64-with-glibc2.39
   Processor:      

Python:
   Implementation: CPython
   Executable:     /home/arch/usbtmc/.venv/bin/python
   Version:        3.11.8
   Compiler:       GCC 13.2.1 20230801
   Architecture:   ('x86', 64)
   Build:          Feb 12 2024 14:50:05 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.2
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available 
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
    '''

Traceback:

'''
$ sds.query('*IDN?')
 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/arch/usbtmc/.venv/lib/python3.11/site-packages/pyvisa/resources/messagebased.py", line 647, in query
    return self.read()
           ^^^^^^^^^^^
  File "/home/arch/usbtmc/.venv/lib/python3.11/site-packages/pyvisa/resources/messagebased.py", line 485, in read
    message = self._read_raw().decode(enco)
              ^^^^^^^^^^^^^^^^
  File "/home/arch/usbtmc/.venv/lib/python3.11/site-packages/pyvisa/resources/messagebased.py", line 441, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arch/usbtmc/.venv/lib/python3.11/site-packages/pyvisa_py/highlevel.py", line 513, in read
    data, status_code = self.sessions[session].read(count)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arch/usbtmc/.venv/lib/python3.11/site-packages/pyvisa_py/usb.py", line 168, in read
    return self._read(
           ^^^^^^^^^^^
  File "/home/arch/usbtmc/.venv/lib/python3.11/site-packages/pyvisa_py/sessions.py", line 792, in _read
    current = reader()
              ^^^^^^^^
  File "/home/arch/usbtmc/.venv/lib/python3.11/site-packages/pyvisa_py/usb.py", line 152, in _usb_reader
    return self.interface.read(count)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arch/usbtmc/.venv/lib/python3.11/site-packages/pyvisa_py/protocols/usbtmc.py", line 481, in read
    response = BulkInMessage.from_bytes(resp)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arch/usbtmc/.venv/lib/python3.11/site-packages/pyvisa_py/protocols/usbtmc.py", line 114, in from_bytes
    msgid, btag, btaginverse = struct.unpack_from("BBBx", data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: unpack_from requires a buffer of at least 4 bytes for unpacking 4 bytes at offset 0 (actual buffer size is 0)

$ sds.query('*IDN?')
'7\n'

'''

0xDEADBEEF-ASAP, 2024-03-31:
pyvisa-info from machine that its working:

Machine Details:
   Platform ID:    Linux-6.8.2-arch2-1-x86_64-with-glibc2.39
   Processor:      

Python:
   Implementation: CPython
   Executable:     /home/arch/usbtmc/.venv/bin/python
   Version:        3.11.8
   Compiler:       GCC 13.2.1 20230801
   Architecture:   ('x86', 64)
   Build:          Feb 12 2024 14:50:05 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.2
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available 
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

0xDEADBEEF-ASAP, 2024-03-31:
Only difference is that one of machines have a USB dongle (for mouse and keyboard and WiFi adapter) - this is not working machine, and second don't have any other USB devices (laptop) co nected - it works just fine.

Maybe this makes some problems?

arr-ee, 2024-03-31:
Can you also share
1. Instrument you’re using
2. Code you’re running in its entirety 

thanks

0xDEADBEEF-ASAP, 2024-03-31:
1. As I wrote above, its siglent SDS 1202X-E (family of devices: sds 1000ex) with latest firmware.
2. Code that I run on PC?
3. 
Its in video also (but no need to watch it, its boring - screen capture).

python console:

import pyvisa
rm = pyvisa.ResourceManager('@py')
rm.list_resources()
    #this produces a oitput>> 'USB0:: ***** ::INSTR'
sds=rm.open_resource('USB0:: ***** ::INSTR')
sds.query('*IDN?')

0xDEADBEEF-ASAP, 2024-04-01:
Without any other usb devices connected - the same error still shows up.

MatthieuDartiailh, 2024-04-01:
Is  your instrument connected through the dongle ? If yes can you try connecting it directly ? I have seen USB hubs mess up USBTMC links before.

0xDEADBEEF-ASAP, 2024-04-02:
Its connected directly - but maybe it is something messed up in hardware. There are usb connectors on back (directly on motherboard and usb connectors on front of pc case connected with motherboard with cables).

Oscilloscope is connected on back (directly to MB), but on front I connected usb dongle (wireless keyboard and mouse).

I will check it tomorrow and give a feedback !

arr-ee, 2024-04-02:
I managed to somewhat reproduce this behaviour with my SDS2000X+. It's quite odd since it appears like we're not managing some buffer properly: after several reads I get data sometimes, and that data sticks across several subsequent reads. Same commands to the same instrument via LAN work without issues.

Will report back once I get a better understanding of what's happening.

arr-ee, 2024-04-02:
Plugging the scope directly to the PC instead of through the hub helped with empty buffer at least.

Trying to read binary data from the scope is still messy, I get chunks of 64 bytes or less at a time + warnings about https://github.com/pyvisa/pyvisa-py/issues/20.

arr-ee, 2024-04-02:
Tried the same (different command, `HCOPy:DATA?`, but result is the same: 200+kB binary response) with an R&S NGE100B, and transfer worked as expected so will move this to a follow-up for https://github.com/pyvisa/pyvisa-py/issues/20.

Re: original error: Matthieu's diagnosis appears correct.

0xDEADBEEF-ASAP, 2024-04-02:
Can it be linked to hardware issues instead? My motherboard MSI B550M has some USB ports that go to CPU and some to b550 chipset ?

[image]

0xDEADBEEF-ASAP, 2024-04-03:
Tried all avaiable USB ports, on all ports device is detected. On all ports the same error shows up.

0xDEADBEEF-ASAP, 2024-04-18:
Updated BIOS to latest - but there is the same issue. Maybe I have to change some BIOS settings?

MatthieuDartiailh, 2024-05-15:
I am sorry to say I am at my wits end here. Let us know if you manage to find the origin of the issue.