pyvisa/pyvisa #815: PyVisa VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded. Even though Library installed

erikcura, 2024-04-02:
Even though I have installed the most recent version of the keysight libraries I still get this error:
VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.
I am not sure how to resolve this.

**Output of `pyvisa-info`**
(powersupply) PS C:\Users\erikc\Documents\Capstone\PowerSupplyCode> pyvisa-info
Machine Details:
   Platform ID:    Windows-11-10.0.22631-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\erikc\Documents\Capstone\PowerSupplyCode\powersupply\Scripts\python.exe
   Version:        3.12.2
   Compiler:       MSC v.1937 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Feb  6 2024 21:26:36 (#tags/v3.12.2:6abddd9)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Could not get more info:
            VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Could not get more info:
            VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.

arr-ee, 2024-04-02:
You can start from checking https://github.com/pyvisa/pyvisa/issues/196, and confirming that files mentioned do, in fact, exist. 

If you're running recent versions of NI VISA, https://github.com/pyvisa/pyvisa/issues/794 might also help with a workaround.

STC26, 2024-12-19:
Hello, 

Sorry for my English ...

Using Python script, the GPIB cable (82357B from Keysight) was not detected, but it was detected using Keysight Connection Expert 2023  (IO Libraries Version:	18.3.29517.2).

Here is my feedback for this error:
 "Erreur lors de la détection : VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded."

Finally, I found (with the help of ChatGPT):
In VS Code, Go to the command palette with Ctrl + Shift + P.
Search for Python: Select Interpreter.

In my case the selected interpreter didn't have to correct path, so I selected the onther one and now it works well.
So I didn't select the "Recommanded" , but the "Mondial" to have the correct path.
[image]

See below the pictures:
with the error:
[image]

with the problem fixed:
[image]

I hope this can be helpfull ...