pyvisa/pyvisa #823: Large write_raw instantaneously fail with VI_ERROR_TMO

Scafir, 2024-05-15:
`write_binary_values` for large amount of data (more than 46'699 bytes) immediatly fail with VI_ERROR_TMO.

What bothered me with this particular issue was that no matter the timeout that I set on my device, I would always immediately get this error, which did not make sense. Turns out, it seem to be a bug in NI visa library. 

To Reproduce
--------------
Adapt and run the following code (any instrument should work, but I did not test that):
```py
import pyvisa
import numpy as np

deviceString = "THURLBY THANDAR, TG5011"
read_termination = '\r\n'
write_termination = '\n'

rm = pyvisa.ResourceManager()

# Also tested in serial mode
instr = rm.open_resource("TCPIP::172.16.10.0::9221::SOCKET",
                         read_termination = read_termination,
                         write_termination = write_termination)
idString = instr.query("*IDN?")

if not idString.startswith(deviceString):
    raise RuntimeError

goodData = np.zeros(23343)
badData  = np.zeros(23344)

instr.write_binary_values('ARB1 ', goodData, datatype='H', is_big_endian=True)
print("PASSED GOOD DATA")
print("Following will fail")

instr.write_binary_values('ARB1 ', badData, datatype='H', is_big_endian=True)

print("This point will not be reached")
```

This should immediatly fail when writing bad data with VI_ERROR_TMO

Workaround
--------------
As the bug lies within the NI library, it is not something that can truly be fixed within pyvisa. However, it is simple enough to work around. One can monkey patch the existing `write_raw` function to only send messages small enough to not trigger the bug withing NI's library:
```py
def write_rawWorkAround(self, message: bytes) -> int:
    """Write a byte message to the device.

    Parameters
    ----------
    message : bytes
        The message to be sent.

    Returns
    -------
    int
        Number of bytes written

    """

    chunks = [message[i:i+20000] for i in range(0, len(message), 20000)]

    count = 0

    for c in chunks:
        count += self.visalib.write(self.session, c)[0]

    return count
```

Then, before writing the large message to the device, monkey-patch (overwrite) the rewritten `write_raw` function:
```py
MessageBasedResource.write_raw = write_rawWorkAround
```

It seems that very large writes on my device can make it unresponsive for some time after the actual write. Which requires me to sleep for a second or so (for 46'700 bytes) before continuing.

**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Linux-5.14.0-362.24.2.el9_3.x86_64-x86_64-with-glibc2.34
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        3.9.18
   Compiler:       GCC 11.4.1 20230605 (Red Hat 11.4.1-2)
   Bits:           64bit
   Build:          Jan  4 2024 00:00:00 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.23.5.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 24118528
         Spec. Version: 7340032
   py:
      Version: 0.7.1
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available 
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```

Thank you for this awesome library :)

horvathp, 2024-07-29:
As an additional data point, I'm facing the same issue despite using the pyvisa-py backend. The instrument is an R&S generator, and I talk to it over hislip. It looks like sending fails around 250KB (a bit larger size than the OP mentioned, but it still fails).

2xB, 2025-02-22:
Although I am not fully confident in this, there is a non-zero chance that this is solved by https://github.com/pyvisa/pyvisa-py/pull/490 . @Scafir and @horvathp , it may be interesting if you could try installing `pyvisa-py` from source since the merge of that pull request and see if that changes anything?

horvathp, 2025-02-25:
Hi @2xB , I just tried it with an Ethernet-connected instrument (this time under Linux), and I was able to upload a large file that has not been working before. Big thanks for fixing this!

2xB, 2025-02-25:
@horvathp Oh, interesting, I somehow missed the important part that you used an Ethernet connection - my fix should only work for USB. Nevertheless, great that the issue is gone for you!

closed 2025-02-25 by MatthieuDartiailh